google-deepmind/mujoco #1243: Attaching a cable to two ends of a bimanual arm

shamilmamedov, 2023-11-29:
Hi,

I'm a student and I'm trying to use MuJoCo for bimanual cable shaping using ABB Yumi.

I'm looking for some help with attaching a cable to two end-effectors of bimanual arm Yumi. 

To be more precise, we want to recreate real cable shaping setup we have in Mujoco. In the left figure below, you'll find the actual setup, while the right displays a visualization in Rviz.

[image]

To recreate this setup, we utilized the cable plugin: setting a cable as a child of the left arm's end-effector and attempting to connect its free end to the right arm's end-effector using an equality constraint (connect or weld). Unfortunately, the constraint did not perform as anticipated. Is there a method to immediately enforce the constraint at the simulation's onset, even if it isn't precisely satisfied for the initial configuration? Is there a better way to implement this closed loop?

Additionally, we experimented with setting `model.qpos0` to a specific value to prevent collisions in the initial configuration. However, upon calling `mujoco.viewer.launch(model, data)`, the simulator reverted to the default `qpos0`. What could be causing this?

Below are model description, main and zip with meshes of Yumi.

<details>
  <summary>XML file</summary>
  <mujoco model="yumi">
    <extension>
        <plugin plugin="mujoco.elasticity.cable"/>
    </extension>

    <statistic meansize="1"/>

    <compiler angle="radian" />
    <visual>
        <quality shadowsize="10000"/>
    </visual>
    <option gravity="0 0 0" iterations="20" integrator="Euler" />
    <size njmax="500" nconmax="100" />
    <asset>
        <mesh name="coarse_body" file="meshes/coarse/body.stl" />
        <mesh name="coarse_link_1" file="meshes/coarse/link_1.stl" />
        <mesh name="coarse_link_2" file="meshes/coarse/link_2.stl" />
        <mesh name="coarse_link_3" file="meshes/coarse/link_3.stl" />
        <mesh name="coarse_link_4" file="meshes/coarse/link_4.stl" />
        <mesh name="coarse_link_5" file="meshes/coarse/link_5.stl" />
        <mesh name="coarse_link_6" file="meshes/coarse/link_6.stl" />
        <mesh name="coarse_link_7" file="meshes/coarse/link_7.stl" />
        <mesh name="base" file="meshes/base.stl" />
        <mesh name="finger" file="meshes/gripper/finger.stl" />
    </asset>
    <worldbody>
        <light diffuse=".5 .5 .5" pos="2 -2 2" dir="-1  1 -1"/>
        <light diffuse=".5 .5 .5" pos="2  2 2" dir="-1 -1 -1"/>
        <geom rgba="1 1 1 1" name="table" type="plane" pos="0 0 0" size="0.4 0.5 0.1" contype="1" conaffinity="1"/>
        <body name="yumi_base_link" pos="-0.4 0 0">
            <inertial pos="0 0 0" mass="0" diaginertia="0 0 0" />
            <body name="yumi_body" pos="0 0 0">
                <inertial pos="0 0 0" quat="0.707107 0 0 0.707107" mass="15" diaginertia="0.6 0.5 0.3" />
                <geom name="yumi_body" type="mesh" rgba="0.8 0.8 0.8 1" mesh="coarse_body" />
                <body name="yumi_link_1_r" pos="0.05355 -0.0725 0.41492" quat="0.219255 -0.407796 0.31429 -0.828765">
                    <inertial pos="0 -0.03 0.12" mass="2" diaginertia="0.1 0.1 0.1" />
                    <joint type="hinge" name="yumi_joint_1_r" pos="0 0 0" axis="0 0 1" limited="true" range="-2.94088 2.94088"/>
                    <geom name="yumi_rlink_1" type="mesh" mesh="coarse_link_1" />
                    <body name="yumi_link_2_r" pos="0.03 0 0.1" quat="0.707107 0.707107 0 0">
                        <inertial pos="0 0 0" mass="2" diaginertia="0.1 0.1 0.1" />
                        <joint type="hinge" name="yumi_joint_2_r" pos="0 0 0" axis="0 0 1" limited="true" range="-2.50455 0.759218"/>
                        <geom name="yumi_rlink_2" type="mesh" mesh="coarse_link_2" />
                        <body name="yumi_link_3_r" pos="-0.03 0.17283 0" quat="0.707107 -0.707107 0 0">
                            <inertial pos="0 0 0" mass="2" diaginertia="0.1 0.1 0.1" />
                            <joint type="hinge" name="yumi_joint_7_r" pos="0 0 0" axis="0 0 1" limited="true" range="-2.94088 2.94088"/>
                            <geom name="yumi_rlink_3" type="mesh" mesh="coarse_link_3" />
                            <body name="yumi_link_4_r" pos="-0.04188 0 0.07873" quat="0.5 0.5 -0.5 0.5">
                                <inertial pos="0 0 0" mass="2" diaginertia="0.1 0.1 0.1" />
                                <joint type="hinge" name="yumi_joint_3_r" pos="0 0 0" axis="0 0 1" limited="true" range="-2.15548 1.39626"/>
                                <geom name="yumi_rlink_4" type="mesh" mesh="coarse_link_4" />
                                <body name="yumi_link_5_r" pos="0.0405 0.16461 0" quat="0.707107 -0.707107 0 0">
                                    <inertial pos="0 0 0" mass="2" diaginertia="0.1 0.1 0.1" />
[... 112 lines cut ...]
model = mujoco.MjModel.from_xml_path(xml_path)
# MjModel contains the model description (quantities that do not change over time)
# model.qpos0[0] = -0.88
# model.qpos0[1] = 0.007
# model.qpos0[7] = 0.8
# model.qpos0[9] = -0.08
# model.qpos0[11] = -2.8

data = mujoco.MjData(model)
# MjData contains the state and quantites that depend on it
# NOTE the derived quantities in mjData need to be explicitely
# propagated by calling (mj_kineamtics)
data.qpos[0] = -0.88
data.qpos[1] = 0.007
data.qpos[7] = 0.8
data.qpos[9] = -0.08
data.qpos[11] = -2.8

mujoco.viewer.launch(model, data)

</details>

[yumi_meshes.zip](https://github.com/google-deepmind/mujoco/files/13504832/yumi_meshes.zip)

We would greatly appreciate any assistance you could provide.

quagla, 2023-11-30:
Hi, would you please be able to create a minimal XML that does not depend on python or external assets/meshes? Maybe try attach the cable to cylinder or a box and see if you can replicate the issue and then let me know and I will look into it. Thanks!

shamilmamedov, 2023-12-02:
Hi @quagla, thank you for quick reply. In simple examples, such as the one below (where the scene.xml is from the cable example), I encounter no issues. However, when dealing with more complex bimanual cable manipulation scenarios, ensuring satisfaction of the equality constraint in the initial configuration isn't straightforward. Is it necessary for these constraints to be satisfied initially? Are there functionalities within Mujoco that can aid in this process, such as an inverse kinematics solver?

<details>
  <summary>XML file</summary>
      <include file="scene.xml"/>
    
      <extension>
        <plugin plugin="mujoco.elasticity.cable"/>
      </extension>
    
      
      <statistic meansize="1"/>
      <option integrator="implicit" />
    
      <visual>
        <global elevation="-30"/>
      </visual>
    
      <compiler autolimits="true"/>
    
      <size memory="2M"/>
    
      <worldbody>
        <body name="left_gripper" pos="0. 0 .6">
          <joint name="left_gripper_joint" type="free" stiffness="0" damping="0" frictionloss="0" armature="0"/>
          <geom name="red_box" type="box" size=".01 .01 .015" rgba="1 0 0 1"/>
            <composite type="cable" curve="s 0 0" count="31 1 1" size="0.5" offset="0 0 -0.025" initial="none">
                <plugin plugin="mujoco.elasticity.cable">
                    <config key="twist" value="1e7"/>
                    <config key="bend" value="2e6"/>
                    <config key="vmax" value="0.05"/>
                </plugin>
                <joint kind="main" damping=".015"/>
                <geom type="capsule" size=".005" rgba=".8 .2 .1 1" condim="1"/>
            </composite>
        </body>
        
        <body name="right_gripper" pos="0.5 0 .6">
          <joint name="right_gripper_joint" type="free" stiffness="0" damping="0" frictionloss="0" armature="0"/>
          <geom name="green_box" type="box" size=".01 .01 .015" rgba="0 1 0 1"/>
        </body>
      </worldbody>
    
      <equality>
        <weld name="right_boundary" body1="B_last" body2="right_gripper" anchor="0.0 0 -0.025"/>
      </equality>
       
      <actuator>
        <velocity name="right_vx" joint="right_gripper_joint" gear="1 0 0 0 0 0" ctrlrange="-.5 .5" kv="30"/>
        <velocity name="right_vy" joint="right_gripper_joint" gear="0 1 0 0 0 0" ctrlrange="-.5 .5" kv="30"/>
        <velocity name="right_vz" joint="right_gripper_joint" gear="0 0 1 0 0 0" ctrlrange="-.5 .5" kv="30"/>
        <velocity name="left_vx" joint="left_gripper_joint" gear="1 0 0 0 0 0" ctrlrange="-.5 .5" kv="30"/>
        <velocity name="left_vy" joint="left_gripper_joint" gear="0 1 0 0 0 0" ctrlrange="-.5 .5" kv="30"/>
        <velocity name="left_vz" joint="left_gripper_joint" gear="0 0 1 0 0 0" ctrlrange="-.5 .5" kv="30"/>
      </actuator>
</details>

quagla, 2023-12-08:
It might be a bit cumbersome but if you want to set the initial position, try to load and save a cable, you will see the structure of an XML file where the cable composite is expanded to a sequence of bodies. You can edit the `pos` of those bodies to get the initial configuration that you wish. Remember to use `flat="true"` in the cable plugin if you do that (if your cable has a straight equilibrium configuration).

Also, if you want a weld constraint at one endpoint of the cable, you can set `initial="none"` in the composite, see https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-composite.

quagla, 2024-01-02:
Closing due to a lack of activity.

closed 2024-01-02 by quagla