google-deepmind/mujoco #1245: Missing PID plugin in pip installation

giorgionicola, 2023-11-30:
Hi,

I'm a researcher and I'm trying to use MuJoCo for defromable object simulation.

I'm looking for some help with to implement PID controllers. I have looked on the [example models](https://github.com/google-deepmind/mujoco/blob/main/model/plugin/actuator/pid.xml) on the repo and pid controllers are used via a plugin. However such a plugin is not available in version 3.0.1 installed via pip. Is it meant to be like that, and pid controllers should be done differently?

traversaro, 2023-11-30:
I am not sure, but it does not seem that 3.0.1 release contains that example model: https://github.com/google-deepmind/mujoco/blob/3.0.1/model/plugin/actuator/pid.xml .

giorgionicola, 2023-11-30:
Ops I did not notice that the commit introducing the pid plugin was very new. I believe you are right and it will be included in the next release. Thanks.

closed 2023-11-30 by giorgionicola