google-deepmind/mujoco #1246: How to create 3D flexes made of tetrahedra from meshes?

DFREANDY, 2023-11-30:
Hi,

I'm a student and I'm trying to use MuJoCo for simulating the deformable objects.

I'm looking for some help with 3D flexes made of tetrahedra from meshes.

I've noticed that you give an example with the Stanford Bunny when explaining the mechanism of 3D flexes made of tetrahedra from meshes. So, could you share the related files of the Stanford Bunny? I want to know the file format which you used to create it. 

Thanks a lot!

closed 2023-11-30 by yuvaltassa

josephrocca, 2024-01-21:
Hi @yuvaltassa, RE this sentence from the [overview](https://github.com/google-deepmind/mujoco/blob/main/doc/overview.rst#flex):

> Automation is provided to load a mesh from a file, construct bodies corresponding to the mesh vertices, construct flex elements corresponding to the mesh faces (or lines or tetrahedra, depending on dimensionality), and obtain a corresponding deformable mesh.

I'm probably misunderstanding, but this seems to imply there's something built-in for this now which makes external use of Gmsh/fTetWild/CoACD unnecessary in 3.0?

yuvaltassa, 2024-02-09:
Ah this is indeed unclear. 

> Automation is provided to load a mesh from a file

Should be "Automation is provided to load a _volumetric_ mesh from a file". You need to bring your own mesh. We are currently working on improving our support for the gmsh format.