google-deepmind/mujoco #1249: force sensor problem

marcos724, 2023-12-01:
Hi,

I'm a student  and I'm trying to use MuJoCo for the task peg in hole using a manipulator(ur5).

I'm looking for some help with force sensor.
Currently i am working on force sensor task in MuJoCo(peg_in_hole),  but i have been struggling with some problems. i made some summaries below explaining. 
1. i have write an 'xml'  adding a `body,`` site` (to work as a sensor) as mentioned in #399:: 

```
    <body name="dummy body" pos="0 0 0.045">
      <inertial pos="0 0 0" mass="0.0001" diaginertia="0 0 0" />
        <geom type="box" pos="0 0 0" size="0.01 0.01 0.01" rgba="0 1 0 0.5"/>
        <site name="ft_sensor" pos="0 0 0.0" rgba="1 0 0 0.3" size="0.01 0.01 0.01" type="box"/>  
    </body>
.......
    <sensor>
        <force name = "force" site="ft_sensor" />
    </sensor>
```

2.  with the changes made in xml file (as shown above) i run in the simulation to measure the force. but i there are some wrong data obtained.
 a) to measure the force, i run the code as shown in a) to measure the force, i run the code as explained in  https://colab.research.google.com/github/google-deepmind/mujoco/blob/main/python/tutorial.ipynb#scrollTo=z-zoBCuBv2Xi: 

```
     force=data.sensor("force").data
     print("-----print force--:  ", force)  

```
  b) running in to the simulation, i obtained a force information without making a contact with other body:  
      info. obtained` [  6.98633476  -0.33842342 -28.42605504]`
   
[image]

  c) when there is a contact.  the force tends to be a huge  info. and the force direction reaction seems to be wrong:
     info. obtained `[-907020.62997521  204220.28079864 1111326.38752532]`
[image]

**would you please help out whether the way i try to get a correct info is right? or, there are other methods i ignore.**

best regards

sunnymints, 2023-12-02:
I  meet same question

yuvaltassa, 2023-12-02:
Could you please make a minimal example so that we can load and reproduce?

sunnymints, 2023-12-02:
> Could you please make a minimal example so that we can load and reproduce?
Thanks for your help!

There are my  description and code of minimal example:

We use hinge joints to directly fix the axis to the square, and the contact force is vertical upwards (I understand it to be the correct direction),as shown in  this fig(xml:peg_square.xml)

[image]

When we load the hinge joint of the axis onto UR5 and take the contact square at the same angle, the direction of force changes(xml:robot_square.xml,code:demo).

[image]

The code and xml:

[mujoco_demo.zip](https://github.com/google-deepmind/mujoco/files/13536195/mujoco_demo.zip)

marcos724, 2023-12-02:
@yuvaltassa 
thanks for your prompt reply:
here is  the xml file:
```
<mujoco model="ur5">
  <option gravity="0 0 0"/>
  <include file="scene.xml"/>
  <compiler angle="radian" meshdir="/home/marcos/Documents/mujoco/collision/"/>
  <statistic meansize="0.2" extent="0.4" center="-3.42017e-10 -0.0501512 -1.90133e-10"/>
  <asset>
    <mesh name="fk" file="fk.STL"/>
  </asset>
  <statistic meansize="0.161957" extent="1.0432" center="0.388318 0.0842553 0.0891814"/>
  <asset>
    <mesh name="base" file="base.stl"/>
    <mesh name="shoulder" file="shoulder.stl"/>
    <mesh name="upperarm" file="upperarm.stl"/>
    <mesh name="forearm" file="forearm.stl"/>
    <mesh name="wrist1" file="wrist1.stl"/>
    <mesh name="wrist2" file="wrist2.stl"/>
    <mesh name="wrist3" file="wrist3.stl"/>
    <mesh name="peg20231110" file="peg20231110.STL"/>
  </asset>
  <worldbody>  
    <geom quat="0 0 0 1" type="mesh" contype="0" conaffinity="0" group="1" density="0" rgba="0.7 0.7 0.7 1" mesh="base"/>
    <geom quat="0 0 0 1" type="mesh" rgba="0.7 0.7 0.7 1" mesh="base"/>
    <body name="shoulder_link" pos="0 0 0.089159" quat="0 0 0 1">
      <inertial pos="0 0 0" mass="3.7" diaginertia="0.0149724 0.0149724 0.0104063"/>
      <joint name="shoulder_pan_joint" pos="0 0 0" axis="0 0 1" range="-6.28319 6.28319"/>
      <geom quat="0 0 0 1" type="mesh" contype="0" conaffinity="0" group="1" density="0" rgba="0.7 0.7 0.7 1" mesh="shoulder"/>
      <geom quat="0 0 0 1" type="mesh" rgba="0.7 0.7 0.7 1" mesh="shoulder"/>
      <body name="upper_arm_link" quat="0.707107 0.707107 0 0">
        <inertial pos="-0.2125 0 0.136" quat="0.707107 0 0.707107 0" mass="8.393" diaginertia="0.133886 0.133886 0.0151074"/>
        <joint name="shoulder_lift_joint" pos="0 0 0" axis="0 0 1" range="-6.28319 6.28319"/>
        <geom pos="0 0 0.136" quat="0.5 0.5 -0.5 -0.5" type="mesh" contype="0" conaffinity="0" group="1" density="0" rgba="0.7 0.7 0.7 1" mesh="upperarm"/>
        <geom pos="0 0 0.136" quat="0.5 0.5 -0.5 -0.5" type="mesh" rgba="0.7 0.7 0.7 1" mesh="upperarm"/>
        <body name="forearm_link" pos="-0.425 0 0">
          <inertial pos="-0.2125 0 0.0165" quat="0.707107 0 0.707107 0" mass="2.275" diaginertia="0.0312168 0.0312168 0.004095"/>
          <joint name="elbow_joint" pos="0 0 0" axis="0 0 1" range="-3.14159 3.14159"/>
          <geom pos="0 0 0.0165" quat="0.5 0.5 -0.5 -0.5" type="mesh" contype="0" conaffinity="0" group="1" density="0" rgba="0.7 0.7 0.7 1" mesh="forearm"/>
          <geom pos="0 0 0.0165" quat="0.5 0.5 -0.5 -0.5" type="mesh" rgba="0.7 0.7 0.7 1" mesh="forearm"/>
          <body name="wrist_1_link" pos="-0.39225 0 0.10915">
            <inertial pos="0 0 0" quat="0.5 0.5 -0.5 0.5" mass="1.219" diaginertia="0.00342844 0.00262426 0.00262426"/>
            <joint name="wrist_1_joint" pos="0 0 0" axis="0 0 1" range="-6.28319 6.28319"/>
            <geom pos="0 0 -0.093" quat="0.707107 0.707107 0 0" type="mesh" contype="0" conaffinity="0" group="1" density="0" rgba="0.7 0.7 0.7 1" mesh="wrist1"/>
            <geom pos="0 0 -0.093" quat="0.707107 0.707107 0 0" type="mesh" rgba="0.7 0.7 0.7 1" mesh="wrist1"/>
            <body name="wrist_2_link" pos="0 -0.09465 -1.9413e-11" quat="0.707107 0.707107 0 0">
              <inertial pos="0 0 0" quat="0.5 0.5 -0.5 0.5" mass="1.219" diaginertia="0.0021942 0.00178515 0.00178515"/>
              <joint name="wrist_2_joint" pos="0 0 0" axis="0 0 1" range="-6.28319 6.28319"/>
              <geom pos="0 0 -0.095" type="mesh" contype="0" conaffinity="0" group="1" density="0" rgba="0.7 0.7 0.7 1" mesh="wrist2"/>
              <geom pos="0 0 -0.095" type="mesh" rgba="0.7 0.7 0.7 1" mesh="wrist2"/>
              <body name="wrist_3_link" pos="0 0.0823 -1.688e-11" quat="0.707107 -0.707107 0 0">
                <inertial pos="0 0 -0.00627169" quat="0.5 0.5 -0.5 0.5" mass="0.3389" diaginertia="0.000132134 0.00011486 0.00011486"/>
                <joint name="wrist_3_joint"  pos="0 0 0" axis="0 0 1" range="-6.28319 6.28319"/>
                <geom pos="0 0 -0.0818" quat="0.707107 0.707107 0 0" type="mesh" contype="0" conaffinity="0" group="1" density="0" rgba="0.7 0.7 0.7 1" mesh="wrist3" friction=".33"/>
                <geom pos="0 0 -0.0823" quat="0.707107 0.707107 0 0" type="mesh" rgba="0.7 0.7 0.7 1" mesh="wrist3"/>
                <geom size="0.005 0.005 0.005" pos="0 0 -0.01" quat="0.5 0.5 -0.5 0.5" type="box"/>
                <geom pos="0 0 0.005" quat="0.5 0.5 -0.5 0.5" type="mesh" contype="0" conaffinity="0" group="1" density="0" rgba="0.7 0.7 0.7 1" mesh="peg20231110"/>
                <geom pos="0 0 0.005" quat="0.5 0.5 -0.5 0.5" type="mesh" rgba="0.7 0.7 0.7 1" mesh="peg20231110" friction=".33"/>
                <site name="endeff" pos="0. 0.0 0" size="0.001"/>
                <body name="dummy body" pos="0 0 0.045">
                  <inertial pos="0 0 0" mass="0.0001" diaginertia="0 0 0" />
                  <geom type="box" pos="0 0 0.0" size="0.01 0.01 0.01" rgba="0 1 0 0.5"/>
                  <site name="ft_sensor" pos="0 0 0.00" rgba="0 0 1 0.3" size="0.01 0.01 0.01" type="box"/>  
                </body>
              </body>
            </body>
          </body>
        </body>
      </body>
[... 43 lines cut ...]
force=data.sensor("force").data
print("---print force: ", force) 
viewer.sync()
#-----------------------position2
for i in range(6):
    data.qpos[i]=pos1[i]   
    data.ctrl[i]= data.qpos[i]  
mj.mj_step(model, data)
force=data.sensor("force").data
print("---print force: ", force) 
viewer.sync()
#-----------------------position3
for i in range(6):
    data.qpos[i]=pos2[i]   
    data.ctrl[i]= data.qpos[i]  
mj.mj_step(model, data)
force=data.sensor("force").data
print("---print force: ", force) 
viewer.sync()
time.sleep(10)
```
`stl. `files attached 
[arm_tool.zip](https://github.com/google-deepmind/mujoco/files/13536246/arm_tool.zip)

best regards!

marcos724, 2023-12-07:
@yuvaltassa 
Hello!
Sorry for continuing to insist. but i am still struggling into the force sensor problem. 
following the #399 . and after may attempts  i still get a wrong force data.
i use a ur5 robot manipulator added a peg(`body)` as an end-effector; at the same time i add the force sensor(`site`) at the end of the peg as shown in the follow figure.
but the force data received is `force[-0.49101407  0.12644676  1.65185966]` and `ntf_force [-0.49101407  0.12644676  1.65185966]` 
1. As the contact is with a planar surface i kept receiving force in `Fx Fy`  isn't the force should be just in `Fz` direction?
2. if the data i have receive is not wrong. i wonder how to just get the force received at  the peg in order to assume the force just between the peg and the planar surface ?
[image]

yuvaltassa, 2023-12-07:
No no, thanks for insisting!

I very much want to diagnose this. Do you think you could make an actually **minimal** test case? Pure XML, no meshes, no arm. Something like a worldbody with a floor (plane) geom, a single vertical slider joint moving a body up and down (position controlled?), one static body at the bottom that can touch the floor with a force sensor attached.

Context:
1. Having the arm creates ambiguity, I need to be able to precisely know what the force sensor is expected to measure (vs what it is measuring).
2. If I'm to write a proper test for this in `mujoco/test/` I cannot have any meshes which come from a copyrighted source.

marcos724, 2023-12-07:
sure!
here is a xml file: simple peg and static body added the sensor at the bottom of the peg.
```
<mujoco model="ur5">
  <option gravity="0 0 0"/>
  <worldbody>
    <body name="peg_link" pos="0.02 0 0.24" >
      <joint pos="0 0 0.0" type="slide" axis="0 0 1"/>
      <geom euler="0 0 0" pos="0 0 0." name="peg" type= "cylinder" size="0.01 0.025" rgba="0.7 0.7 0.7 0.8" contype="0" conaffinity="0"/>
      <inertial pos="0 0 0.015" mass="0.424" diaginertia="0.00019875 0.00019875 0.00019080"/>
      <body name="dummy_body" pos="0 0 -0.025">
        <inertial pos="0 0 0" mass="0.000000001" diaginertia="0.000000001 0.000000001 0.000000001"/>
        <geom  pos="0 0 0.0025" size=".01 .01 .0025" type="box" rgba="0 0 1 0.3"/>                   
        <site name="contact_sensor" pos="0 0 0.0" rgba="1 0 0 1" size="0.0005 0.0005 0.0005" type="sphere"/> 
      </body>
    </body>
    <body name="box_obj" pos="0 0 0.1">
      <geom pos="0 0 0.0" name="box_obj" type="box" size= "0.1 0.1 0.1" rgba="0.7 0.7 0.7 1" />
    </body>
  </worldbody>
  <sensor>
        <force name="contact_force" site="contact_sensor"/>
	</sensor>
</mujoco>
```
after running with simple xml (one peg with an static body ) the obtained force  data seems to be right. but the problem comes ; when i want to add the peg into the robot manipulator( added at ee_link), the force data and the force direction tends to be wrong(not the same as when i run a simple peg and static body). 
perhaps after adding the peg into the manipulator.  the manipulator  robot affects the expected force data we want to obtain? if that so, is there a method or a way to ignore  it, and just obtain the force data between the peg and the box.

thanks for your time!
regards.

yuvaltassa, 2023-12-08:
Hmm, so this example does not really reproduce the issue, right?

I wonder if there are small vibrations coming from the arm controller? Could you try reducing the timestep to see if that changes anything?

marcos724, 2023-12-09:
- yes.
- I just run a simple `mj_forward` with `data.qpos` joint position,  as shown below.  And still the force data seems to be wrong. 
```
-model = mj.MjModel.from_xml_path('.........../ur5_rob.xml')
data = mj.MjData(model)
viewer = mj_v.launch_passive(model, data)
pos0=[0.169, -1.224, -2.087, -1.400, 1.57, 1.74]
pos1=[ 0.16880386, -1.22476103, -2.08876971, -1.39885805,  1.57080307,
        1.73958779]
#----------------------position 1
for i in range(6):
    data.qpos[i]=pos0[i]   
mj.mj_forward(model, data) 
force=data.sensor("contact_force").data
print("---print force: ", force) 
viewer.sync()
#-----------------------position2
for i in range(6):
    data.qpos[i]=pos1[i]   
mj.mj_forward(model, data)
force=data.sensor("contact_force").data
print("---print force: ", force) 
viewer.sync()
```
I appreciate your time and help. 

Regards!

yuvaltassa, 2023-12-10:
Okay, could you try to make the simplified model slightly more elaborate, say two hinge joints rather than one slider, and see if you can reproduce the issue?

marcos724, 2023-12-11:
- after a modification in the simply model( changing the joints), the results are right ( as show in the fig. )
[image]
- the problem is still when i add the peg at the arm; instead of use `mj_step` i use `mj_forward`.  or am i using in a wrong way the functions(mj_step, mj_forward) so that can affect the force results?

yuvaltassa, 2023-12-11:
`mj_forward` is fine, doesn't affect the results.

I have a suspicion: do you understand that the **measured values are in the frame of the sensor site** ?
Visualize your site frames in the original model, perhaps that will illuminate.

[image]

closed 2023-12-18 by yuvaltassa