google-deepmind/mujoco #1262: Is it possible for MJX to render in 'human' mode

gautica, 2023-12-06:
Hi,
I would like to ask if MJX offers similar render_mode as in gymnasium, where I can render in 'human' mode. And man can also control simulation with hot keys, for instance, whitespace to pause simulation, c to show the contact force, etc., thank you.

gautica, 2023-12-07:
Another question is, can all environments be rendered during training in a way similar to isaacgym? Thanks

btaba, 2023-12-08:
Hi @gautica have you tried the viewer [here](https://github.com/google-deepmind/mujoco/blob/main/mjx/mujoco/mjx/viewer.py)? This might be what you are looking for, let us know if not

For rendering during training, or saving policy checkpoints, you can use the following callback function 

https://github.com/google/brax/blob/16304037a36b1d9c8c0b3084f57d1159627b636b/brax/training/agents/ppo/train.py#L102

closed 2023-12-08 by btaba