google-deepmind/mujoco #1264: [`MJX`] does `MJX` have its own renderer or does it rely on MuJoCo's renderer

Kallinteris-Andreas, 2023-12-07:
Hi,

I'm a maintainer of [Gymnasium](https://gymnasium.farama.org/) & [Gymnasium-Robotics](https://robotics.farama.org/), and I'm trying to use `MuJoCo-MJX` for "prototyping `MJX`-based RL environments in Gymnasium".

Here is the model tested: [Gymnasium/HalfCheetah](https://github.com/Farama-Foundation/Gymnasium/blob/main/gymnasium/envs/mujoco/assets/half_cheetah.xml) (though it should be not relevant for this question)

The `MJX` [tutorial](https://colab.research.google.com/github/google-deepmind/mujoco/blob/main/mjx/tutorial.ipynb#scrollTo=9f2ME2WbA5Ip) in the section "Define a render utility function", you create a copy of the `mjx_data` into `d` (`type(d)==mujoco.mj_data`) and using it with base `MuJoCo`'s render function to render the `rgb_array`.

```py
def get_image(state: State, camera: str) -> np.ndarray:
  """Renders the environment state."""
  d = mujoco.MjData(env.model)
  # write the mjx.Data into an mjData object
  mjx.device_get_into(d, state.pipeline_state)
  mujoco.mj_forward(env.model, d)
  # use the mjData object to update the renderer
  renderer.update_scene(d, camera=camera)
  return renderer.render()
```

Didn't `brax` have its own render utilities?
Does `MJX` plan to add its own render functions, or it is the plan to keep using `MuJoCo`'s render functions.

Thanks!

btaba, 2023-12-08:
Hi @Kallinteris-Andreas , `brax` had its owner render utilities but they are not compatible with `MJX`. We plan to keep using `MuJoCo` render functions in `MJX`.

If `MJX` is ported back to `brax` (needs a lot of plumbing), we'll likely keep the `brax` HTML rendering there

closed 2023-12-08 by btaba

rs545837, 2024-01-03:
Instead of saving or showing the video with the help of mediap-py library is there also a way to render the model with the help of mujoco viewer while the RL training is going on.

btaba, 2024-01-03:
Hey @rs545837 , [here](https://github.com/google-deepmind/mujoco/blob/main/mjx/mujoco/mjx/viewer.py) is how the MuJoCo viewer is used with MJX. You could try loading a checkpoint and running a policy using that sample code

For checkpointing in Brax see: https://github.com/google/brax/issues/299
For loading a model in Brax see: https://github.com/google/brax/discussions/403

Kallinteris-Andreas, 2024-01-03:
in `gymnasium` you can do `gymnasium.make("Ant-v5", render_mode="human")`

lucattyy, 2024-01-29:
> Hi,
> 
> I'm a maintainer of [Gymnasium](https://gymnasium.farama.org/) & [Gymnasium-Robotics](https://robotics.farama.org/), and I'm trying to use `MuJoCo-MJX` for "prototyping `MJX`-based RL environments in Gymnasium".
> 
> Here is the model tested: [Gymnasium/HalfCheetah](https://github.com/Farama-Foundation/Gymnasium/blob/main/gymnasium/envs/mujoco/assets/half_cheetah.xml) (though it should be not relevant for this question)
> 
> The `MJX` [tutorial](https://colab.research.google.com/github/google-deepmind/mujoco/blob/main/mjx/tutorial.ipynb#scrollTo=9f2ME2WbA5Ip) in the section "Define a render utility function", you create a copy of the `mjx_data` into `d` (`type(d)==mujoco.mj_data`) and using it with base `MuJoCo`'s render function to render the `rgb_array`.
> 
> ```python
> def get_image(state: State, camera: str) -> np.ndarray:
>   """Renders the environment state."""
>   d = mujoco.MjData(env.model)
>   # write the mjx.Data into an mjData object
>   mjx.device_get_into(d, state.pipeline_state)
>   mujoco.mj_forward(env.model, d)
>   # use the mjData object to update the renderer
>   renderer.update_scene(d, camera=camera)
>   return renderer.render()
> ```
> 
> Didn't `brax` have its own render utilities? Does `MJX` plan to add its own render functions, or it is the plan to keep using `MuJoCo`'s render functions.
> 
> Thanks!

Excuse me. Have you completed the development? where can I see the program if it is done.
Thanks!

Kallinteris-Andreas, 2024-01-31:
@lucattyy https://github.com/Farama-Foundation/Gymnasium/issues/833

Andrew-Luo1, 2024-03-10:
If you're doing RL using depth-images, MJX rendering via mjx.ray can be useful; see my response [here](https://github.com/google-deepmind/mujoco/issues/1485). This MJX implementation of a depth-camera can render up to millions of FPS.