google-deepmind/mujoco #1265: Problems with 'viewer.launch_passive' usage

lgpz, 2023-12-07:
Hi,

I'm a student and I'm trying to use MuJoCo for simulation.

I'm looking for some help with viewer.launch_passive.

here is my code:
```
import mujoco
from mujoco import viewer

class MujocoSimulator:
    def __init__(self, model_path, duration=3.8):
        self.model = mujoco.MjModel.from_xml_path(model_path)
        self.data = mujoco.MjData(self.model)
        self.duration = duration
        self.dof = 6
        self.init()

    def run(self):
        mujoco.mj_resetData(self.model, self.data)
        while self.data.time < self.duration:
            self.data.qpos[0] = self.data.qpos[0] + 0.01
            mujoco.mj_step(self.model, self.data)
        print('sim out')

    def init(self):
        for i in range(self.dof):
            self.data.qpos[i] = 0
            self.data.qvel[i] = 0.0
        viewer.launch_passive(self.model, self.data)
        print('inited')

simulator = MujocoSimulator("../assets/universal_robots_ur5e/scene.xml")
simulator.run()
```
and here is outcome:
```
/usr/bin/python3.7 /home/lgpz/Desktop/share/mujoco_attitude_planning-master/tests/my_text02.py 
inited
sim out

Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)
```
Additional Notes:
1. The simulation window flickers
2. Using "viewer.launch" is normal.
3. Running in a Linux virtual machine

lbusellato, 2023-12-07:
You should store the viewer instance:

```
self.viewer = viewer.launch_passive(self.model, self.data) 
```

Then in run() you should sync the simulation and the viewer by calling:

```
self.viewer.sync()
```

lgpz, 2023-12-07:
Thanks! It run. But the error code "Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)" is still there. Can anyone help me to analyse the possible reasons for this, the video of the run is as follows

https://github.com/google-deepmind/mujoco/assets/152043285/66d6f041-bf0e-485b-9934-142ee1f28a57

Why does it fall down at the end, is this normal?

lgpz, 2023-12-07:
Ok,thanks @lbusellato  .

yuvaltassa, 2023-12-07:
Thanks @lbusellato !

If someone could make a consistently reproducible minimal test case with the exit failure that would be appreciated.

Closing for now.