google-deepmind/mujoco #1269: [`MJX`] How to create batched `mjx.Data`?

Kallinteris-Andreas, 2023-12-08:
Hi,

I'm a maintainer of [Gymnasium](https://gymnasium.farama.org/) & [Gymnasium-Robotics](https://robotics.farama.org/), and I'm trying to use `MuJoCo-MJX` for "prototyping `MJX`-based RL environments in Gymnasium".

The documentation for `device_get_into` indicated that `mjx_data` may be batched (aka vectorized)
https://github.com/google-deepmind/mujoco/blob/1db0a9946fca461ed17493a34e03dcbcf8694ab9/mjx/mujoco/mjx/_src/device.py#L233-L249

And the [MJX documentation page](https://mujoco.readthedocs.io/en/3.0.1/mjx.html#structs) says "`mjx.Data` support adding batch dimensions", but `mjx.make_data()` does not have an argument for making batched `mjx.Data`
https://github.com/google-deepmind/mujoco/blob/623201ec76cdfd52d8ceced9d5f81a309ee88af1/mjx/mujoco/mjx/_src/io.py#L131

How can I create batched `mjx.Data`, do I have to use `jax.vmap`?, can you give an example?

Thanks!

yuvaltassa, 2023-12-08:
I believe this is covered in the [colab](https://mujoco.readthedocs.io/en/latest/mjx.html#tutorial-notebook)?

closed 2023-12-08 by yuvaltassa

btaba, 2023-12-08:
Hi @Kallinteris-Andreas , it is indeed in the colab [here](https://colab.sandbox.google.com/github/google-deepmind/mujoco/blob/main/mjx/tutorial.ipynb#scrollTo=65mIPj6DQNNa). An even simpler example would be:

```python
batched_data = jax.vmap(lambda _, x: x, in_axes=(0, None))(jp.arange(100), data)
```