google-deepmind/mujoco #1271: Facing issue with the Equality constraint

rishabkr29, 2023-12-09:
Hi,

I'm a student and I'm trying to use MuJoCo for my thesis. 

I'm looking for some help with equality Constraint. I wanted to connect the rope to the offset site and the cylinder. And ultimately I wanted to rotate the cylinder so that I can actuate the links.

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
<!--<option timestep="0.005" iterations="100" solver="Newton" tolerance="1e-10" gravity = "0 -10 0"/>
<compiler autolimits="true"/>-->
<!--<size nconmax="500" njmax="1500" nkey="1"/>-->

<!--<default>
    <geom solref="1e-2 5" solimp=".5 .99 1e-3" friction=".5"/>
</default>-->

  <visual>
    <rgba haze=".3 .3 .3 1"/>
  </visual>

<asset>
    <texture type="skybox" builtin="gradient" rgb1="0.6 0.6 0.6" rgb2="0 0 0" width="512" height="512"/>

    <texture name="texplane" type="2d" builtin="checker" rgb1=".25 .25 .25" rgb2=".3 .3 .3" width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>

    <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
</asset>
  
  <worldbody>
    <light diffuse=".5 .5 .5" pos="0 0 10" dir="0 0 -1" />
    <geom type="plane" size="0.4 0.4 0.05" rgba="0.8 0.8 0.8 1" material="matplane" />
    <body name="Extrusion">
        <geom type="box" pos="0 0.4 0"  euler="0 0 0" size="0.4 0.05 0.05" rgba="0.7 0.7 0.7 1"/>
        <geom type="box" pos="0 -0.4 0"  euler="0 0 0" size="0.4 0.05 0.05" rgba="0.7 0.7 0.7 1"/>
        <geom type="box" pos="0.4 0 0" euler="0 0 0" size="0.05 0.4 0.05" rgba="0.7 0.7 0.7 1"/>
        <geom type="box" pos="-0.4 0 0" euler="0 0 0" size="0.05 0.4 0.05" rgba="0.7 0.7 0.7 1"/>
    </body>

    <body name="Motor1" pos="-0.4 0.15 0.06">
      <joint name="M1"  type="hinge" axis="0 0 1"  damping="1" />
      <geom type="cylinder" size=".03 .05" rgba=".2 .2 .7 1"/>
      <site name= "M1" pos="0 0 0" size="0.005"  />
    </body>
     <body name="Motor2" pos="-0.4 0.3  0.06">
      <joint name="M2"  type="hinge" axis="0 0 1"  damping="1" />
      <geom type="cylinder" size=".03 .05" rgba=".2 .2 .7 1"/>
      <site name= "M2" pos="0 0 0" size="0.005"  />
    </body>
     <body name="Motor3" pos="-0.4 -0.15 0.06">
      <joint name="M3"  type="hinge" axis="0 0 1"  damping="1" />
      <geom type="cylinder" size=".03 .05" rgba=".2 .2 .7 1"/>
      <site name= "M3" pos="0 0 0" size="0.005"  />
    </body>
     <body name="Motor4" pos="-0.4 -0.3  0.06">
      <joint name="M4"  type="hinge" axis="0 0 1"  damping="1" />
      <geom type="cylinder" size=".03 .05" rgba=".2 .2 .7 1"/>
      <site name= "M4" pos="0 0 0" size="0.005"  />
    </body>
 

    <body name = "Link1" pos="-0.3 0 0.06">
      <geom name="1st Link" type="box" pos="0 0 0"  size=".1 0.02 0.01" />
      <joint name="1st Revolute Joint" pos="-0.1 0 0" type="hinge" axis="0 0 1" limited="true" range="-120 120" damping="0.1" />
      <geom name="1st Revolute Joint" type="cylinder" pos="-0.1 0 0"  size=".05 .01" rgba="0 .9 0 1"  mass="0" group="1" />
      <geom name="offset1" type="box" pos="0 0 0" size=".01 0.05 0.01" />
      <site name="l1" pos="0 0.05 0"  size="0.008" rgba="0.8 .9 0 1" />
      <site name="l2" pos="0 -0.05 0" size="0.008" rgba="0.8 .9 0 1" />  

      <body name="DB0" pos="0 0.05 0" euler="0 0 135">
        <composite type="rope" count="11 1 1" spacing="0.023" offset="0 0 0" prefix="D">
          <joint kind="main" damping="0.01" armature="0.00001"/>
          <geom type="capsule" size=".005 .01" rgba=".8 .2 .1 1"/>
        </composite>
      </body> 
      <body name="CB0" pos="0 -0.05 0" euler="0 0 -135">
        <composite type="rope" count="11 1 1" spacing="0.023"  offset="0 0 0" prefix="C" >
          <joint kind="main" damping="0.01" armature="0.00001"/>
          <geom type="capsule" size=".005 .01" rgba=".8 .2 .1 1"/>
        </composite>
      </body> 

      <body pos="0.2 0 0">
        <geom name="2nd Link" type="box" pos="0 0 0"  size=".1 0.02 .01" rgba="0.9 .9 0 1" />
        <joint name="2nd Revolute Joint" pos="-0.1 0 0" type="hinge" axis="0 0 1" limited="true" range="-120 120" damping="0.1" />
        <geom name="2nd Revolute Joint" type="cylinder" pos="-0.1 0 0"  size=".05 .01" rgba="0 0 0 1"  mass="0" group="1"/>
        <geom name="offset2" type="box" pos="0 0 0" size=".01 0.05 0.01" />
        <site name="l3" pos="0 0.05 0" size="0.008" rgba="0 .9 0 1" />
        <site name="l4" pos="0 -0.05 0" size="0.008" rgba="0 .9 0 1" />  

        <body name="EB0" pos="0 0.05 0" euler="0 0 135">
        <composite type="rope" count="31 1 1"  spacing="0.023" offset="0 0 0" prefix="E">
          <joint kind="main" damping="0.01" armature="0.00001"/>
          <geom type="capsule" size=".005 .01" rgba=".8 .2 .1 1"/>
        </composite>
        </body> 
        <body name="FB0" pos="0 -0.05 0" euler="0 0 -135">
        <composite type="rope" count="31 1 1" spacing="0.023" offset="0 0 0" prefix="F" >
          <joint kind="main" damping="0.01" armature="0.00001"/>
          <geom type="capsule" size=".005 .01" rgba=".8 .2 .1 1"/>
        </composite>
        </body> 
      
      </body> 
    </body>
</worldbody>

<equality>
    <connect name = "rope1" body1="Motor1" body2="DB10" anchor="-0.4 0.15 0.06" />
</equality>

<equality>
    <connect name = "rope2" body1="Motor2" body2="CB10" anchor="-0.4 0.3  0.06"  />
</equality>

<equality>
    <connect name = "rope3" body1="Motor3" body2="EB30" anchor="-0.4 -0.15 0.06"   />
</equality>

<equality>
    <connect name = "rope4" body1="Motor4" body2="FB30" anchor="-0.4 0.15 0.06"   />
</equality>

<actuator >
    <motor name="Motor1" joint ="M1"  />
</actuator>
<actuator >
    <motor name="Motor2" joint ="M2"  />
</actuator>
<actuator >
    <motor name="Motor3" joint ="M3"  />
</actuator>
<actuator >
    <motor name="Motor4" joint ="M4" />
</actuator>

</mujoco>
```

</details>

Here is a screenshot / video, illustrating my question:

https://github.com/google-deepmind/mujoco/assets/110694658/9d8bb6e8-1d75-4b23-b9b2-21685b56d656

HoangGiang93, 2023-12-09:
Please use the current MuJoCo version 3.0.1 and update the model. MuJoCo droped this composite one year ago.
https://github.com/google-deepmind/mujoco/commit/3de2873883aa467738f84b8214fd6bc5730c7526#diff-2c576e4fc746579964f3f1ef41621ea94a4496df1686ff48f3bde77595850341

rishabkr29, 2023-12-10:
> Please use the current MuJoCo version 3.0.1 and update the model. MuJoCo droped this composite type one year ago. [3de2873#diff-2c576e4fc746579964f3f1ef41621ea94a4496df1686ff48f3bde77595850341](https://github.com/google-deepmind/mujoco/commit/3de2873883aa467738f84b8214fd6bc5730c7526#diff-2c576e4fc746579964f3f1ef41621ea94a4496df1686ff48f3bde775958503

```XML
<mujoco>
  <option timestep="0.001" jacobian="dense"/>
  <size nconmax="500" njmax="1500" nkey="1"/>
  
  <extension>
    <plugin plugin="mujoco.elasticity.cable"/>
  </extension>

  <default>
    <geom solref="1e-2 5" solimp=".5 .99 1e-3" friction=".5"/>
  </default>

  <visual>
    <rgba haze=".3 .3 .3 1"/>
  </visual>

<asset>
    <texture type="skybox" builtin="gradient" rgb1="0.6 0.6 0.6" rgb2="0 0 0" width="512" height="512"/>

    <texture name="texplane" type="2d" builtin="checker" rgb1=".25 .25 .25" rgb2=".3 .3 .3" width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>

    <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
</asset>
  
  <worldbody>
    <light diffuse=".5 .5 .5" pos="0 0 10" dir="0 0 -1" />
    <geom type="plane" size="0.4 0.4 0.05" rgba="0.8 0.8 0.8 1" material="matplane" />
    <body name="Extrusion">
        <geom type="box" pos="0 0.4 0"  euler="0 0 0" size="0.4 0.05 0.05" rgba="0.7 0.7 0.7 1"/>
        <geom type="box" pos="0 -0.4 0"  euler="0 0 0" size="0.4 0.05 0.05" rgba="0.7 0.7 0.7 1"/>
        <geom type="box" pos="0.4 0 0" euler="0 0 0" size="0.05 0.4 0.05" rgba="0.7 0.7 0.7 1"/>
        <geom type="box" pos="-0.4 0 0" euler="0 0 0" size="0.05 0.4 0.05" rgba="0.7 0.7 0.7 1"/>
    </body>

    <body name="Motor1" pos="-0.4 0.15 0.06">
      <joint name="M1"  type="hinge" axis="0 0 1"  damping="1" />
      <geom type="cylinder" size=".03 .05" rgba=".2 .2 .7 1"/>
      <site name= "M1" pos="0 0 0" size="0.005"  />
    </body>
     <body name="Motor2" pos="-0.4 0.3  0.06">
      <joint name="M2"  type="hinge" axis="0 0 1"  damping="1" />
      <geom type="cylinder" size=".03 .05" rgba=".2 .2 .7 1"/>
      <site name= "M2" pos="0 0 0" size="0.005"  />
    </body>
     <body name="Motor3" pos="-0.4 -0.15 0.06">
      <joint name="M3"  type="hinge" axis="0 0 1"  damping="1" />
      <geom type="cylinder" size=".03 .05" rgba=".2 .2 .7 1"/>
      <site name= "M3" pos="0 0 0" size="0.005"  />
    </body>
     <body name="Motor4" pos="-0.4 -0.3  0.06">
      <joint name="M4"  type="hinge" axis="0 0 1"  damping="1" />
      <geom type="cylinder" size=".03 .05" rgba=".2 .2 .7 1"/>
      <site name= "M4" pos="0 0 0" size="0.005"  />
    </body>
 

    <body name = "Link1" pos="-0.3 0 0.06">
      <geom name="1st Link" type="box" pos="0 0 0"  size=".1 0.02 0.01" />
      <joint name="1st Revolute Joint" pos="-0.1 0 0" type="hinge" axis="0 0 1" limited="true" range="-120 120" damping="0.1" />
      <geom name="1st Revolute Joint" type="cylinder" pos="-0.1 0 0"  size=".05 .01" rgba="0 .9 0 1"  mass="0" group="1" />
      <geom name="offset1" type="box" pos="0 0 0" size=".01 0.05 0.01" />
      <site name="l1" pos="0 0.05 0"  size="0.008" rgba="0.8 .9 0 1" />
      <site name="l2" pos="0 -0.05 0" size="0.008" rgba="0.8 .9 0 1" />  

      <body name="DB0" pos="0 0.05 0" euler="0 0 135">
        <composite type="cable" count="5 1 1" curve="s 0 0" offset="0 0 0" prefix="D">
          <joint kind="main" damping="0.01" armature="0.00001"/>
          <geom type="capsule" size=".005 .01" rgba=".8 .2 .1 1"/>
        </composite>
      </body> 
      <body name="CB0" pos="0 -0.05 0" euler="0 0 -135">
        <composite type="cable" count="5 1 1" curve="s 0 0"  offset="0 0 0" prefix="C" >
          <joint kind="main" damping="0.01" armature="0.00001"/>
          <geom type="capsule" size=".005 .01" rgba=".8 .2 .1 1"/>
        </composite>
      </body> 

      <body pos="0.2 0 0">
        <geom name="2nd Link" type="box" pos="0 0 0"  size=".1 0.02 .01" rgba="0.9 .9 0 1" />
        <joint name="2nd Revolute Joint" pos="-0.1 0 0" type="hinge" axis="0 0 1" limited="true" range="-120 120" damping="0.1" />
        <geom name="2nd Revolute Joint" type="cylinder" pos="-0.1 0 0"  size=".05 .01" rgba="0 0 0 1"  mass="0" group="1"/>
        <geom name="offset2" type="box" pos="0 0 0" size=".01 0.05 0.01" />
        <site name="l3" pos="0 0.05 0" size="0.008" rgba="0 .9 0 1" />
        <site name="l4" pos="0 -0.05 0" size="0.008" rgba="0 .9 0 1" />  

        <body name="EB0" pos="0 0.05 0" euler="0 0 135">
        <composite type="cable" count="5 1 1"  curve="s 0 0" offset="0 0 0" prefix="E">
          <joint kind="main" damping="0.01" armature="0.00001"/>
          <geom type="capsule" size=".005 .01" rgba=".8 .2 .1 1"/>
        </composite>
        </body> 
        <body name="FB0" pos="0 -0.05 0" euler="0 0 -135">
        <composite type="cable" count="11 1 1" curve="s 0 0"  offset="0 0 0" prefix="F" >
          <joint kind="main" damping="0.01" armature="0.00001"/>
          <geom type="capsule" size=".005 .01" rgba=".8 .2 .1 1"/>
        </composite>
        </body> 
      
      </body> 
    </body>
</worldbody>

<equality>
    <connect name = "cable1" body1="Motor1" body2="DB_last" anchor="-0.4 0.2 0.06" />
</equality>

<equality>
    <connect name = "cable2" body1="Motor2" body2="CB_last" anchor="-0.4 0.3  0.06" />
</equality>

<equality>
    <connect name = "cable3" body1="Motor3" body2="EB_last" anchor="-0.4 -0.2 0.06" />
</equality>

<equality>
    <connect name = "cable4" body1="Motor4" body2="FB_last" anchor="-0.4 0.2 0.06" />
</equality>

<actuator >
    <motor name="Motor1" joint ="M1"  />
</actuator>
<actuator >
    <motor name="Motor2" joint ="M2"  />
</actuator>
<actuator >
    <motor name="Motor3" joint ="M3"  />
</actuator>
<actuator >
    <motor name="Motor4" joint ="M4" />
</actuator>

</mujoco>
```

Uploading MuJoCo _ MuJoCo Model 2023-12-10 10-08-37.mp4…

rishabkr29, 2023-12-10:
> > Please use the current MuJoCo version 3.0.1 and update the model. MuJoCo droped this composite type one year ago. [[3de2873](https://github.com/google-deepmind/mujoco/commit/3de2873883aa467738f84b8214fd6bc5730c7526)#diff-2c576e4fc746579964f3f1ef41621ea94a4496df1686ff48f3bde77595850341]([3de2873#diff-2c576e4fc746579964f3f1ef41621ea94a4496df1686ff48f3bde775958503](https://github.com/google-deepmind/mujoco/commit/3de2873883aa467738f84b8214fd6bc5730c7526#diff-2c576e4fc746579964f3f1ef41621ea94a4496df1686ff48f3bde775958503)
> 
> Updated the model still problem persists.
> 
> ```
> <mujoco>
>   <option timestep="0.001" jacobian="dense"/>
>   <size nconmax="500" njmax="1500" nkey="1"/>
>   
>   <extension>
>     <plugin plugin="mujoco.elasticity.cable"/>
>   </extension>
> 
>   <default>
>     <geom solref="1e-2 5" solimp=".5 .99 1e-3" friction=".5"/>
>   </default>
> 
>   <visual>
>     <rgba haze=".3 .3 .3 1"/>
>   </visual>
> 
> <asset>
>     <texture type="skybox" builtin="gradient" rgb1="0.6 0.6 0.6" rgb2="0 0 0" width="512" height="512"/>
> 
>     <texture name="texplane" type="2d" builtin="checker" rgb1=".25 .25 .25" rgb2=".3 .3 .3" width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>
> 
>     <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
> </asset>
>   
>   <worldbody>
>     <light diffuse=".5 .5 .5" pos="0 0 10" dir="0 0 -1" />
>     <geom type="plane" size="0.4 0.4 0.05" rgba="0.8 0.8 0.8 1" material="matplane" />
>     <body name="Extrusion">
>         <geom type="box" pos="0 0.4 0"  euler="0 0 0" size="0.4 0.05 0.05" rgba="0.7 0.7 0.7 1"/>
>         <geom type="box" pos="0 -0.4 0"  euler="0 0 0" size="0.4 0.05 0.05" rgba="0.7 0.7 0.7 1"/>
>         <geom type="box" pos="0.4 0 0" euler="0 0 0" size="0.05 0.4 0.05" rgba="0.7 0.7 0.7 1"/>
>         <geom type="box" pos="-0.4 0 0" euler="0 0 0" size="0.05 0.4 0.05" rgba="0.7 0.7 0.7 1"/>
>     </body>
> 
>     <body name="Motor1" pos="-0.4 0.15 0.06">
>       <joint name="M1"  type="hinge" axis="0 0 1"  damping="1" />
>       <geom type="cylinder" size=".03 .05" rgba=".2 .2 .7 1"/>
>       <site name= "M1" pos="0 0 0" size="0.005"  />
>     </body>
>      <body name="Motor2" pos="-0.4 0.3  0.06">
>       <joint name="M2"  type="hinge" axis="0 0 1"  damping="1" />
>       <geom type="cylinder" size=".03 .05" rgba=".2 .2 .7 1"/>
>       <site name= "M2" pos="0 0 0" size="0.005"  />
>     </body>
>      <body name="Motor3" pos="-0.4 -0.15 0.06">
>       <joint name="M3"  type="hinge" axis="0 0 1"  damping="1" />
>       <geom type="cylinder" size=".03 .05" rgba=".2 .2 .7 1"/>
>       <site name= "M3" pos="0 0 0" size="0.005"  />
>     </body>
>      <body name="Motor4" pos="-0.4 -0.3  0.06">
>       <joint name="M4"  type="hinge" axis="0 0 1"  damping="1" />
>       <geom type="cylinder" size=".03 .05" rgba=".2 .2 .7 1"/>
>       <site name= "M4" pos="0 0 0" size="0.005"  />
>     </body>
>  
> 
>     <body name = "Link1" pos="-0.3 0 0.06">
>       <geom name="1st Link" type="box" pos="0 0 0"  size=".1 0.02 0.01" />
>       <joint name="1st Revolute Joint" pos="-0.1 0 0" type="hinge" axis="0 0 1" limited="true" range="-120 120" damping="0.1" />
>       <geom name="1st Revolute Joint" type="cylinder" pos="-0.1 0 0"  size=".05 .01" rgba="0 .9 0 1"  mass="0" group="1" />
>       <geom name="offset1" type="box" pos="0 0 0" size=".01 0.05 0.01" />
>       <site name="l1" pos="0 0.05 0"  size="0.008" rgba="0.8 .9 0 1" />
>       <site name="l2" pos="0 -0.05 0" size="0.008" rgba="0.8 .9 0 1" />  
> 
>       <body name="DB0" pos="0 0.05 0" euler="0 0 135">
[... 48 lines cut ...]
> </equality>
> 
> <equality>
>     <connect name = "cable4" body1="Motor4" body2="FB_last" anchor="-0.4 0.2 0.06" />
> </equality>
> 
> 
> <actuator >
>     <motor name="Motor1" joint ="M1"  />
> </actuator>
> <actuator >
>     <motor name="Motor2" joint ="M2"  />
> </actuator>
> <actuator >
>     <motor name="Motor3" joint ="M3"  />
> </actuator>
> <actuator >
>     <motor name="Motor4" joint ="M4" />
> </actuator>
> 
> 
> </mujoco>
> ```

https://github.com/google-deepmind/mujoco/assets/110694658/ffe48968-7224-4b09-9caf-e69f922bceed

rishabkr29, 2023-12-10:
@quagla I am not able to modify the cable length, as spacing is not being supported in the cable type.

Balint-H, 2023-12-11:
Try the `size` attribute in the composite element. The second element of the inner geom's size is ignored, the length is determined by the composite element's (1D) size.

Balint-H, 2023-12-11:
I see, in case you haven't yet, check out the "belt" and "cable" examples in the model folder of the repo, both have examples on controlling composite geometry.

rishabkr29, 2023-12-11:
Yes, I have seen it.  Right now my concern is why my equality constraint of  two bodies are not being holding ? and what shall I make it possible in such a way so that when I give the rotation to motor the cables get wrap or unwarp.

HoangGiang93, 2023-12-11:
I fixed your model, in your last model the motors cant rotate because of self-collision. I think you need to reconsider using different approach than puting connect equality between the motor body and the tendon last. I understand what you want to archieve but that shouldn't be how it works
```
<mujoco>
  <option timestep="0.001" jacobian="dense"/>
  <size nconmax="500" njmax="1500" nkey="1" />

  <extension>
    <plugin plugin="mujoco.elasticity.cable" />
  </extension>

  <default>
    <geom solref="1e-2 5" solimp=".5 .99 1e-3" friction=".5" />
  </default>

  <visual>
    <rgba haze=".3 .3 .3 1" />
  </visual>

  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.6 0.6 0.6" rgb2="0 0 0" width="512"
      height="512" />

    <texture name="texplane" type="2d" builtin="checker" rgb1=".25 .25 .25" rgb2=".3 .3 .3"
      width="512" height="512" mark="cross" markrgb=".8 .8 .8" />

    <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true" />
  </asset>

  <worldbody>
    <light diffuse=".5 .5 .5" pos="0 0 10" dir="0 0 -1" />
    <geom type="plane" size="0.4 0.4 0.05" rgba="0.8 0.8 0.8 1" material="matplane" />
    <body name="Extrusion">
      <geom type="box" pos="0 0.4 0" euler="0 0 0" size="0.4 0.05 0.05" rgba="0.7 0.7 0.7 1" />
      <geom type="box" pos="0 -0.4 0" euler="0 0 0" size="0.4 0.05 0.05" rgba="0.7 0.7 0.7 1" />
      <geom type="box" pos="0.4 0 0" euler="0 0 0" size="0.05 0.4 0.05" rgba="0.7 0.7 0.7 1" />
      <geom type="box" pos="-0.4 0 0" euler="0 0 0" size="0.05 0.4 0.05" rgba="0.7 0.7 0.7 1" />
    </body>

    <body name="Motor1" pos="-0.4 0.15 0.08">
      <joint name="M1" type="hinge" axis="0 0 1" damping="1" />
      <geom type="cylinder" size=".03 .03" rgba=".2 .2 .7 1" />
      <site name="M1" pos="0 0 0" size="0.005" />
    </body>
    <body name="Motor2" pos="-0.4 0.3 0.08">
      <joint name="M2" type="hinge" axis="0 0 1" damping="1" />
      <geom type="cylinder" size=".03 .03" rgba=".2 .2 .7 1" />
      <site name="M2" pos="0 0 0" size="0.005" />
    </body>
    <body name="Motor3" pos="-0.4 -0.15 0.08">
      <joint name="M3" type="hinge" axis="0 0 1" damping="1" />
      <geom type="cylinder" size=".03 .03" rgba=".2 .2 .7 1" />
      <site name="M3" pos="0 0 0" size="0.005" />
    </body>
    <body name="Motor4" pos="-0.4 -0.3 0.08">
      <joint name="M4" type="hinge" axis="0 0 1" damping="1" />
      <geom type="cylinder" size=".03 .03" rgba=".2 .2 .7 1" />
      <site name="M4" pos="0 0 0" size="0.005" />
    </body>

    <body name="Link1" pos="-0.3 0 0.06">
      <geom name="1st Link" type="box" pos="0 0 0" size=".1 0.02 0.01" />
      <joint name="1st Revolute Joint" pos="-0.1 0 0" type="hinge" axis="0 0 1" limited="true"
        range="-120 120" damping="0.1" />
      <geom name="1st Revolute Joint" type="cylinder" pos="-0.1 0 0" size=".05 .01" rgba="0 .9 0 1"
        mass="0" group="1" />
      <geom name="offset1" type="box" pos="0 0 0" size=".01 0.05 0.01" />
      <site name="l1" pos="0 0.05 0" size="0.008" rgba="0.8 .9 0 1" />
      <site name="l2" pos="0 -0.05 0" size="0.008" rgba="0.8 .9 0 1" />

      <body name="DB0" pos="0 0.05 0" euler="0 0 135">
        <composite type="cable" count="10 1 1" curve="s 0 0" offset="0 0 0" prefix="D">
          <joint kind="main" damping="0.01" armature="0.00001" />
          <geom type="capsule" size=".005 .01" rgba=".8 .2 .1 1" />
        </composite>
      </body>
      <body name="CB0" pos="0 -0.05 0" euler="0 0 -135">
        <composite type="cable" count="10 1 1" curve="s 0 0" offset="0 0 0" prefix="C">
          <joint kind="main" damping="0.01" armature="0.00001" />
          <geom type="capsule" size=".005 .01" rgba=".8 .2 .1 1" />
        </composite>
      </body>

      <body pos="0.2 0 0">
        <geom name="2nd Link" type="box" pos="0 0 0" size=".1 0.02 .01" rgba="0.9 .9 0 1" />
        <joint name="2nd Revolute Joint" pos="-0.1 0 0" type="hinge" axis="0 0 1" limited="true"
          range="-120 120" damping="0.1" />
        <geom name="2nd Revolute Joint" type="cylinder" pos="-0.1 0 0" size=".05 .01" rgba="0 0 0 1"
          mass="0" group="1" />
        <geom name="offset2" type="box" pos="0 0 0" size=".01 0.05 0.01" />
        <site name="l3" pos="0 0.05 0" size="0.008" rgba="0 .9 0 1" />
        <site name="l4" pos="0 -0.05 0" size="0.008" rgba="0 .9 0 1" />

        <body name="EB0" pos="0 0.05 0" euler="0 0 135">
          <composite type="cable" count="10 1 1" curve="s 0 0" offset="0 0 0" prefix="E">
            <joint kind="main" damping="0.01" armature="0.00001" />
            <geom type="capsule" size=".005 .01" rgba=".8 .2 .1 1" />
          </composite>
        </body>
        <body name="FB0" pos="0 -0.05 0" euler="0 0 -135">
          <composite type="cable" count="10 1 1" curve="s 0 0" offset="0 0 0" prefix="F">
            <joint kind="main" damping="0.01" armature="0.00001" />
            <geom type="capsule" size=".005 .01" rgba=".8 .2 .1 1" />
          </composite>
        </body>

      </body>
    </body>
  </worldbody>

  <!-- <equality>
    <connect name = "cable1" body1="Motor1" body2="DB_last" anchor="0 0 0" />
    <connect name = "cable2" body1="Motor2" body2="CB_last" anchor="0 0 0 " />
    <connect name = "cable3" body1="Motor3" body2="EB_last" anchor="0 0  0 " />
    <connect name = "cable4" body1="Motor4" body2="FB_last" anchor="0 0 0" />
</equality> -->

  <actuator>
    <motor name="Motor1" joint="M1" />
    <motor name="Motor2" joint="M2" />
    <motor name="Motor3" joint="M3" />
    <motor name="Motor4" joint="M4" />
  </actuator>

</mujoco>
```

rishabkr29, 2023-12-11:
Thanks, @HoangGiang93 but I am not using tendon in my case instead I am using composite of type cable.

rishabkr29, 2023-12-14:
> Try the `size` attribute in the composite element. The second element of the inner geom's size is ignored, the length is determined by the composite element's size (size[0] specifically). Also, in case the wrapping is not essential in your model and you only need lengthening and shortening, have you considered using tendons?

Thanks😊 , I found some alternative to actuate the cable length.

kevinzakka, 2024-01-03:
@rishabkr29 It sounds like you've solved your issue. Going to close, feel free to re-open if you encounter further issues.

closed 2024-01-03 by kevinzakka