google-deepmind/mujoco #1273: How to debug efficiently in MJX

damianliumin, 2023-12-12:
Hi,

I'm a student and I'm trying to use MuJoCo (MJX) for a research project.

I'm wondering how to debug efficiently in MJX. 
In the code below, the first two iterations take about 80 seconds to run respectively, while the later iterations takes less than 0.08 seconds. 
I do not understand why the first two iterations take so long. I thought it was related to JIT, but after disabling JIT by setting the environment variable JAX_DISABLE_JIT=1, it takes even longer (over 300s).

I think whether there is anything wrong with the code. Besides, is there any method to debug efficiently in MJX. If the initial runs are always so slow, it would be very inconvenient to develop new environments with MJX.

```python
model = mujoco.MjModel.from_xml_path("assets/robotiq_2f85/scene.xml")
data = mujoco.MjData(model)
renderer = mujoco.Renderer(model)

mjx_model = mjx.device_put(model)
mjx_data = mjx.device_put(data)
forward_jit = jax.jit(mjx.forward)

for i in range(10):
  tik = time.time()
  mjx_data = forward_jit(mjx_model, mjx_data)
  tok = time.time()
  print(i, tok - tik)
```
The asset can be found in https://github.com/google-deepmind/mujoco_menagerie/tree/main/robotiq_2f85. Actually, the running time varies with the complexity of assets.

closed 2023-12-23 by damianliumin

panwliu, 2024-01-09:
Hi @damianliumin , did you get a solution? I got the same issue but couldn't figure out why.

damianliumin, 2024-01-09:
> Hi @damianliumin , did you get a solution? I got the same issue but couldn't figure out why.

Sorry, I don't have a solution yet. I have decided not to use MJX in my project.

erikfrey, 2024-01-09:
Hi @panwliu and @damianliumin, a few pointers for debugging:

- As of MuJoCo 3.1 we have a better API for placing data on device, `mjx.put_model` and `mjx.put_data`, please switch to those.  The old functions did not preserve array shapes correctly, so you are probably seeing double JITs in that for-loop when you call mjx.step.  Switching to the new functions should make it so that you pay the JIT penalty just once.

- If you want a simple way of inspecting behavior in MJX, check out the supplied [viewer.py](https://github.com/google-deepmind/mujoco/blob/main/mjx/mujoco/mjx/viewer.py) - it's not nearly as fully featured as the native MuJoCo simulate, but you can at least eyeball behavior quickly and move ctrl values and so on.

- In general our philosophy for MJX is that you should be able to do most of your model development in native MuJoCo, and then when you're ready to hook it up to RL just switch to MJX and it should just work - if that is not the case, please let us know!  Tools like [jax.debug.print](https://jax.readthedocs.io/en/latest/debugging/print_breakpoint.html) are quite handy and work well on GPU if you are seeing an issue.  However, this shouldn't be necessary, if you have a simulation that's working correctly in MuJoCo and not in MJX, that's a bug - we'll look into it.

- In our roadmap are some tools for AOT'ing JIT / caching JIT output to disk, so that you can rapidly develop in MJX without waiting for JIT every time you restart your process... no ETA on that work though, but that will be a big help for sure.

- For your particular model, I see a lot of meshes.  While MJX supports convex<>convex, they are not ideal performance wise.  You would see lower JIT times and much faster training performance if you switched to primitive shapes.  Just looking at the shape of the robotiq grippers, they seem ideally shaped for primitives.  If you would like an example of a menagerie model that has been optimized for MJX, see the [Barkour](https://github.com/google-deepmind/mujoco_menagerie/blob/main/google_barkour_vb/barkour_vb_mjx.xml) robot and accompanying [training tutorial](https://colab.research.google.com/github/google-deepmind/mujoco/blob/main/mjx/tutorial.ipynb).

panwliu, 2024-01-09:
Thanks @erikfrey ! I'd suggest putting the "philosophy for MJX" into MJX document so that developers are clear about the overall development flow. It could be very helpful.