google-deepmind/mujoco #1274: Loading xml into mujoco viewer leads to crash with the following ERROR: Auxiliary buffer size exceeds maximum allowed by OpenGL implementation

brudermueller, 2023-12-12:
Hi,

I'm a  PhD student  and I'm trying to use MuJoCo for contact-rich robot manipulation
I have just upgraded my Mujoco version to the newest version, as I want to use newer features, e.g. in the passive launch of the viewer and in modifying the scene.  
With mujoco_210, the viewer was working perfectly. Now, each time I run `python3 -m mujoco.viewer`, the viewer launches, but the moment I drag an xml file into the viewer (for any xml-file, even very basic ones), such as the one pasted below, I get the following error in my terminal: 
`ERROR: Auxiliary buffer size exceeds maximum allowed by OpenGL implementation

Press Enter to exit ...
`
I was trying to find out any driver requirements for the mujoco versions, but couldn't find any information. 
I would appreciate any kind of leads on how to solve that problem. 
Many thanks in advance! 

- Mujoco version: mujoco 3.0.1
- Python version: Python 3.10.13
- NVIDIA-SMI 535.129.03 (driver version)
- Cuda version: 11.7

<mujoco model="mug">
<visual>
<headlight ambient="0 0 0" diffuse="0 0 0" specular="1 1 1"/>
<quality shadowsize="8192"/>
</visual>
<asset>
<texture name="mug" file="mug.png" type="2d"/>
<material name="mug" texture="mug" specular="1" shininess="1"/>
<mesh file="mug.obj" scale=".01 .01 .01"/>
<material name="floor" reflectance=".1"/>
</asset>
<default>
<default class="collision_mug">
<geom type="box" group="3"/>
<default class="cup">
<geom size="0.0026 0.00704 0.033"/>
</default>
<default class="handle">
<geom size="0.0025 0.00328 0.0055"/>
</default>
</default>
</default>
<worldbody>
<light pos=".4 -.4 .3" dir="-2 2 -1.5" diffuse=".6 .6 .6"/>
<light pos="-.2 -.4 .3" dir="1 2 -1.5" diffuse=".6 .6 .6"/>
<geom type="plane" size="5 5 .01" rgba="1 1 1 1" material="floor"/>
<body name="mug">
<freejoint/>
<geom type="mesh" mesh="mug" material="mug" euler="90 0 0" group="1" contype="0" conaffinity="0"/>
<geom class="cup" pos="0.0418 0 0.038" euler="0 0 0"/>
<geom class="cup" pos="0.0398 0.0129 0.038" euler="0 0 18"/>
<geom class="cup" pos="0.0339 0.0246 0.038" euler="0 0 36"/>
<geom class="cup" pos="0.0246 0.0339 0.038" euler="0 0 54"/>
<geom class="cup" pos="0.0129 0.0398 0.038" euler="0 0 72"/>
<geom class="cup" pos="0 0.0418 0.038" euler="0 0 90"/>
<geom class="cup" pos="-0.0129 0.0398 0.038" euler="0 0 108"/>
<geom class="cup" pos="-0.0246 0.0339 0.038" euler="0 0 126"/>
<geom class="cup" pos="-0.0339 0.0246 0.038" euler="0 0 144"/>
<geom class="cup" pos="-0.0398 0.0129 0.038" euler="0 0 162"/>
<geom class="cup" pos="-0.0418 0 0.038" euler="0 0 180"/>
<geom class="cup" pos="-0.0398 -0.0129 0.038" euler="0 0 198"/>
<geom class="cup" pos="-0.0339 -0.0246 0.038" euler="0 0 216"/>
<geom class="cup" pos="-0.0246 -0.0339 0.038" euler="0 0 234"/>
<geom class="cup" pos="-0.0129 -0.0398 0.038" euler="0 0 252"/>
<geom class="cup" pos="0 -0.0418 0.038" euler="0 0 270"/>
<geom class="cup" pos="0.0129 -0.0398 0.038" euler="0 0 288"/>
<geom class="cup" pos="0.0246 -0.0339 0.038" euler="0 0 306"/>
<geom class="cup" pos="0.0339 -0.0246 0.038" euler="0 0 324"/>
<geom class="cup" pos="0.0398 -0.0129 0.038" euler="0 0 342"/>
<geom class="cup" type="cylinder" size="0.045 0.0026" pos="0 0 0.0026"/>
<body name="handle" pos="0.056 0 0.0395" euler="90 0 0">
<geom class="handle" pos="0.0193 0 0" euler="0 0 0"/>
<geom class="handle" pos="0.0184 0.00568 0" euler="0 0 17.1"/>
<geom class="handle" pos="0.0159 0.0108 0" euler="0 0 34.3"/>
<geom class="handle" pos="0.012 0.0151 0" euler="0 0 51.4"/>
<geom class="handle" pos="0.00703 0.0179 0" euler="0 0 68.6"/>
<geom class="handle" pos="0.00144 0.0192 0" euler="0 0 85.7"/>
<geom class="handle" pos="0.00144 -0.0192 0" euler="0 0 274"/>
<geom class="handle" pos="0.00703 -0.0179 0" euler="0 0 291"/>
<geom class="handle" pos="0.012 -0.0151 0" euler="0 0 309"/>
<geom class="handle" pos="0.0159 -0.0108 0" euler="0 0 326"/>
<geom class="handle" pos="0.0184 -0.00568 0" euler="0 0 343"/>
<geom class="collision_mug" size="0.007 0.0025 0.0055" pos="-0.007 -0.0195 0"/>
<geom class="collision_mug" size="0.007 0.0025 0.0055" pos="-0.007 0.0195 0"/>
</body>
</body>
</worldbody>
</mujoco>

brudermueller, 2023-12-12:
Additional note: this is only the case for the newest mujoco release. That error does not occur under Mujoco 3.0.0.

nimrod-gileadi, 2023-12-12:
What operating system are you on?

brudermueller, 2023-12-12:
Ubuntu 20.04

nimrod-gileadi, 2023-12-13:
I don't have a solution for you, and I was unable to reproduce this bug.
However, we just pushed version 3.1.0 of the mujoco package. Could you upgrade and try again? We should have a better error message to debug the issue.

brudermueller, 2023-12-14:
I guess it's some weird versioning issue.... The issue persists under 3.1.0, while everything is working fine under 3.0.0. 
The error message is now the following: 
`ERROR: Auxiliary buffer height exceeds maximum allowed by OpenGL implementation: 16566 > 16384`

yuvaltassa, 2024-01-02:
Not clear how to help you without being able to reproduce. 

It looks like your system is trying to open a huge window for some reason. Does reducing [offsamples](https://mujoco.readthedocs.io/en/latest/XMLreference.html#visual-quality-offsamples) help?

Closing due to inability to reproduce.

vikashplus, 2024-01-06:
I just faced the same issue `Auxiliary buffer size exceeds maximum allowed by OpenGL implementation` 
I have a (non-public) model where everything works in `mujoco-3.0.0` and fails with the above error after `mujoco>=3.0.1`

Jendker, 2024-01-17:
I can confirm the issue @vikashplus encounters with Mujoco version>=3.0.1:
```
ERROR: Auxiliary buffer height exceeds maximum allowed by OpenGL implementation: 21642 > 16384
```
I am running the code on Intel MBP 16 GB RAM what would explain the buffer height limit of 16384. Did Mujoco's memory footprint change with version 3.0.1?

vikashplus, 2024-01-20:
@yuvaltassa @nimrod-gileadi -- There is more evidence coming in on this issue. Will it be okay to re-open this issue?

yuvaltassa, 2024-01-22:
Thanks @brudermueller! 

@saran-t will be back in February and hopefully be able to diagnose/fix this.

combinatorylogic, 2024-01-27:
I'm able to consistently reproduce the bug with Mesa softpipe (to exclude the differences between opengl implementations).

combinatorylogic, 2024-01-27:
The largest chunks contributing to the aux buffer height: Physics (4314), Visualisation (4212), Rendering (2646) and Group Enable (2646). Adding up to 16632.

saran-t, 2024-02-02:
Something happened between 3.0.0 and 3.0.1 that caused the calculated max height for the left UI pane to increase. Given that none of our team is able to reproduce the issue locally, it'll be really helpful if someone who's affected can help us bisect to see which commit between 3.0.0 and 3.0.1 this issue first appeared.

Separately, the way that we're currently drawing the UI panes are a bit silly. We make a draw buffer that's as tall as the UI pane can possibly be if it's fully expanded, then draw everything, then clip to viewport. This is inefficient both in compute and memory usage. A better approach would be to do a check on each UI element and just don't draw if it's completely outside the bounds.

yuvaltassa, 2024-02-02:
We've figured this out. It has to do with an unfortunate design choice of the UI framework (described above by @saran-t) combined with the limitations of some GL backends, combined with high-resolution screens. We might figure out a better way to solve this but in the meantime we'll just reduce the maximum default font size by one click.

@vikashplus, if you avoid opening both the Joint and Actuator tabs in the rightUI you should avoid a crash.

Please try building from source and let us know if this "fix" works for you.

vikashplus, 2024-02-03:
I gave it a quick spin. I was also able to make it reproducible on a public model 
> ./simulate https://github.com/MyoHub/myo_sim/blob/main/leg/myolegs.xml

What I observe for this model
- loads fine under the default setting
- It crashes with following error when I boost the font to 300% 
```
MuJoCo version 3.1.2
ERROR: Auxiliary buffer height exceeds maximum allowed by OpenGL implementation: 16656 > 16384
```

yuvaltassa, 2024-02-03:
Ya this is why the commit said "workaround" rather than "fixed"... 😜

Please avoid 300% for now.

vikashplus, 2024-02-17:
@yuvaltassa  -- I have a large model that crashes on load (default font to 200%). Any workaround that I can use?

hello-binit, 2024-03-26:
Another workaround that doesn't require rebuilding is launching the viewer with `python -m mujoco.viewer`, reducing the font size to 50% or 100%, and then dragging the model file into the viewer.

Looking forward to a proper fix for this :slightly_smiling_face:

yuvaltassa, 2024-06-24:
Hi folks, this is now fixed at head. 
Please build from source and let us know if there is any remaining issue...