google-deepmind/mujoco #1276: contact force problem

sunnymints, 2023-12-13:
**How to ask for help**

First, read our quick guide to
[asking good questions](https://github.com/google-deepmind/mujoco#asking-questions).
Below is a template for you to use:

Hi,

I'm a student and I'm trying to use MuJoCo for _robot drl____.

I'm looking for some help with contact force.

Here is a model which explains my question:
I'm very sorry to post again to ask for help and occupy everyone's resources. I have previously posted in this issuse (https://github.com/google-deepmind/mujoco/issues/1249) the problem of inconsistent directions of the peg block surface contact force.

<details>
  <summary>minimal XML</summary>

```XML
<mujoco model="ur5">
  <option gravity="0 0 0"/>
  <compiler angle="radian" meshdir="./ur_description/meshes/ur5/collision/"/>
  <statistic meansize="0.2" extent="0.4" center="-3.42017e-10 -0.0501512 -1.90133e-10"/>
  <asset>
    <mesh name="fk" file="fk.STL"/>
  </asset>
  <statistic meansize="0.161957" extent="1.0432" center="0.388318 0.0842553 0.0891814"/>
  <asset>
    <mesh name="base" file="base.stl"/>
    <mesh name="shoulder" file="shoulder.stl"/>
    <mesh name="upperarm" file="upperarm.stl"/>
    <mesh name="forearm" file="forearm.stl"/>
    <mesh name="wrist1" file="wrist1.stl"/>
    <mesh name="wrist2" file="wrist2.stl"/>
    <mesh name="wrist3" file="wrist3.stl"/>
    <mesh name="peg20231110" file="peg20231110.STL"/>
  </asset>
  <worldbody>
	<body name="FK_link"  pos = "-0.35  -0.01 0.3"  euler="1.5707963267948966 0 0">
      <geom euler="0 0 0" type="mesh" rgba="0.7 0.7 0.7 1" mesh="fk"/>
 </body>

                 <!-- <body name="peg_link">
                 
                <joint name="peg_jiont" type= "hinge" pos="0 0 0" axis="0 0 1" range="-6.28319 6.28319"/>

                <geom pos="-0.35  0.08 0.3" quat="0.5 0.5 -1.5 0.5" type="mesh" rgba="0.7 0.7 0.7 1" mesh="peg20231110"/>
                </body> -->

<geom quat="0 0 0 1" type="mesh" contype="0" conaffinity="0" group="1" density="0" rgba="0.7 0.7 0.7 1" mesh="base"/>
    <geom quat="0 0 0 1" type="mesh" rgba="0.7 0.7 0.7 1" mesh="base"/>
    <body name="shoulder_link" pos="0 0 0.089159" quat="0 0 0 1">
      <inertial pos="0 0 0" mass="3.7" diaginertia="0.0149724 0.0149724 0.0104063"/>
      <joint name="shoulder_pan_joint" pos="0 0 0" axis="0 0 1" range="-6.28319 6.28319"/>
      <geom quat="0 0 0 1" type="mesh" contype="0" conaffinity="0" group="1" density="0" rgba="0.7 0.7 0.7 1" mesh="shoulder"/>
      <geom quat="0 0 0 1" type="mesh" rgba="0.7 0.7 0.7 1" mesh="shoulder"/>
      <body name="upper_arm_link" quat="0.707107 0.707107 0 0">
        <inertial pos="-0.2125 0 0.136" quat="0.707107 0 0.707107 0" mass="8.393" diaginertia="0.133886 0.133886 0.0151074"/>
        <joint name="shoulder_lift_joint" pos="0 0 0" axis="0 0 1" range="-6.28319 6.28319"/>
        <geom pos="0 0 0.136" quat="0.5 0.5 -0.5 -0.5" type="mesh" contype="0" conaffinity="0" group="1" density="0" rgba="0.7 0.7 0.7 1" mesh="upperarm"/>
        <geom pos="0 0 0.136" quat="0.5 0.5 -0.5 -0.5" type="mesh" rgba="0.7 0.7 0.7 1" mesh="upperarm"/>
        <body name="forearm_link" pos="-0.425 0 0">
          <inertial pos="-0.2125 0 0.0165" quat="0.707107 0 0.707107 0" mass="2.275" diaginertia="0.0312168 0.0312168 0.004095"/>
          <joint name="elbow_joint" pos="0 0 0" axis="0 0 1" range="-3.14159 3.14159"/>
          <geom pos="0 0 0.0165" quat="0.5 0.5 -0.5 -0.5" type="mesh" contype="0" conaffinity="0" group="1" density="0" rgba="0.7 0.7 0.7 1" mesh="forearm"/>
          <geom pos="0 0 0.0165" quat="0.5 0.5 -0.5 -0.5" type="mesh" rgba="0.7 0.7 0.7 1" mesh="forearm"/>
          <body name="wrist_1_link" pos="-0.39225 0 0.10915">
            <inertial pos="0 0 0" quat="0.5 0.5 -0.5 0.5" mass="1.219" diaginertia="0.00342844 0.00262426 0.00262426"/>
            <joint name="wrist_1_joint" pos="0 0 0" axis="0 0 1" range="-6.28319 6.28319"/>
            <geom pos="0 0 -0.093" quat="0.707107 0.707107 0 0" type="mesh" contype="0" conaffinity="0" group="1" density="0" rgba="0.7 0.7 0.7 1" mesh="wrist1"/>
            <geom pos="0 0 -0.093" quat="0.707107 0.707107 0 0" type="mesh" rgba="0.7 0.7 0.7 1" mesh="wrist1"/>
            <body name="wrist_2_link" pos="0 -0.09465 -1.9413e-11" quat="0.707107 0.707107 0 0">
              <inertial pos="0 0 0" quat="0.5 0.5 -0.5 0.5" mass="1.219" diaginertia="0.0021942 0.00178515 0.00178515"/>
              <joint name="wrist_2_joint" pos="0 0 0" axis="0 0 1" range="-6.28319 6.28319"/>
              <geom pos="0 0 -0.095" type="mesh" contype="0" conaffinity="0" group="1" density="0" rgba="0.7 0.7 0.7 1" mesh="wrist2"/>
              <geom pos="0 0 -0.095" type="mesh" rgba="0.7 0.7 0.7 1" mesh="wrist2"/>
              <body name="wrist_3_link" pos="0 0.0823 -1.688e-11" quat="0.707107 -0.707107 0 0">
                <inertial pos="0 0 -0.00627169" quat="0.5 0.5 -0.5 0.5" mass="0.3389" diaginertia="0.000132134 0.00011486 0.00011486"/>
                <joint name="wrist_3_joint" pos="0 0 0" axis="0 0 1" range="-6.28319 6.28319"/>
                <geom pos="0 0 -0.0818" quat="0.707107 0.707107 0 0" type="mesh" contype="0" conaffinity="0" group="1" density="0" rgba="0.7 0.7 0.7 1" mesh="wrist3"/>
                <geom pos="0 0 -0.0823" quat="0.707107 0.707107 0 0" type="mesh" rgba="0.7 0.7 0.7 1" mesh="wrist3"/>
                <geom size="0.005 0.005 0.005" pos="0 0 -0.01" quat="0.5 0.5 -0.5 0.5" type="box"/>
                <body name="peg_link">
                <joint name="peg_jiont" type= "hinge" pos="0 0 0" axis="0 0 1" range="-6.28319 6.28319"/>
                <geom pos="0 0 0.005" quat="0.5 0.5 -1.5 0.5" type="mesh" contype="0" conaffinity="0" group="1" density="0" rgba="0.7 0.7 0.7 1" mesh="peg20231110"/> 
                <geom pos="0 0 0.005" quat="0.5 0.5 -1.5 0.5" type="mesh" rgba="0.7 0.7 0.7 1" mesh="peg20231110"/>
                <site name="ft_sensor" pos="0 0 0.0" rgba="1 0 0 0.3" size="0.01 0.01 0.01" type="box"/>  

                </body>
              </body>
            </body>
          </body>
        </body>
      </body>
    </body> 
  </worldbody>
 <sensor>
        <force name = "force" site="ft_sensor" />
    </sensor>
</mujoco>

```

</details>

   I am currently unable to solve this problem, but I would like to isolate the force effects from the robotic arm and only consider the force issues on the peg and block surface.  We learned in the Selection summary of the Collision detection chapter(https://mujoco.readthedocs.io/en/stable/computation/index.html?highlight=%20collision%20detection%20#collision-detection) that we can modify `mjModel.opt.collision` to select collision targets (I only need to detect peg and block surfaces)
but, we use code to find the `mjModel.opt.collision` is failed, because no `.collision` in the `mjModel.opt`:

```
m = mujoco.MjModel.from_xml_path('robot_square.xml')
print(m.opt)
```

```
apirate: 100.0
  cone: 0
  density: 0.0
  disableactuator: 0
  disableflags: 0
  enableflags: 0
  gravity: array([0., 0., 0.])
  impratio: 1.0
  integrator: 0
  iterations: 100
  jacobian: 2
  ls_iterations: 50
  ls_tolerance: 0.01
  magnetic: array([ 0. , -0.5,  0. ])
  mpr_iterations: 50
  mpr_tolerance: 1e-06
  noslip_iterations: 0
  noslip_tolerance: 1e-06
  o_friction: array([1.e+00, 1.e+00, 5.e-03, 1.e-04, 1.e-04])
  o_margin: 0.0
  o_solimp: array([9.0e-01, 9.5e-01, 1.0e-03, 5.0e-01, 2.0e+00])
  o_solref: array([0.02, 1.  ])
  sdf_initpoints: 40
  sdf_iterations: 10
  solver: 2
  timestep: 0.002
  tolerance: 1e-08
  viscosity: 0.0
```

    Then we wanted to try using the `contact/⁠exclude` tag, but we couldn't find a detailed demo. We made the following modifications ourselves, but it didn't work.
```XML
 <contact>
        <exclude body1="peg_link", body2="wrist_3_joint"/>
    </contact>
```

    So I need the following help: 
1. Can you provide examples of the correct use of `mjModel. opt. collation `and` contact/exclude`; 
2. If my above approach is not implemented and only considers the contact force between the axis and the surface of the block, can I provide a feasible solution.

sunnymints, 2023-12-13:
> **How to ask for help**
> 
> First, read our quick guide to [asking good questions](https://github.com/google-deepmind/mujoco#asking-questions). Below is a template for you to use:
> 
> Hi,
> 
> I'm a student and I'm trying to use MuJoCo for _robot drl____.
> 
> I'm looking for some help with contact force.
> 
> Here is a model which explains my question: I'm very sorry to post again to ask for help and occupy everyone's resources. I have previously posted in this issuse (#1249) the problem of inconsistent directions of the peg block surface contact force.
> minimal XML
> 
> I am currently unable to solve this problem, but I would like to isolate the force effects from the robotic arm and only consider the force issues on the peg and block surface. We learned in the Selection summary of the Collision detection chapter(https://mujoco.readthedocs.io/en/stable/computation/index.html?highlight=%20collision%20detection%20#collision-detection) that we can modify `mjModel.opt.collision` to select collision targets (I only need to detect peg and block surfaces) but, we use code to find the `mjModel.opt.collision` is failed, because no `.collision` in the `mjModel.opt`:
> 
> ```
> m = mujoco.MjModel.from_xml_path('robot_square.xml')
> print(m.opt)
> ```
> 
> ```
> apirate: 100.0
>   cone: 0
>   density: 0.0
>   disableactuator: 0
>   disableflags: 0
>   enableflags: 0
>   gravity: array([0., 0., 0.])
>   impratio: 1.0
>   integrator: 0
>   iterations: 100
>   jacobian: 2
>   ls_iterations: 50
>   ls_tolerance: 0.01
>   magnetic: array([ 0. , -0.5,  0. ])
>   mpr_iterations: 50
>   mpr_tolerance: 1e-06
>   noslip_iterations: 0
>   noslip_tolerance: 1e-06
>   o_friction: array([1.e+00, 1.e+00, 5.e-03, 1.e-04, 1.e-04])
>   o_margin: 0.0
>   o_solimp: array([9.0e-01, 9.5e-01, 1.0e-03, 5.0e-01, 2.0e+00])
>   o_solref: array([0.02, 1.  ])
>   sdf_initpoints: 40
>   sdf_iterations: 10
>   solver: 2
>   timestep: 0.002
>   tolerance: 1e-08
>   viscosity: 0.0
> ```
> 
> ```
> Then we wanted to try using the `contact/⁠exclude` tag, but we couldn't find a detailed demo. We made the following modifications ourselves, but it didn't work.
> ```
> 
> ```
>  <contact>
>         <exclude body1="peg_link", body2="wrist_3_joint"/>
>     </contact>
> ```
> 
> ```
> So I need the following help: 
> ```
> 
>     1. Can you provide examples of the correct use of `mjModel. opt. collation `and` contact/exclude`;
> 
>     2. If my above approach is not implemented and only considers the contact force between the axis and the surface of the block, can I provide a feasible solution.

We have made the following modifications to the <contact> based on the example in` humanoid.xml`, but there are still no changes.
`  <contact>
    <exclude body1="peg_link" body2="wrist_3_link"/>
    <exclude body1="peg_link" body2="wrist_2_link"/>
    <exclude body1="peg_link" body2="wrist_1_link"/>
     <exclude body1="peg_link" body2="forearm_link"/>
     <exclude body1="peg_link" body2="upper_arm_link"/>
      <exclude body1="peg_link" body2="shoulder_link"/>
    <pair geom1="peg" geom2="FK"/>

  </contact>`

yuvaltassa, 2023-12-13:
Sorry, `mjOption.collision` was removed, we need to update the docs. In any case it would not have helped you. Or I think It would not have helped you, since I don't understand the problem.

Please try to articulate what you want using a minimal model that we can load, that displays the problem you are encountering.

yuvaltassa, 2024-01-02:
Closing due to lack of activity.

closed 2024-01-02 by yuvaltassa