google-deepmind/mujoco #1277: How to set contact points to constraint the specific link of one Robots?

JameScottX, 2023-12-13:
Hi,
I'm a student and I'm trying to use MuJoCo to develop the locomotion algorithm of legged robots.
I'm looking for some help with that I can not find a way to constrain the robot's foot or some other part of the body even though there is no geometry collision.
For me, I concur very much if the MuJoCo can constrain any points as I want. If there is any solution I would appreciate quite.

yuvaltassa, 2023-12-13:
Use [equality constraints](https://mujoco.readthedocs.io/en/latest/XMLreference.html#equality).

closed 2023-12-13 by yuvaltassa

JameScottX, 2023-12-13:
Thank you very much for your reply so quickly! Coincidentally, I just have the same idea as you recommended. I had applied the equality constraint since I tried to get a closure chain of a robot leg. But, here is the thing, I found that the position residual always exists which I realized is a strong PD constraint. So, is there any convenient solution to reduce the residual, such as turning the PD gain?
By the way, I just watched the video of your presentation on the MIT Robotics channel and read your several papers,  which inspired me a lot. I really favor the idea that "predictive control is like surfing."

JameScottX, 2023-12-13:
> Use [equality constraints](https://mujoco.readthedocs.io/en/latest/XMLreference.html#equality).

Thank you very much for your reply so quickly! Coincidentally, I just have the same idea as you recommended. I had applied the equality constraint since I tried to get a closure chain of a robot leg. But, here is the thing, I found that the position residual always exists which I realized is a strong PD constraint. So, is there any convenient solution to reduce the residual, such as turning the PD gain?
By the way, I just watched the video of your presentation on the MIT Robotics channel and read your several papers, which inspired me a lot. I really favor the idea that "predictive control is like surfing."

yuvaltassa, 2023-12-13:
You can tune equality constraints to be stiffer using the [solver parameters](https://mujoco.readthedocs.io/en/latest/modeling.html#solver-parameters).