google-deepmind/mujoco #1278: Wrong value silently used when URDF limit is really big (~1e38)

traversaro, 2023-12-13:
Minimum Working Example (available at https://gist.githubusercontent.com/traversaro/a07efb8c9448ade704d8aace46f17e6f/raw/79e6aa32f0fdfa8558e5859bbe2ab7f22bcd966c/reproduce.py):
~~~
wget https://gist.githubusercontent.com/traversaro/a07efb8c9448ade704d8aace46f17e6f/raw/79e6aa32f0fdfa8558e5859bbe2ab7f22bcd966c/reproduce.py
python reproduce.py
~~~

on Ubuntu 22.04 with apt dependencies, both with 3.0.1 and https://github.com/google-deepmind/mujoco/commit/0d37670698b49e78594c635b7f54682ce51ec95d it fails with output:
~~~
<mujoco model="cartpole">
  <compiler angle="radian"/>
  <statistic meansize="0.5" extent="2" center="0 0 0.25"/>
  <worldbody>
    <body name="pole">
      <inertial pos="0 0 0.5" mass="0.1" diaginertia="0.00833396 0.00833396 1.25e-06"/>
      <joint name="pivot" pos="0 0 0" axis="1 0 0" range="-2147483648 -2147483648"/>
    </body>
  </worldbody>
</mujoco>

Traceback (most recent call last):
  File "/home/traversaro/idyntreepypi/reproduce.py", line 38, in <module>
    mj_model = mujoco.MjModel.from_xml_string(model_mjcf_string)
ValueError: Error: range[0] should be smaller than range[1] in joint 'pivot' (id = 0)
Object name = pivot, id = 0, line = 7, column = -1
~~~

The problem is that for some reason the URDF joint:
~~~xml
        <limit effort="500.0" velocity="10.0" lower="-3.4028234663852886e+38" upper="3.4028234663852886e+38"/> 
~~~
get translated in the mjcf joint:
~~~xml
      <joint name="pivot" pos="0 0 0" axis="1 0 0" range="-2147483648 -2147483648"/>
~~~

While in theory one on the URDF side could use a `continuous` joint instead of a `revolute` joint with such huge limits, if the number it is too great to parse it would be great if there was an error or at least a warning instead of silently using a wrong value.

traversaro, 2023-12-13:
cc @flferretti

traversaro, 2023-12-13:
Digging a bit deeper, this seems to be some kind of saturation with a 32-bit integer. That could make sense, but what is quite surprising is that for some reason any big number >= 1e10 gets saturated with a negative value, i.e. `-2147483648`.

traversaro, 2023-12-13:
> Do you want to put a breakpoint [here](https://github.com/google-deepmind/mujoco/blob/0d37670698b49e78594c635b7f54682ce51ec95d/src/xml/xml_urdf.cc#L471) and step through? I expect this should not be hard to figure out. If not, we'll get to it at some point.
> 
> Thanks!

Thanks for the pointer! It seems that the value is correctly loaded in memory, so I guess the problem is in the export (i.e. mj_saveLastXML).

yuvaltassa, 2023-12-13:
Thanks for this, fix incoming.

traversaro, 2023-12-13:
Great, thanks a lot!

commit 5b2c98f8 (2023-12-13): When saving XMLs, don't round floats that are bigger than INT_MAX. Fixes #1278

PiperOrigin-RevId: 590692020
Change-Id: Icd11d20f47bc4163eb4378f4cb031b8fd9d13612
closed 2023-12-13 by copybara-service[bot]