google-deepmind/mujoco #1280: Passive viewer memory allocation error

lbusellato, 2023-12-14:
Hello,
this is a followup to #1219 . Since I am still having the issues described there, but the issue topic is different, I opened a new one.

So, I have the same body using the elasticity plugin and a flexcomp:

<details>
  <summary>scene XML</summary>

```XML
<mujoco model="deformable">
   <extension>
      <plugin plugin="mujoco.elasticity.solid"/>
   </extension>
   <size memory="1G"/>
   <worldbody>
      <flexcomp type="grid" count="4 4 4" spacing=".05 .05 .05" pos=".5 .5 .2"
               radius=".0" rgba="0 .7 .7 1" name="softbody" dim="3" mass="0.1">
         <contact condim="3" solref="0.01 1" solimp=".95 .99 .0001" selfcollide="none"/>
         <edge damping="1"/>
         <plugin plugin="mujoco.elasticity.solid">
            <config key="poisson" value="0.2"/>
            <!--Units are in Pa (SI)-->
            <config key="young" value="5e4"/>
         </plugin>
      </flexcomp>
   </worldbody>
</mujoco>
```

</details>

I am using the following script to launch the passive viewer and to load the xml:

<details>
  <summary>python script</summary>

```python
import mujoco
import mujoco.viewer
from os.path import abspath, dirname, join

path = abspath(dirname(__file__))

m = mujoco.MjModel.from_xml_path(join(path, 'deformable.xml'))
d = mujoco.MjData(m)

with mujoco.viewer.launch_passive(m, d) as viewer:
    while viewer.is_running():
        mujoco.mj_step(m, d)
        viewer.sync()

```

</details>

My setup is:

- Python 3.8.10
- Mujoco 3.10
- Intel integrated / Nvidia graphics

The problems I'm still having are:

1. Running the script above results always in the error message:
```
ERROR: mj_stackAlloc: insufficient memory: max = 0, available = 0, requested = 24 (ne = 0, nf = 0, nefc = 0, ncon = 0)
```
2. The memory allocation with `<size memory="..."/>` still isn't working properly. Using suffixes does not seem to work, in the sense that in the above error message `max` is always 0. Just using numbers works (the max value in the error message is correct), but strangely anything above `33000` results again in `max = 0`.

Using `simulate`, these issues are not present and everything works as intended.

What I tried:

1. Switch between intel and nvidia graphics with prime-select, no change
2. Build mujoco from source, no change
3. Install mujoco from github head with `pip install mujoco --pre -f https://py.mujoco.org/`, no change

I am out of ideas. Any help is greatly appreciated.

nimrod-gileadi, 2023-12-14:
Thanks. This does look like the same issue, but indeed we only fixed it in simulate and not the passive viewer. Apologies.

nimrod-gileadi, 2023-12-14:
Actually, the cause is quite different, and only affects the passive viewer. We know how to fix this, just going through code review.

closed 2023-12-18 by copybara-service[bot]

ykim104, 2024-02-22:
I have the similar problem and I just installed latest release. i get this error even when not using the passive viewer

commit 7cb7c87f (2023-12-18): Fix a few issues with the passive viewer.

1. Create an arena for the mjData instance used by the passive viewer visualization.

When using the passive viewer, stuff gets copied from the real mjData into a minimal struct.
That struct didn't have a stack, and now visualization for Flex does stack allocs.

2. Add missing a missing field in scene state for flex visualization.

3. Fix a memory leak where mjvScene wasn't released on exit in the passive viewer.

4. Add some locks in places where the render thread and Simulate::Sync collide.

This fixes #1280.

PiperOrigin-RevId: 591891676
Change-Id: [... cut]