google-deepmind/mujoco #1285: Collision detection and maximum number of contact points using hfield

heleiduan, 2023-12-18:
Hi,

I'm looking for some help to better understand collision detection for hfield, because the simulation speed relates to this aspect a lot. From the [hfield documentation](https://mujoco.readthedocs.io/en/stable/XMLreference.html#asset-hfield), `The number of possible contacts between a height field and a geom is limited to 9; any contacts beyond that are discarded.` However, when simulating with a minimal example below, it looks like `box` geom (10cm cube) has 16 contact points from `data.contact`, and `capsule` geom has 12. It seems the number of contact points is related to the size of the geoms, but is the maximum of 9 contacts enforced regardless of the size of geoms?

Here is a model with minimal running code which explains my question using MuJoCo 3.0.1. Thank you!

<details>
  <summary>minimal XML</summary>

```XML
<mujoco model='robot'>
  <compiler angle='degree' eulerseq='xyz' meshdir='stl'/>
  <option iterations='50' solver='Newton' timestep='0.0005' gravity='0 0 -9.80665' cone='elliptic'>
  </option>

  <asset>
    <hfield name='hfield0' nrow='400' ncol='400' size="20 20 20 0.1"/>
  </asset>

  <default>
  <default class='robot'>
      <geom group='1'/>
    </default>
    <geom conaffinity='0' condim='1' contype='0' solref='0.005 1'/>
    <pair condim='3' solref='0.005 1'/>
    <joint solreflimit='0.001 1' />
  </default>

  <worldbody>
    <geom name='floor' pos='0 0 0' hfield='hfield0' type='hfield' condim='3' priority='100'/>
    <light mode='trackcom' pos='0 0 5' dir='0 0 -1'/>
    <light directional='true' pos='0 0 3' dir='0 0 -1' diffuse='0.2 0.2 0.2' specular='0 0 0'/>
    <body name='box' pos='0 0 1' euler='0 0 0' childclass='robot'>
      <geom name='box' pos='0 0 0' size='0.1 0.1 0.1' type='box' rgba='0 1 0 1.0' mass='50.0'/>
      <joint type='free'/>
    </body>

    <body name='capsule' pos='2 0 1' euler='0 0 0' childclass='robot'>
      <geom name='capsule' pos='0 0 0' size='0.1 0.2' type='capsule' rgba='1.0 0 0 1.0' mass='50.0'/>
      <joint type='free'/>
    </body>
  </worldbody>

  <contact>
    <pair geom1="box" geom2="floor"/>
    <pair geom1="capsule" geom2="floor"/>
  </contact>

</mujoco>
```

</details>

Here is the python script to run the model file, 
<details>
  <summary>python script</summary>

```
import mujoco as mj
import mujoco.viewer
import time

model_path = "./hfield_test.xml"
m = mj.MjModel.from_xml_path(model_path)
d = mj.MjData(m)

hfield_radius_x, hfield_radius_y, hfield_max_z, hfield_min_z = m.hfield('hfield0').size
print(f"hfield pixel size: {m.hfield_nrow} x {m.hfield_ncol}")
print(f"hfield side : {hfield_radius_x*2}m x {hfield_radius_y*2}m")
print(f"hfield resolution: {hfield_radius_x*2/m.hfield_nrow} x {hfield_radius_y*2/m.hfield_ncol} m/pixel")
input("Press Enter to continue...")

box_geom_id = mj.mj_name2id(m, mj.mjtObj.mjOBJ_GEOM, "box")
capsule_geom_id = mj.mj_name2id(m, mj.mjtObj.mjOBJ_GEOM, "capsule")

with mujoco.viewer.launch_passive(m, d) as viewer:
    while True:
        step_start = time.time()
        mj.mj_step(m, d)
        viewer.sync()
        time_until_next_step = m.opt.timestep - (time.time() - step_start)
        if time_until_next_step > 0:
            time.sleep(time_until_next_step)

        box_contact_points = 0
        capsule_contact_points = 0
        for i, contact in enumerate(d.contact):
            if contact.geom1 == box_geom_id or contact.geom2 == box_geom_id:
                box_contact_points += 1
            if contact.geom1 == capsule_geom_id or contact.geom2 == capsule_geom_id:
                capsule_contact_points += 1
        print(f"box contact points: {box_contact_points}, capsule contact points: {capsule_contact_points}")
```
</details>

nimrod-gileadi, 2023-12-20:
This line of the documentation is very very out of date. Thanks for pointing it out.

In versions before MuJoCo 1.4 (released 2016), the limit of contacts between any two geoms was 9.
Since version 1.5 (2017), the limit has been 50, and that's what caps hfield contacts.

The constant is `mjMAXCONPAIR` in [mjmodel.h](https://github.com/google-deepmind/mujoco/blob/3.1.1/include/mujoco/mjmodel.h#L29).

closed 2023-12-20 by copybara-service[bot]

heleiduan, 2023-12-20:
@nimrod-gileadi Thank you for updating this! One more question on it, to speed up sampling speed, it is possible to overwrite  `mjMAXCONPAIR` in the model file to reduce the contact limit?

nimrod-gileadi, 2023-12-20:
To override mjMAXCONPAIR you would need to rebuild MuJoCo from source after you edit mjmodel.h.

But if you run into the contact limit you'll probably get unphysical behaviour, because the contacts are not dropped in a clever way - you might only get contacts on one side of the object, with the other side falling through the hfield.

heleiduan, 2023-12-20:
@nimrod-gileadi Thanks! I am more leaning towards reducing the number of contacts, due to slower sampling speed caused by hfield. Given `box` geom is usually 4 at minimum, it looks like reducing from 50 to a lower number can speed up simulation by a lot.

commit 2a46ca37 (2023-12-20): Update the documentation on the maximum number of contacts from hfield collisions.

The docs were incorrectly asserting that the limit is 9 contacts, which hasn't been true since version 1.5.

This fixes #1285.

PiperOrigin-RevId: 592526249
Change-Id: Id45d73666fec8dcfd18d11605975e787f2cfeba2