google-deepmind/mujoco #1286: joint qpos and framepos outputs are different after calling mj_step

lbergmann1, 2023-12-20:
Hi,

I'm a PhD student and I'm trying to use MuJoCo for motion planning. 
I use a 'framepos' sensor to measure the current position of an object. The object is controlled by an actuator which outputs a force in x-direction of the global frame.

Here a minimal example code including a model xml string:

```python
import mujoco

xml = """
<?xml version="1.0" encoding="utf-8"?>
<mujoco model="planar_transport_system">
    <compiler angle="radian" coordinate="local" />
    <option timestep="0.001" cone="elliptic" jacobian="auto" gravity="0 0 -9.81"/>
    
    <worldbody>        
        <body name="body_0" pos="0 0 0" gravcomp="1">
            <joint name="joint_0" type="free" damping="0"></joint>
            <geom name="geom_0" type="box" size="0.0775 0.0775 0.006" mass="1" pos="0 0 0" rgba="0 1 0 1"/>
        </body> 
        
        <light directional="true" ambient="0.2 0.2 0.2" diffuse="0.8 0.8 0.8" specular="0.3 0.3 0.3" castshadow="false" pos="0 0 4" dir="0 0 -1" name="light0" />
    </worldbody>

    <sensor>
        <framepos name="pos_sensor" objtype="xbody" objname="body_0"/>
    </sensor>

    <actuator>
        <general name="actuator_x" joint="joint_0" gear="1 0 0 0 0 0" dyntype="none" gaintype="fixed" gainprm="1 0 0" biastype="none"/>
    </actuator>
</mujoco>
"""

model = mujoco.MjModel.from_xml_string(xml)
data = mujoco.MjData(model)

# joint 
joint_name = "joint_0"
joint_id = model.joint(joint_name).id
joint_addr = model.jnt_qposadr[joint_id]
joint_dim = 7
# actuator
actuator_name = "actuator_x"
actuator_id = model.actuator(actuator_name).id
# sensor
pos_sensor_name = "pos_sensor"
pos_sensor_view = model.sensor(pos_sensor_name)
pos_sensor_id = pos_sensor_view.id
pos_sensor_adr = pos_sensor_view.adr[0]
pos_sensor_dim = pos_sensor_view.dim[0]

for step in range(0,10):
    data.ctrl[actuator_id] = 2
    mujoco.mj_step(model, data, nstep=1)

    print(f"step {step} - sensor output: {data.sensordata[pos_sensor_adr]:.6f}, joint qpos: {data.qpos[joint_addr]:.6f}")

```

Here is the output of my example code (for simplicity, I only considered the x components):
```
step 0 - sensor output: 0.000000, joint qpos: 0.000002
step 1 - sensor output: 0.000002, joint qpos: 0.000006
step 2 - sensor output: 0.000006, joint qpos: 0.000012
step 3 - sensor output: 0.000012, joint qpos: 0.000020
step 4 - sensor output: 0.000020, joint qpos: 0.000030
step 5 - sensor output: 0.000030, joint qpos: 0.000042
step 6 - sensor output: 0.000042, joint qpos: 0.000056
step 7 - sensor output: 0.000056, joint qpos: 0.000072
step 8 - sensor output: 0.000072, joint qpos: 0.000090
step 9 - sensor output: 0.000090, joint qpos: 0.000110
```
I was wondering why the sensor output is 0 after calling 'mj_step()'. So I compared the sensor values to the joint qpos. Both should output the current position of the object w.r.t. the global frame. As shown in the example output, the sensor outputs and the qpos differ by exactly one call of `mj_step()`. Is this the desired behavior? For this particular example, I would have expected the sensor outputs and joint qpos to be similar in each step.

### System Info
- OS: Linux-6.2.0-39-generic-x86_64-with-glibc2.35 # 40~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Nov 16 10:53:04 UTC 2
- Python: 3.11.6
- MuJoCo: 3.1.1

Many thanks in advance!

nimrod-gileadi, 2023-12-20:
Although it's counter-intuitive, this is the expected behaviour.

Take a look at the [simulation pipeline](https://mujoco.readthedocs.io/en/stable/computation/index.html#simulation-pipeline) docs to see what happens when you call mj_step.

qpos is an **input** to forward dynamics, and based on this input, the accelerations and all the sensor values are computed. So your sensor readings will always correspond to the qpos at the start of the step.

The very last part of mj_step is integration, where the computed accelerations are used to advance qpos, so the qpos you read at the end of the step corresponds to the next step's position.

Does that make sense?

yuvaltassa, 2024-01-02:
If you want to get the "latest" sensor readings, you can always call `mj_step1` after each `mj_step`. This is expensive but will update the sensor values to the current state. Note that this cannot work for sensors that report force or acceleration, since these are also a function of the control, which is perhaps unknown (since it might be a function of the sensor values).

closed 2024-01-02 by yuvaltassa