google-deepmind/mujoco #1291: [`MJX`] loading `mjx_model.actuator_ctrlrange` is not JITable

Kallinteris-Andreas, 2023-12-22:
Hi,

I'm a maintainer of [Gymnasium](https://gymnasium.farama.org/) & the project manager of [Gymnasium-Robotics](https://robotics.farama.org/), and I'm trying to use `MuJoCo-MJX` for "prototyping `MJX`-based RL environments in `Gymnasium`, `Gymnasium-Robotics`, `Metaworld`, `MO-Gymnasium`".

In the Gymnasium Environment API, we have the [`action_space`](https://gymnasium.farama.org/api/env/#gymnasium.Env.action_space) member variable that specifies the set of valid actions the agent can perform.

I am trying to set the `low` and `high` limits of the `action_space` using the `mjx_model.actuator_ctrlrange` constant member variable of the loaded model (this is the way we do it with `mujoco` environments, we load `model.actuator_ctrlrange`), but I get this error:
(my `test.py` just JIT's the `__init__` and runs it)
```sh
$ JAX_TRACEBACK_FILTERING=off python3.11 test.py
JIT success
Traceback (most recent call last):
  File "/home/master-andreas/gym/mjx/test.py", line 39, in <module>
    jit_test()
  File "/home/master-andreas/.local/lib/python3.11/site-packages/jax/_src/traceback_util.py", line 177, in reraise_with_filtered_traceback
    return fun(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^
  File "/home/master-andreas/.local/lib/python3.11/site-packages/jax/_src/pjit.py", line 256, in cache_miss
    outs, out_flat, out_tree, args_flat, jaxpr = _python_pjit_helper(
                                                 ^^^^^^^^^^^^^^^^^^^^
  File "/home/master-andreas/.local/lib/python3.11/site-packages/jax/_src/pjit.py", line 162, in _python_pjit_helper
    args_flat, _, params, in_tree, out_tree, _ = infer_params_fn(
                                                 ^^^^^^^^^^^^^^^^
  File "/home/master-andreas/.local/lib/python3.11/site-packages/jax/_src/api.py", line 325, in infer_params
    return pjit.common_infer_params(pjit_info_args, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/master-andreas/.local/lib/python3.11/site-packages/jax/_src/pjit.py", line 486, in common_infer_params
    jaxpr, consts, canonicalized_out_shardings_flat = _pjit_jaxpr(
                                                      ^^^^^^^^^^^^
  File "/home/master-andreas/.local/lib/python3.11/site-packages/jax/_src/pjit.py", line 963, in _pjit_jaxpr
    jaxpr, final_consts, out_type = _create_pjit_jaxpr(
                                    ^^^^^^^^^^^^^^^^^^^
  File "/home/master-andreas/.local/lib/python3.11/site-packages/jax/_src/linear_util.py", line 348, in memoized_fun
    ans = call(fun, *args)
          ^^^^^^^^^^^^^^^^
  File "/home/master-andreas/.local/lib/python3.11/site-packages/jax/_src/pjit.py", line 916, in _create_pjit_jaxpr
    jaxpr, global_out_avals, consts = pe.trace_to_jaxpr_dynamic(
                                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/master-andreas/.local/lib/python3.11/site-packages/jax/_src/profiler.py", line 314, in wrapper
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/home/master-andreas/.local/lib/python3.11/site-packages/jax/_src/interpreters/partial_eval.py", line 2203, in trace_to_jaxpr_dynamic
    jaxpr, out_avals, consts = trace_to_subjaxpr_dynamic(
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/master-andreas/.local/lib/python3.11/site-packages/jax/_src/interpreters/partial_eval.py", line 2225, in trace_to_subjaxpr_dynamic
    ans = fun.call_wrapped(*in_tracers_)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/master-andreas/.local/lib/python3.11/site-packages/jax/_src/linear_util.py", line 190, in call_wrapped
    ans = self.f(*args, **dict(self.params, **kwargs))
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/master-andreas/gym/mjx/test.py", line 7, in test
    f_env = HalfCheetahMJXEnv()
            ^^^^^^^^^^^^^^^^^^^
  File "/home/master-andreas/gym/mjx/Gymnasium-kalli/gymnasium/envs/mjx/half_cheetah_j_v5.py", line 50, in __init__
    MJXEnv.__init__(
  File "/home/master-andreas/gym/mjx/Gymnasium-kalli/gymnasium/envs/mjx/mjx_env.py", line 110, in __init__
    low=np.array(low_action_bound),
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/master-andreas/.local/lib/python3.11/site-packages/jax/_src/core.py", line 641, in __array__
    raise TracerArrayConversionError(self)
jax.errors.TracerArrayConversionError: The numpy.ndarray conversion method __array__() was called on traced array with shape float32[6].
The error occurred while tracing the function test at /home/master-andreas/gym/mjx/test.py:6 for jit. This value became a tracer due to JAX operations on these lines:

  operation a:f32[6,2] = device_put[device=None src=None] b
    from line /home/master-andreas/.local/lib/python3.11/site-packages/mujoco/mjx/_src/io.py:121:18 (put_model)
See https://jax.readthedocs.io/en/latest/errors.html#jax.errors.TracerArrayConversionError
```

But when I load `mjx_model.nu` it works fine (which is what your [tutorial](https://colab.research.google.com/github/google-deepmind/mujoco/blob/main/mjx/tutorial.ipynb#scrollTo=ccujYeJ5XOhx) does), but this does not satisfy our requirements because we need to load the bounds along with the size of the action space.

```py
class MJXEnv(gymnasium.functional.FuncEnv[...]):
    """The Base class for MJX Environments in Gymnasium."""
    def __init__(self, full_path, ...):
        self.model = mujoco.MjModel.from_xml_path(fullpath)
        self.mjx_model = mjx.put_model(self.model)

        # Set action space

        # This works
        self.action_space = gymnasium.spaces.Box(
            low=np.array(-np.ones(self.mjx_model.nu)),
            high=np.array(np.ones(self.mjx_model.nu)),
            dtype=np.float32,
        )
        # This does not work
        low_action_bound, high_action_bound = self.mjx_model.actuator_ctrlrange.T
        self.action_space = gymnasium.spaces.Box(
            low=low_action_bound,
            high=high_action_bound,
            dtype=jnp.float32
        )
      ...
    ...
```

Thanks!

Note: this is likely my last issue for MJX base-class for the release of `Gymansium-MJX` environments (hopefully).

btaba, 2024-01-02:
Hi @Kallinteris-Andreas 

`self.mjx_model.actuator_ctrlrange` is a jax Array, see here:

https://github.com/google-deepmind/mujoco/blob/1c7809e54261009698b5baab569faf700c8f8dad/mjx/mujoco/mjx/_src/types.py#L491

which gets ["traced"](https://jax.readthedocs.io/en/latest/notebooks/thinking_in_jax.html#jit-mechanics-tracing-and-static-variables). When you try to create the gymnasium.spaces.Box, you are trying to materialize a traced array, which is what the error states.

Instead you can do:
```python
        self.action_space = gymnasium.spaces.Box(
            low=self.model.actuator_ctrl_range[0],
            high=self.model.actuator_ctrl_range[1],
            dtype=np.float32
        )
```

closed 2024-01-02 by btaba

Kallinteris-Andreas, 2024-01-02:
Thanks!, it works!

For reference: I added the transpose operation
```py
        self.action_space = gymnasium.spaces.Box(
            low=self.model.actuator_ctrlrange.T[0],
            high=self.model.actuator_ctrlrange.T[1],
            dtype=np.float32
        )
```