google-deepmind/mujoco #1292: Python Types are missing :(

Artur-Galstyan, 2023-12-23:
**Is your feature request related to a problem? Please describe.**
Pyright LSP cannot find the MjData, MjModel etc. types 😢 

**Describe the solution you'd like**
A potential solution could be adding python stubs, like a`mujoco.pyi` file or something?

[image]

FYI: I was checking out Brax when I noticed this

kevinzakka, 2024-01-02:
Hi @Artur-Galstyan, I was able to use [pybind11-stubgen](https://github.com/sizmailov/pybind11-stubgen) and point Visual Code to the generated stubs to get type hints.

```
pip install pybind11-stubgen
pybind11-stubgen mujoco  # Installs stubs in $HOME/stubs
```

Now in visual code, I added the following to my `settings.json`:

```
"python.languageServer": "Pylance",
"python.analysis.stubPath": "~/stubs",
```

and the result:

[image]

closed 2024-01-02 by kevinzakka

mluogh-xdof, 2025-08-20:
I was able to get this to work using Pylance typechecking in CI as well. 

I had to patch the `_enum.pyi` file created because the function bindings created expect an int type which works in C because of the way enums are basically ints there, but the Python typechecker throws an error. So we have to patch them to be in IntEnums.

Here's the script I used for this,

https://gist.github.com/mluogh-xdof/805cd47daf7d3690495c74cb1141ddb7

I also ran it to generate bindings using Annotated because there were some typing errors with the standard numpy annotations due to dimension annotations (which might be fixed by this currently unmerged PR https://github.com/sizmailov/pybind11-stubgen/issues/264)

```
pybind11-stubgen mujoco -o typings/  --numpy-array-wrap-with-annotated
```

Now it doesn't just give typehints but gives reasonable typechecking.

kevinzakka, 2025-08-20:
Thanks @mluogh-xdof (and hi @wuphilipp). FYI @saran-t

horizon-blue, 2026-10-25:
Cross-posting from [my comment in #244](https://github.com/google-deepmind/mujoco/issues/244#issuecomment-6206331537):

For anyone still running into this, I've published an unofficial `mujoco-stubs` package [on PyPI](https://pypi.org/project/mujoco-stubs/), following [PEP 561](https://peps.python.org/pep-0561/#stub-only-packages). It contains version-matched releases covering MuJoCo 3.4.0 and above, building on the approach discussed here (many thanks to @kevinzakka and @mluogh-xdof for the stubs generation and patching instructions):

```bash
pip install "mujoco-stubs==3.17.0.*" # or replace with your matching MuJoCo version
```

Longer term, it'd be great if MuJoCo team can consider shipping official type stubs in MuJoCo's Python package, or as a separate `-stubs` package (I'll be happy to transfer the package name if needed!). In the meantime, I hope this helps anyone else running into the same issue. 😄

Artur-Galstyan, 2026-10-26:
I also just maintain a fork where I let Astra Max just add the types for me. Took one hour but now I have types and docstrings everywhere and I will -I guess-just keep using my fork instead and keep it in sync with upstream.