google-deepmind/mujoco #1293: Question for Mujoco's NVIDIA GPU Acceleration Performance

Agnel-Wang, 2023-12-25:
**Description**
Hi,
I'm trying to implement the dexterous hand manipulation RL task in the simulation.
I'm focusing on the need for accurate **tactile sensor** data for fingers.
**Problem**
When using Mujoco RL task, I observed that the GPU acceleration speed is slow.
The figure mentioned in the official documentation [mjx-the-sharp-bits] (https://mujoco.readthedocs.io/en/stable/mjx.html#mjx-the-sharp-bits) is not that good either.
I also tried using Isaac gym, which is fast but the data is not as real as mujoco.
**Expected Improvements**
Can you tell me when you are going to optimise the performance of Nvidia GPU acceleration for Mujoco (not the current simplified model collisions)?
And the approximate time of release of the corresponding version.
For example, in the last month or two, I would be willing to go ahead and wait for that version to be released

btaba, 2024-01-02:
Hi @Agnel-Wang 

What is the specific model you are trying to run? With some model changes in the XML, you may be able to get the speed you're looking for. If not, we can use your example to prioritize feature improvements.

Agnel-Wang, 2024-01-05:
It has nothing to do with the specific model.

I hope mujoco could support High Performance GPU-Based Physics Simulation(Nvidia GPU) and highly parallel simulations with 4096 envs , like Isaac Gym.