google-deepmind/mujoco #1297: mujoco error

dffsd156, 2023-12-30:
I downloaded mujoco 2.1.1 version, compiled it, and pressed simulate,
 but the simulate window pops up for 3 seconds and disappears immediately.
 What's the problem?

<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
  <worldbody>
    <light pos=".4 -.4 .3" dir="-2 2 -1.5" diffuse=".6 .6 .6"/>
    <light pos="-.2 -.4 .3" dir="1 2 -1.5" diffuse=".6 .6 .6"/>
    <geom type="plane" size=".5 .5 .01"/>
    <body name="ball" pos="0 0 0.05">
      <freejoint/>
      <geom size=".03"/>
    </body>
    <body name="hammer" pos="-.05 0 0.18">
      <joint axis="0 1 0"/>
      <geom type="capsule" size=".01" fromto="0 0 0 -.15 0 0"/>
      <geom type="box" size=".02 .02 .03" pos="-.15 0 0"/>
    </body>
  </worldbody>
</mujoco>
```

</details>

Here is a screenshot / video, illustrating my question:
[image]

yuvaltassa, 2024-01-02:
Please give us the details of your hardware and software setup.

dffsd156, 2024-01-03:
> Please give us the details of your hardware and software setup.

[image]
[image]
[image]
[image]
[image]

yuvaltassa, 2024-01-05:
Why are you using version 2.2.1?

Does this issue also happen with the latest version?

dffsd156, 2024-01-06:
> Why are you using version 2.1.1?
> 
> Does this issue also happen with the latest version?

Yes, I'm using mujoco 2.2.1 and when I download the latest version, only the version appears on the screen and the window disappears right away

kevinzakka, 2024-08-30:
Closing because stale, please re-open if you are still running into this issue.

closed 2024-08-30 by kevinzakka