google-deepmind/mujoco #1298: It's hard to me to understand what's the point of FindOrFetch macro and how to override its behavior.

keszegrobert, 2023-12-30:
Hi everyone, I'm an engineer and trying to use MuJoCo for my hobby project. I am trying to build the sources with a minimal build time, so I would use a conan recipe for the dependencies. Most of mujoco's dependencies have their recipes already in the conan center, so I am able to install them once and if they could be integrated, I wouldn't have to rebuild them through mujoco's build ever. [conanfile.py.txt](https://github.com/google-deepmind/mujoco/files/13799259/conanfile.py.txt)
I've looked at the cmake scripts and realized that I cannot use the findorfetch macro with my externally built dependency packages without a broader patching. There are a couple of reasons I am saying this:
1. FindOrFetch.cmake has the same copy in 3 folders:  <root>/cmake, <root>/simulate/cmake,  <root>/sample/cmake .
2. findorfetch macro "calls" are used with USE_SYSTEM_PACKAGE mainly switched OFF instead of using a CMake option to switch it on. I cannot use these packages from external sources, only when I introduce some cmake options for the calls.
3. the FindOrFetch macro in the USE_SYSTEM_PACKAGE case tries to find the targets and if they are not defined then find_package is called. If I wonder how that should work in the case USE_SYSTEM_PACKAGE is ON, I would first call find_package, and then if the ${PACKAGE_NAME}_FOUND is not set, only then I would download them with FetchContent...
4. Some dependencies come only from external sources by using find_package, for example Threads, Python3 and can be of any version. Isn't that a problem when reproducibility is one of the main goals of the project? 
5. Some other dependencies are fetched with FetchContent and not FindOrFetch,  like lodepng, MarchingCube, eigen, gtest. So they cannot be overloaded externally. 

I would like to open a pull request or more with my patches

traversaro, 2023-12-31:
I guess you could be interested inhttps://github.com/google-deepmind/mujoco/pull/937 .

keszegrobert, 2023-12-31:
> I guess you could be interested inhttps://github.com/[/pull/937](https://github.com/google-deepmind/mujoco/pull/937) . Other related issue: #1137 .

Thank you, I have reviewed it. But it is only partially covers the issues I have listed here