google-deepmind/mujoco #1301: Assistance Required for Handling Memory Limitations and Plant Model Simulation in MuJoCo

kankanzheli, 2024-01-01:
Dear MuJoCo Support Team,

I am currently facing a challenging issue with a complex simulation in MuJoCo, and I'm seeking your expertise to find a solution. I am using MuJoCo on an Ubuntu system with 32GB of RAM.

Issue 1: Memory Limitation Error
During my simulation, I encounter a memory allocation error:
ValueError: Error: engine error: mj_stackAlloc: insufficient memory: max = 2567796100, available = 1070765764, requested = 1496482632 (ne = 0, nf = 15, nefc = 15, ncon = 0).

This error occurs in a complex model derived from a voxel representation of an environment scanned by a camera. Each voxel is represented as a cube in the XML file, with each having a free joint and associated collision coefficients for simulating interactions with a robotic arm. Due to the nature of my research, simplifying this model is almost impossible.

Could you provide guidance on how to avoid this memory error and allow MuJoCo to utilize more memory to accommodate the complexity of my model?

Issue 2: Simulating Plant Model Behavior
Additionally, I am working on a plant model. In this simulation, when the robotic arm collides with the plant, the plant is displaced, and I need it to return to its original position once the robotic arm is moved away. This behavior is crucial for my research, but it’s important to implement it in a way that minimizes memory usage.

Could you advise on the best approach to simulate this kind of plant behavior in MuJoCo, especially considering the memory constraints?

I appreciate any advice or resources you can provide, as these issues are critical for the success of my project. Thank you for your time and assistance.

nimrod-gileadi, 2024-01-02:
Any chance you could attach an example XML file where this happens?
Also, what is the number of cubes you create in your model?

kankanzheli, 2024-01-02:
I randomly selected some voxels from a large voxel model and used them in mujoco. A part of my xml file is showed in here. And the number of cubess is more than 2000.
  <worldbody>
    <body name="voxel_22458">
      <geom type="box" size="0.005 0.005 0.005" rgba="0.204 0.357 0.286 1" pos="-0.08 -0.08 0.0" contype="2" conaffinity="1"/>
      <joint name="voxel_22458" type="free" damping="0.1"/>
    </body>
    <body name="voxel_22459">
      <geom type="box" size="0.005 0.005 0.005" rgba="0.239 0.369 0.329 1" pos="-0.07 -0.09 0.0" contype="2" conaffinity="1"/>
      <joint name="voxel_22459" type="free" damping="0.1"/>
    </body>
    <body name="voxel_22460">
      <geom type="box" size="0.005 0.005 0.005" rgba="0.161 0.294 0.278 1" pos="-0.05 -0.1 0.0" contype="2" conaffinity="1"/>
      <joint name="voxel_22460" type="free" damping="0.1"/>
    </body>
    <body name="voxel_22461">
      <geom type="box" size="0.005 0.005 0.005" rgba="0.161 0.294 0.278 1" pos="-0.04 -0.1 0.0" contype="2" conaffinity="1"/>
      <joint name="voxel_22461" type="free" damping="0.1"/>
    </body>
........

nimrod-gileadi, 2024-01-02:
One important change you should make to your model is to specify the `pos` attribute on the `body` and not the `geom` tag.

Some computations are more efficient if the centre of mass of a body is aligned with the position of the body, which won't be the case in your model. There might be implications to collision detection too.

nimrod-gileadi, 2024-01-02:
When you say "more than 2000", do you mean a lot more? Can you give an exact number for when it fails, or at least an upper bound rather than a lower bound?

We have one O(nbody^2) stack allocation [here](https://github.com/google-deepmind/mujoco/blob/3.1.1/src/engine/engine_collision_driver.c#L289), but 2000 would be way too small to account for the size of alloc you got in your error message (which suggests 27k voxels). Given the names of the voxels in your example I'll assume you meant 20000 rather than 2000. :)

kankanzheli, 2024-01-02:
When the ValueError: Error: engine error: mj_stackAlloc: insufficient memory: max = 2567796100, available = 1070765764, requested = 1496482632 (ne = 0, nf = 15, nefc = 15, ncon = 0). happened, I use 2700 cubes in the xml. 
The xml codes showed the number like 22458. For the complete model, the number of cubes is more than 27k. But I just take part of cubes from the complete model. The number just means the number of the cube in the complete model.

kankanzheli, 2024-01-03:
Sure, here's a suggested reply in English to the message you received:

Subject: Re: Clarification on Voxel Count and Memory Issue

Dear [Recipient's Name],

Thank you for your response. Upon reviewing my XML file, I realized that it contains 2276 cubes. Further inspection led me to discover that it's likely other components in the XML file, not related to the cubes, that are contributing to the increased memory usage. These components are parts of the robot.

Below, I've included a more detailed snippet of the XML code for your reference and a better understanding of the structure:

`<mujoco>
  <size nstack="300000000" /> 
  <option gravity="0 0 0" />
  <asset>
    
    <mesh name="frame" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/frame.stl" />
    <mesh name="wheel" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/wheel.stl" />
    <mesh name="Y_linear" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/Y_linear.stl" />
    <mesh name="X_linear" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/X_linear.stl" />
    <mesh name="Z1_linear" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/Z1_linear.stl" />
    <mesh name="Z2_linear" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/Z2_linear.stl" />
    <mesh name="Z3_linear" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/Z3_linear.stl" />
    <mesh name="yaw" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/yaw.stl" />
    <mesh name="pitch" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/pitch.stl" />
    <mesh name="subarm_base" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/subarm_base.STL" />
    <mesh name="joint1" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/joint1.STL" />
    <mesh name="joint2" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/joint2.STL" />
    <mesh name="joint3" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/joint3.STL" />
    <mesh name="joint4" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/joint4.STL" />
    <mesh name="harvest_tool" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/harvest_tool.STL" />
  </asset>

  
  <worldbody>
    <body name="base_body">
      
      <body name="frame">
        <inertial diaginertia="222.35 143.37 209.67" mass="1e-2" pos="0 0 0.66" />
        
        <geom name="frame" type="mesh" mesh="frame" pos="0 0 0" contype="0" conaffinity="0" />
          
        <body name="wheel_rr" pos="-0.894 -0.633 0.206">
          <inertial diaginertia="0.1 0.18 0.1" mass="1e-2" pos="0 0 0" />
          <geom name="wheel_rr" type="mesh" mesh="wheel" pos="0 0 0" euler="0 0 3.14159" contype="0" conaffinity="0" />
          <joint name="wheel_rr_joint" type="hinge" axis="0 -1 0" limited="false" damping="0.1" frictionloss="0.1" />
        </body>
        <body name="wheel_lr" pos="-0.894 0.633 0.206">
          <inertial diaginertia="0.1 0.18 0.1" mass="1e-2" pos="0 0 0" />
          <geom name="wheel_lr" type="mesh" mesh="wheel" pos="0 0 0" euler="0 0 3.14159" contype="0" conaffinity="0" />
          <joint name="wheel_lr_joint" type="hinge" axis="0 -1 0" limited="false" damping="0.1" frictionloss="0.1" />
        </body>
        <body name="wheel_rf" pos="0.894 -0.633 0.206">
          <inertial diaginertia="0.1 0.18 0.1" mass="1e-2" pos="0 0 0" />
          <geom name="wheel_rf" type="mesh" mesh="wheel" pos="0 0 0" euler="0 0 0" contype="0" conaffinity="0" />
          <joint name="wheel_rf_joint" type="hinge" axis="0 1 0" limited="false" damping="0.1" frictionloss="0.1" />
        </body>
        <body name="wheel_lf" pos="0.894 0.633 0.206">
          <inertial diaginertia="0.1 0.18 0.1" mass="1e-2" pos="0 0 0" />
          <geom name="wheel_lf" type="mesh" mesh="wheel" pos="0 0 0" euler="0 0 0" contype="0" conaffinity="0" />
          <joint name="wheel_lf_joint" type="hinge" axis="0 1 0" limited="false" damping="0.1" frictionloss="0.1" />
        </body>
        
        <body name="Y_linear" pos="0 0 1.137">
          <inertial diaginertia="1e-4 1e-4 1e-4" mass="1e-2" pos="0.01 -0.02 -0.1" />
          <geom name="Y_linear" type="mesh" mesh="Y_linear" pos="0 0 0" contype="1" conaffinity="0" />
          <joint name="Y_linear_joint" type="slide" axis="0 -1 0" pos="0 0 0" damping="0.1" frictionloss="0.1" limited="true" range="-0.3455 0.3455" />
          <body name="X_linear" pos="0 0 -0.002">
            <inertial diaginertia="1e-4 1e-4 1e-4" mass="1e-2" pos="-0.04 -0.01 -0.2" />
            <geom name="X_linear" type="mesh" mesh="X_linear" pos="0 0 0" contype="1" conaffinity="0" />
            <joint name="X_linear_joint" type="slide" axis="1 0 0" pos="0 0 0" damping="0.1" frictionloss="0.1" limited="true" range="-0.225 0.281" />
[... 74 lines cut ...]
    <body name="voxel_49025">
      <geom type="box" size="0.005 0.005 0.005" rgba="0.18 0.235 0.271 1" pos="0.19 0.1 0.52" contype="2" conaffinity="1" />
      <joint name="voxel_49025" type="free" damping="0.1" />
    </body>
  </worldbody>
  <actuator>
    <motor name="wheel_rr_joint_motor" joint="wheel_rr_joint" ctrllimited="true" ctrlrange="-3.9 3.9" forcelimited="true" forcerange="-450 450" />
[... cut ...]
    <position name="joint2_position" joint="joint2_joint" ctrllimited="true" ctrlrange="-2.01 2.01" forcelimited="true" forcerange="-25 25" />
    <position name="joint3_position" joint="joint3_joint" ctrllimited="true" ctrlrange="-2.01 2.01" forcelimited="true" forcerange="-25 25" />
    <position name="joint4_position" joint="joint4_joint" ctrllimited="true" ctrlrange="-2.01 2.01" forcelimited="true" forcerange="-25 25" />
  </actuator>
<tendon /></mujoco>`

I hope this provides more clarity on the issue. I look forward to any suggestions or insights you might have regarding the memory allocation and how to optimize it.

kankanzheli, 2024-01-03:
Thank you for your response. Upon reviewing my XML file, I realized that it contains 2276 cubes. Further inspection led me to discover that it's likely other components in the XML file, not related to the cubes, that are contributing to the increased memory usage. These components are parts of the robot.

Below, I've included a more detailed snippet of the XML code for your reference and a better understanding of the structure:

`<mujoco>
  <size nstack="300000000" /> 
  <option gravity="0 0 0" />
  <asset>
    <mesh name="frame" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/frame.stl" />
    <mesh name="wheel" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/wheel.stl" />
    <mesh name="Y_linear" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/Y_linear.stl" />
    <mesh name="X_linear" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/X_linear.stl" />
    <mesh name="Z1_linear" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/Z1_linear.stl" />
    <mesh name="Z2_linear" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/Z2_linear.stl" />
    <mesh name="Z3_linear" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/Z3_linear.stl" />
    <mesh name="yaw" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/yaw.stl" />
    <mesh name="pitch" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/pitch.stl" />
    <mesh name="subarm_base" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/subarm_base.STL" />
    <mesh name="joint1" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/joint1.STL" />
    <mesh name="joint2" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/joint2.STL" />
    <mesh name="joint3" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/joint3.STL" />
    <mesh name="joint4" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/joint4.STL" />
    <mesh name="harvest_tool" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/harvest_tool.STL" />
  </asset>
  <worldbody>
    <body name="base_body">
      
      <body name="frame">
        <inertial diaginertia="222.35 143.37 209.67" mass="1e-2" pos="0 0 0.66" />
        
        <geom name="frame" type="mesh" mesh="frame" pos="0 0 0" contype="0" conaffinity="0" />
          
        <body name="wheel_rr" pos="-0.894 -0.633 0.206">
          <inertial diaginertia="0.1 0.18 0.1" mass="1e-2" pos="0 0 0" />
          <geom name="wheel_rr" type="mesh" mesh="wheel" pos="0 0 0" euler="0 0 3.14159" contype="0" conaffinity="0" />
          <joint name="wheel_rr_joint" type="hinge" axis="0 -1 0" limited="false" damping="0.1" frictionloss="0.1" />
        </body>
        <body name="wheel_lr" pos="-0.894 0.633 0.206">
          <inertial diaginertia="0.1 0.18 0.1" mass="1e-2" pos="0 0 0" />
          <geom name="wheel_lr" type="mesh" mesh="wheel" pos="0 0 0" euler="0 0 3.14159" contype="0" conaffinity="0" />
          <joint name="wheel_lr_joint" type="hinge" axis="0 -1 0" limited="false" damping="0.1" frictionloss="0.1" />
        </body>
        <body name="wheel_rf" pos="0.894 -0.633 0.206">
          <inertial diaginertia="0.1 0.18 0.1" mass="1e-2" pos="0 0 0" />
          <geom name="wheel_rf" type="mesh" mesh="wheel" pos="0 0 0" euler="0 0 0" contype="0" conaffinity="0" />
          <joint name="wheel_rf_joint" type="hinge" axis="0 1 0" limited="false" damping="0.1" frictionloss="0.1" />
        </body>
        <body name="wheel_lf" pos="0.894 0.633 0.206">
          <inertial diaginertia="0.1 0.18 0.1" mass="1e-2" pos="0 0 0" />
          <geom name="wheel_lf" type="mesh" mesh="wheel" pos="0 0 0" euler="0 0 0" contype="0" conaffinity="0" />
          <joint name="wheel_lf_joint" type="hinge" axis="0 1 0" limited="false" damping="0.1" frictionloss="0.1" />
        </body>
        <body name="Y_linear" pos="0 0 1.137">
          <inertial diaginertia="1e-4 1e-4 1e-4" mass="1e-2" pos="0.01 -0.02 -0.1" />
          <geom name="Y_linear" type="mesh" mesh="Y_linear" pos="0 0 0" contype="1" conaffinity="0" />
          <joint name="Y_linear_joint" type="slide" axis="0 -1 0" pos="0 0 0" damping="0.1" frictionloss="0.1" limited="true" range="-0.3455 0.3455" />
          <body name="X_linear" pos="0 0 -0.002">
            <inertial diaginertia="1e-4 1e-4 1e-4" mass="1e-2" pos="-0.04 -0.01 -0.2" />
            <geom name="X_linear" type="mesh" mesh="X_linear" pos="0 0 0" contype="1" conaffinity="0" />
            <joint name="X_linear_joint" type="slide" axis="1 0 0" pos="0 0 0" damping="0.1" frictionloss="0.1" limited="true" range="-0.225 0.281" />
            <body name="Z1_linear" pos="0 0 0">
              <inertial diaginertia="1e-4 1e-4 1e-4" mass="1e-2" pos="-0.04 0 -0.24" />
              <geom name="Z1_linear" type="mesh" mesh="Z1_linear" pos="0 0 0" contype="1" conaffinity="0" />
              <joint name="Z1_linear_joint" type="slide" axis="0 0 -1" pos="0 0 0" damping="0.1" frictionloss="0.1" limited="true" range="0 0.276" />
              <body name="Z2_linear" pos="0 0 0">
                <inertial diaginertia="1e-4 1e-4 1e-4" mass="1e-2" pos="-0.01 0 -0.25" />
                <geom name="Z2_linear" type="mesh" mesh="Z2_linear" pos="0 0 0" contype="1" conaffinity="0" />
[... cut ...]
    <position name="joint2_position" joint="joint2_joint" ctrllimited="true" ctrlrange="-2.01 2.01" forcelimited="true" forcerange="-25 25" />
    <position name="joint3_position" joint="joint3_joint" ctrllimited="true" ctrlrange="-2.01 2.01" forcelimited="true" forcerange="-25 25" />
    <position name="joint4_position" joint="joint4_joint" ctrllimited="true" ctrlrange="-2.01 2.01" forcelimited="true" forcerange="-25 25" />
  </actuator>
<tendon /></mujoco>`

I hope this provides more clarity on the issue. I look forward to any suggestions or insights you might have regarding the memory allocation and how to optimize it.

kankanzheli, 2024-01-03:
Thank you for your response. Upon reviewing my XML file, I realized that it contains 2276 cubes. Further inspection led me to discover that it's likely other components in the XML file, not related to the cubes, that are contributing to the increased memory usage. These components are parts of the robot.

Below, I've included a more detailed snippet of the XML code for your reference and a better understanding of the structure.
I hope this provides more clarity on the issue. I look forward to any suggestions or insights you might have regarding the memory allocation and how to optimize it.

`<mujoco>
  <size nstack="300000000" /> 
  <option gravity="0 0 0" />
  <asset>
    <mesh name="frame" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/frame.stl" />
    <mesh name="wheel" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/wheel.stl" />
    <mesh name="Y_linear" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/Y_linear.stl" />
    <mesh name="X_linear" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/X_linear.stl" />
    <mesh name="Z1_linear" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/Z1_linear.stl" />
    <mesh name="Z2_linear" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/Z2_linear.stl" />
    <mesh name="Z3_linear" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/Z3_linear.stl" />
    <mesh name="yaw" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/yaw.stl" />
    <mesh name="pitch" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/pitch.stl" />
    <mesh name="subarm_base" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/subarm_base.STL" />
    <mesh name="joint1" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/joint1.STL" />
    <mesh name="joint2" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/joint2.STL" />
    <mesh name="joint3" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/joint3.STL" />
    <mesh name="joint4" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/joint4.STL" />
    <mesh name="harvest_tool" file="/home/ohya_student/Reinforcement learning/agricultural_robot/STL/harvest_tool.STL" />
  </asset>
  <worldbody>
    <body name="base_body">
      
      <body name="frame">
        <inertial diaginertia="222.35 143.37 209.67" mass="1e-2" pos="0 0 0.66" />
        
        <geom name="frame" type="mesh" mesh="frame" pos="0 0 0" contype="0" conaffinity="0" />
          
        <body name="wheel_rr" pos="-0.894 -0.633 0.206">
          <inertial diaginertia="0.1 0.18 0.1" mass="1e-2" pos="0 0 0" />
          <geom name="wheel_rr" type="mesh" mesh="wheel" pos="0 0 0" euler="0 0 3.14159" contype="0" conaffinity="0" />
          <joint name="wheel_rr_joint" type="hinge" axis="0 -1 0" limited="false" damping="0.1" frictionloss="0.1" />
        </body>
        <body name="wheel_lr" pos="-0.894 0.633 0.206">
          <inertial diaginertia="0.1 0.18 0.1" mass="1e-2" pos="0 0 0" />
          <geom name="wheel_lr" type="mesh" mesh="wheel" pos="0 0 0" euler="0 0 3.14159" contype="0" conaffinity="0" />
          <joint name="wheel_lr_joint" type="hinge" axis="0 -1 0" limited="false" damping="0.1" frictionloss="0.1" />
        </body>
        <body name="wheel_rf" pos="0.894 -0.633 0.206">
          <inertial diaginertia="0.1 0.18 0.1" mass="1e-2" pos="0 0 0" />
          <geom name="wheel_rf" type="mesh" mesh="wheel" pos="0 0 0" euler="0 0 0" contype="0" conaffinity="0" />
          <joint name="wheel_rf_joint" type="hinge" axis="0 1 0" limited="false" damping="0.1" frictionloss="0.1" />
        </body>
        <body name="wheel_lf" pos="0.894 0.633 0.206">
          <inertial diaginertia="0.1 0.18 0.1" mass="1e-2" pos="0 0 0" />
          <geom name="wheel_lf" type="mesh" mesh="wheel" pos="0 0 0" euler="0 0 0" contype="0" conaffinity="0" />
          <joint name="wheel_lf_joint" type="hinge" axis="0 1 0" limited="false" damping="0.1" frictionloss="0.1" />
        </body>
        <body name="Y_linear" pos="0 0 1.137">
          <inertial diaginertia="1e-4 1e-4 1e-4" mass="1e-2" pos="0.01 -0.02 -0.1" />
          <geom name="Y_linear" type="mesh" mesh="Y_linear" pos="0 0 0" contype="1" conaffinity="0" />
          <joint name="Y_linear_joint" type="slide" axis="0 -1 0" pos="0 0 0" damping="0.1" frictionloss="0.1" limited="true" range="-0.3455 0.3455" />
          <body name="X_linear" pos="0 0 -0.002">
            <inertial diaginertia="1e-4 1e-4 1e-4" mass="1e-2" pos="-0.04 -0.01 -0.2" />
            <geom name="X_linear" type="mesh" mesh="X_linear" pos="0 0 0" contype="1" conaffinity="0" />
            <joint name="X_linear_joint" type="slide" axis="1 0 0" pos="0 0 0" damping="0.1" frictionloss="0.1" limited="true" range="-0.225 0.281" />
            <body name="Z1_linear" pos="0 0 0">
              <inertial diaginertia="1e-4 1e-4 1e-4" mass="1e-2" pos="-0.04 0 -0.24" />
              <geom name="Z1_linear" type="mesh" mesh="Z1_linear" pos="0 0 0" contype="1" conaffinity="0" />
              <joint name="Z1_linear_joint" type="slide" axis="0 0 -1" pos="0 0 0" damping="0.1" frictionloss="0.1" limited="true" range="0 0.276" />
              <body name="Z2_linear" pos="0 0 0">
                <inertial diaginertia="1e-4 1e-4 1e-4" mass="1e-2" pos="-0.01 0 -0.25" />
[... cut ...]
                              <geom name="joint4" type="mesh" mesh="joint4" pos="0 0 0" contype="1" conaffinity="0" />
                              <joint name="joint4_joint" type="hinge" pos="0 0 0" axis="0 0 1" damping="0.1" armature="0.01" frictionloss="0.1" limited="true" range="-115 115" />
                            </body>
                          </body>
                        </body>
                      </body>
                    </body>
                  </body>
                </body>
              </body>
            </body>
          </body>
        </body>
      </body>
    </body>`

kankanzheli, 2024-01-03:
`  <body name="target">
      
      <geom name="target" type="sphere" size="0.02" rgba="1 0 0 1" pos="0 0 0" contype="2" conaffinity="1" />
      <joint name="target" type="free" damping="10" />
    </body>
    <body name="voxel_790">
      <geom type="box" size="0.005 0.005 0.005" rgba="0.098 0.149 0.141 1" pos="0.16 0.07 0.41" contype="2" conaffinity="1" />
      <joint name="voxel_790" type="free" damping="0.1" />
    </body>
    <body name="voxel_791">
      <geom type="box" size="0.005 0.005 0.005" rgba="0.098 0.149 0.141 1" pos="0.17 0.07 0.41" contype="2" conaffinity="1" />
      <joint name="voxel_791" type="free" damping="0.1" />
    </body>
........2276 cubes`

and

`  
  <actuator>
    <motor name="wheel_rr_joint_motor" joint="wheel_rr_joint" ctrllimited="true" ctrlrange="-3.9 3.9" forcelimited="true" forcerange="-450 450" />
    <motor name="wheel_lr_joint_motor" joint="wheel_lr_joint" ctrllimited="true" ctrlrange="-3.9 3.9" forcelimited="true" forcerange="-450 450" />
    <motor name="wheel_rf_joint_motor" joint="wheel_rf_joint" ctrllimited="true" ctrlrange="-3.9 3.9" forcelimited="true" forcerange="-450 450" />
    <motor name="wheel_lf_joint_motor" joint="wheel_lf_joint" ctrllimited="true" ctrlrange="-3.9 3.9" forcelimited="true" forcerange="-450 450" />
    <position name="Y_linear_position" joint="Y_linear_joint" ctrllimited="true" ctrlrange="-0.3455 0.3455" forcelimited="true" forcerange="-230 230" />
    <position name="X_linear_position" joint="X_linear_joint" ctrllimited="true" ctrlrange="-0.225 0.281" forcelimited="true" forcerange="-110 110" />
    <position name="Z1_linear_position" joint="Z1_linear_joint" ctrllimited="true" ctrlrange="0 0.276" forcelimited="true" forcerange="-300 300" />
    <position name="Z2_linear_position" joint="Z2_linear_joint" ctrllimited="true" ctrlrange="0 0.325" forcelimited="true" forcerange="-300 300" />
    <position name="Z3_linear_position" joint="Z3_linear_joint" ctrllimited="true" ctrlrange="0 0.207" forcelimited="true" forcerange="-180 180" />
    <position name="yaw_position" joint="yaw_joint" ctrllimited="true" ctrlrange="-3.14159 3.14159" forcelimited="true" forcerange="-1.7 1.7" />
    <position name="pitch_position" joint="pitch_joint" ctrllimited="true" ctrlrange="-3.14159 0.523599" forcelimited="true" forcerange="-25 25" />
    <position name="joint1_position" joint="joint1_joint" ctrllimited="true" ctrlrange="-2.01 2.01" forcelimited="true" forcerange="-25 25" />
    <position name="joint2_position" joint="joint2_joint" ctrllimited="true" ctrlrange="-2.01 2.01" forcelimited="true" forcerange="-25 25" />
    <position name="joint3_position" joint="joint3_joint" ctrllimited="true" ctrlrange="-2.01 2.01" forcelimited="true" forcerange="-25 25" />
    <position name="joint4_position" joint="joint4_joint" ctrllimited="true" ctrlrange="-2.01 2.01" forcelimited="true" forcerange="-25 25" />
  </actuator>
<tendon /></mujoco>`

kevinzakka, 2024-01-03:
@kankanzheli If you can attach a zip file with a fully loadable XML model, that would make debugging this much easier.

yuvaltassa, 2024-01-05:
Did you try using `<option solver="CG"/>` ?

Currently the CG solver is the only one which is fully sparsified, this is mentioned in item 6 [here](https://mujoco.readthedocs.io/en/latest/changelog.html#id24)

kankanzheli, 2024-01-11:
I have attached the XML file below. To display the robot model, I need to call the stl file. I simplified the robot model. Each robot component is replaced by an identical rectangular prism (errors still occur).
[1.zip](https://github.com/google-deepmind/mujoco/files/13898837/1.zip)

kankanzheli, 2024-01-12:
Using <option solver="CG"/> solved my problem

kankanzheli, 2024-01-12:
Using option solver="CG" sloved my problem

closed 2024-01-12 by kankanzheli