google-deepmind/mujoco #1302: Help in finding the right qpos and ctrl values while defining the keyframe

rs545837, 2024-01-01:
I have been getting errors like 
```
ValueError: Error: key 0: invalid qpos size, expected length 35
Object name = standing, id = 0
```
I am trying to understand the correct meaning of qpos and ctrl for models. I am trying to train my model using MJX. When I was reading the MJX tutorial - [here](https://colab.research.google.com/github/google-deepmind/mujoco/blob/main/mjx/tutorial.ipynb#scrollTo=cHJCbESGA7Rk), I was confused when qpos was also used to save the joint angles for the Barkour v0 Quadruped, and was being further used. Can someone try to elaborate these attributes a little more?

nimrod-gileadi, 2024-01-02:
qpos is an array that holds all the joint position values for all joints in the model.

For normal robot joints (e.g. hinges, sliders), there's one entry per joint. For free joints (the "joint" between a free floating object and the world), there are 7 entries: 3 for the position and 4 for the quaternion representing the object's orientation, and the order in the array is the same order that they're specified in the MuJoCo model XML.

ctrl is an array that holds the control commands for the robot. The length of the array is the number of actuators in the model.

See the [Computation](https://mujoco.readthedocs.io/en/stable/computation/index.html#general-framework) chapter in the docs.

rs545837, 2024-01-04:
Is there a way to access the actuators directly or I have to access it through mjData.joint(i).name, where i is the index for actuators. 
I didn't find anything about just accessing values like qpos, qvel, xpos etc. for just actuators.

yuvaltassa, 2024-01-05:
Your question doesn't make sense. Actuators are not directly associated with dofs. For example an actuator can be attached to a tendon that will pull on multiple joints. Certainly an actuator doesn't have a position.

Try to say what you are trying to achieve, rather than where you got stuck trying to achieve it.

rs545837, 2024-01-05:
I am sorry for my stupid way of asking. I can very well see I was not clear in the question. So basically I have two questions as follows:

**First One**
`qpos` stores the value of joint position or joint angles.
I want to define functions that can access joint positions and joint angles, so how to do that.
[image]

**Second One**
I just noticed that the [tutorial.py](https://colab.research.google.com/github/google-deepmind/mujoco/blob/main/mjx/tutorial.ipynb#scrollTo=BQDG6NQ1CbZD) for MJX got changed. I was basically trying to make some functions in the MjxEnv class according to the prev tutorial, that returns things like joint_torques, joint_positions, joint_velocities but for only actuated joints. For accessing those particular joints, I used an array of all the actuated joints.
[image]
Then I am trying to return these stored values as:
[image]

My question was, we know that information about control commands is stored in ctrl, so is there also an option of array as well where the actuated joints are directly stored, and I can find info just about them, or is this the only way.

yuvaltassa, 2024-01-05:
`model.actuator("actuator_name" or actuator_id).trnid[0]` will give you the id of the joint actuated by this actuator, assuming it is connected to a joint. I hope this answers your question.

maheshr24, 2024-08-23:
@yuvaltassa Hey I'm solving IK for a robotic arm but when I input as data.qpos this doesnt work I guess the input should be with data.ctrl how do I map qpos to ctrl?