google-deepmind/mujoco #1304: Difference between the PD control and Model based control

rishabkr29, 2024-01-01:
Hi,

I'm a student  and I'm trying to use MuJoCo for Thesis.

I'm looking for some help with type of control. I saw this youtube video https://www.youtube.com/watch?v=6B_NDL0ff1c which explain Pd control and model-based control. what type of error do pd control encounter ? How to find the gain values for pd control ?

Balint-H, 2024-01-02:
I recommend looking at tutorial papers and textbooks on the topic (such as chapter 10 in [this book](https://www.cds.caltech.edu/~murray/books/AM08/pdf/am08-complete_22Feb09.pdf), or [this book for a human movement centric discussion](https://direct-mit-edu.iclibezp1.cc.ic.ac.uk/books/book/3988/Human-RoboticsNeuromechanics-and-Motor-Control)).

A short oversimplified response: 
You can get the current state of the controlled system (e.g. the joint angles), and its rate of change (or velocity). You can compare this to some desired state and velocity to get an error signal. You can then simply apply a response that's linearly proportional to the magnitude of this error, that applies a change in the system that reduces the error. The ideal parameters will depend a lot on the details of the system. For example, if the object that you are moving is heavier, you should have a stronger response (with greater proportional and derivative gains) if you want to move as fast and responsive as when moving a lighter object. You can try adjusting/tuning the gains until you get the behaviour you are looking for. 

Alternatively, to tailor the behaviour of your system it is operating, you can use information available from your model of the system (for example the mass matrix, which is readily available in MuJoCo). This can help achieve optimality in cases where simple PD would fail (e.g. minimizing energy usage, etc).

rishabkr29, 2024-01-02:
Thanks @Balint-H for quick overview of the PID control, 

I am interested in the error of the singles when I am using direct PID. By saying direct PID , I mean you are having some reference state signal and you are taking actual state signal form the model and by using the PID controller with the error signals you are giving it the actuator ofcourse with the some gains (in my case : 2R cable-driven model). Here my curiosity lies is what kind of error does it will encounter ? I understand it will depend on various details of the system. Is there any ideal way to validate the stability of the system using Root locus, or Nyquist ( at the same time I understand it is a highly non-linear system) but If I consider the Kinematic Control only.

rishabkr29, 2024-01-02:
@kevinzakka #1303 have been referred here. I am still not convince the theoretical way to compute the gains. I understand the intuitive way to tuned it, but if you are directly using PD control and then what will be the **Plant** equation being used here ? I have done with it but need to prove the stability of the model, for which I need the plant equation.

rishabkr29, 2024-01-03:
@Balint-H I am looking for state Identification of actuator. Basically I wanted to Know what is the low level control of the actuator, how the control input singles are being used by the MuJoCokl, what are the governing equation for it. I am curious to write the transfer function.

Balint-H, 2024-01-03:
For details on the actuation model, you can find info in the modelling section of the documentation [[here](https://mujoco.readthedocs.io/en/stable/computation/index.html#actuation-model)], and [[here](https://mujoco.readthedocs.io/en/stable/modeling.html#actuators)]. 

To identify a transfer function of a system / plant that is nonlinear, I don't have too much experience. My first instinct would be to systematically sweep a range of actuation frequencies and amplitudes, then fit a transfer function on the recorded frequency response behaviour. However, I wouldn't be surprised if that is a wasteful approach, and there's a better/analytical method to this end. In particular, you could use the Jacobian, which MuJoCo can calculate for you, and assume linearization around a point, in which case the system/plant TF can approximated as a simple amplifier (**Please someone correct me if I'm wrong!**).