google-deepmind/mujoco #1307: Undersanding auto connect feature

vikashplus, 2024-01-02:
I'm curious about what the auto-connect feature draws. I thought it was the kinematic chain. But I often see things are a bit off. 

For example, consider the feet of the humanoid model. The connects are in transitioning through points that are neighter the body frames, nor the joint positions. 

[image]
[image]

kevinzakka, 2024-01-02:
It connects joints and body coms as mentioned [here](https://github.com/google-deepmind/mujoco/blob/1c7809e54261009698b5baab569faf700c8f8dad/include/mujoco/mjvisualize.h#L123).

vikashplus, 2024-01-02:
Do you understand, why it is conceptually useful? 

I'm not sure it's implemented properly, even at its face value. For example 
   1. The connect shooting back from the ankle in the humanoid is neither the joint-location nor the body COM.
   2. Below is another model with connect having weird patterns at the knee, elbow, and something that shoots up in the sky from the torso. Again these aren't joints or COMs.
[image]

kevinzakka, 2024-01-03:
I modified the visualization code to plot all body COMs along with the joints and autoconnect. I get the below for `model/humanoid/humanoid.xml`:

[image]

I would encourage you to plot the body CoMs in your model, you may have weird inertial values?

kevinzakka, 2024-01-03:
@vikashplus I just [inspected the code](https://github.com/google-deepmind/mujoco/blob/main/src/engine/engine_vis_visualize.c) and it looks like when inertias are turned on, [the body frame visualization will switch from xpos to xipos](https://github.com/google-deepmind/mujoco/blob/main/src/engine/engine_vis_visualize.c#L904-L905) so no need to visualize the spheres as I did. So just redo your inspection with inertias turned on (hit I). Might be hard to see with all the red but it should confirm that autoconnect is correctly drawn.

closed 2024-01-03 by kevinzakka

vikashplus, 2024-01-04:
It gets confused for bodies with no geoms. 
I still don't understand what's the use of this visualization. Won't it be more useful if it was rendering just the kinematic chain?

yuvaltassa, 2024-01-05:
@vikashplus the code is [right here](https://github.com/google-deepmind/mujoco/blob/721e2d5589d3fdafd440009374a31521214088b7/src/engine/engine_vis_visualize.c#L1910), how do you want to change it? If you think there is something better just modify the code and send us a PR. This is very easy, I know you can do it 🙂 

If that's too hard, attach a minimal model that doesn't do what you expect and we can fix it.