google-deepmind/mujoco #1309: How to build Simulate app in my own project for windows

flilqhu, 2024-01-04:
Hi,

I'm an engineer and I'm trying to use MuJoCo for robot simulation.

I'm looking for some help with simulate app.

I am not familiar with opengl, the rendering examples on the official website do not have the powerful features in simulate app for rendering ui, but there is no example on how to build an simulate app. For concise is there any example on building an app using only mujoco glfw DLLS and their LiBs, thank you so much.

Balint-H, 2024-01-05:
I believe this tutorial series covers the basics of creating custom GLFW apps with MuJoCo: https://pab47.github.io/mujocopy.html

You can also use the Unity plugin to create apps with additional UI building tools/rendering functionality.

yuvaltassa, 2024-01-05:
The link provided by @Balint-H is pretty old, I'm not sure how relevant it still is.

Regarding building simulate-like apps, simulate itself is such and example, which you can [build from source](https://mujoco.readthedocs.io/en/latest/programming/index.html#building-from-source) and modify as you like.

Another specific example is @thowell's [MJPC](https://github.com/google-deepmind/mujoco_mpc?tab=readme-ov-file#overview) app.

closed 2024-01-05 by yuvaltassa