google-deepmind/mujoco #1310: Non-holonomic link for sphere robot.

nor-code, 2024-01-04:
Hi, @yuvaltassa !
I would like to continue previous not finished our discussion: https://github.com/google-deepmind/mujoco/issues/450

For this robot there are two control moments. The first moment turns the wheel around axis that perpendicular to his plane and passing through its center. The other moment the rotate the wheel relative to the vertical axis (was implemented).

I still haven't been able to implement the following: I want the robot shell to be driven by a wheel. (of course, in this case the wheel with the platform should not turn over.) The platform is sliping on the robot shell.

I don't understand how this implement in mujoco. 
Is this even possible at the moment?
I hope very much on your help

Video (real model ): https://disk.yandex.ru/i/f-ik3ki-pQMxuA

Here is a model which explains my question:
```XML
<mujoco>
    <compiler autolimits="true"/>

    <visual>
      <headlight diffuse="0.6 0.6 0.6"  ambient="0.3 0.3 0.3" specular="0 0 0"/>
      <rgba haze="0.15 0.25 0.35 1"/>
    </visual>

    <asset>
        <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="3072"/>
        <texture type="2d" name="groundplane" builtin="checker" mark="edge" rgb1="0.2 0.3 0.4" rgb2="0.1 0.2 0.3" markrgb="0.8 0.8 0.8" width="300" height="300"/>
        <material name="groundplane" texture="groundplane" texuniform="true" texrepeat="5 5" reflectance="0.2"/>
    </asset>

    <visual>
        <map force="0.1" zfar="30"/>
        <rgba haze="0.15 0.25 0.35 1"/>
        <quality shadowsize="2048"/>
        <global offwidth="800" offheight="800"/>
    </visual>

  <default>
    <joint damping=".02" actuatorfrcrange="-0.5 0.5"/>
    <default class="wheel">
      <geom type="cylinder" size=".04 .01" rgba=".5 .5 1 1"/>
    </default>
    <default class="decor">
      <site type="box" rgba=".5 1 .5 1"/>
    </default>
  </default>
    
  <worldbody>
      <light pos="0 0 1.5" dir="0 0 -1" directional="true"/>
      <geom name="floor" size="0 0 0.25" type="plane" material="groundplane" condim="3"/>
      <camera name="fixed" pos="0 0 1" zaxis="0 -1 0"/>

      <light pos="0 0 6"/>
      <light directional="false" diffuse=".2 .2 .2" specular="0 0 0" pos="0 0 5" dir="0 0 -1" castshadow="false"/>

      <body name="sphere_shell">
          <freejoint/>
          <geom type="sphere" size=".181 .18"  pos="0 0 .18" rgba=".0 .0 .0 .2"/>
          
          <body name="skeleton" pos="0 0 .04">
              <!--<joint name="ball" type="ball" frictionloss=".15"/> -->
              <geom name="wheel_axis" pos="0 0 0" type="cylinder" size=".005 .03" zaxis="0 1 0"  contype="0" conaffinity="0"/>
              <geom name="fork_one" pos="0 .03 .03" type="cylinder" size=".005 .035"/>
              <geom name="fork_two" pos="0 -.03 .03" type="cylinder" size=".005 .035"/>
              <geom name="wheel_axis_top" pos="0 0 .06" type="cylinder" size=".005 .03" zaxis="0 1 0"/>
              <geom name="rotor" pos="0 0 .075" type="cylinder" size=".005 .01"/>

              <body name="robot_platform" pos="0 0 .085">
                  <joint name="right"/>
                  <geom name="platform" type="cylinder" size=".15 .005" rgba=".0 .0 .3 .5"/>
                  <site class="decor" pos=".10 0 .005" size=".006 .025 .001"/>
              </body>

              <body name="wheel"  zaxis="0 1 0">
                  <joint name="left" />
                  <geom class="wheel"/>
                  <site class="decor" size=".006 .025 .012"/>
                  <site class="decor" size=".025 .006 .012"/>
              </body>
            </body>
      </body>

  </worldbody>

   <!-- <tendon>
        <fixed name="forward">
          <joint joint="left" coef=".5"/>
          <joint joint="ball" coef="-.5"/>>
        </fixed>
    </tendon> -->

  <!--<equality> not working  
	<joint joint1="ball" joint2="left" polycoef="0 1 1 0 0"/>
  </equality>-->

  <actuator>
	<motor name="plat" joint="right" ctrlrange="-1 1"/>
	<motor name="wheel" joint="left" ctrlrange="-1 1"/>
  </actuator>
</mujoco>
```

yuvaltassa, 2024-01-05:
Yes, this robot can certainly be modeled, and you are on the right track.

The issue is that you have the two internal joints in parallel, but they need to be serial. You want something like:
```
<body (spherical shell with freejoint)>
  <body (ballast the makes the wheel point down)>
    <joint (roll)/>
    <joint (turn)/>
```
Maybe the order of roll and turn needs to be reversed, I'm not sure.

Try playing with this and see how far you get. If you fail i can try and help.

nor-code, 2024-01-05:
> Yes, this robot can certainly be modeled, and you are on the right track.
>
> The issue is that you have the two internal joints in parallel, but they need to be serial. You want something like:
>
> ``` [...]

I tried simplified my problem. Find one solution, but as you can see on [video](https://disk.yandex.ru/i/l4oqM9ZIoby5nA) (start from 5s) wheel not turn relative his axes.

```
<mujoco>
    <compiler autolimits="true"/>

    <visual>
      <headlight diffuse="0.6 0.6 0.6"  ambient="0.3 0.3 0.3" specular="0 0 0"/>
      <rgba haze="0.15 0.25 0.35 1"/>
    </visual>

    <asset>
        <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="3072"/>
        <texture type="2d" name="groundplane" builtin="checker" mark="edge" rgb1="0.2 0.3 0.4" rgb2="0.1 0.2 0.3" markrgb="0.8 0.8 0.8" width="300" height="300"/>
        <material name="groundplane" texture="groundplane" texuniform="true" texrepeat="5 5" reflectance="0.2"/>
    </asset>

    <visual>
        <map force="0.1" zfar="30"/>
        <rgba haze="0.15 0.25 0.35 1"/>
        <quality shadowsize="2048"/>
        <global offwidth="800" offheight="800"/>
    </visual>

  <default>
    <joint damping=".02"/>
    <default class="wheel">
      <geom type="cylinder" size=".04 .01" rgba=".5 .5 1 1"/>
    </default>
    <default class="decor">
      <site type="box" rgba=".5 1 .5 1"/>
    </default>
  </default>

  <worldbody>
      <light pos="0 0 1.5" dir="0 0 -1" directional="true"/>
      <geom name="floor" size="0 0 0.25" type="plane" material="groundplane" condim="3"/>
      <camera name="fixed" pos="0 0 1" zaxis="0 -1 0"/>

      <light pos="0 0 6"/>
      <light directional="false" diffuse=".2 .2 .2" specular="0 0 0" pos="0 0 5" dir="0 0 -1" castshadow="false"/>

        <body name="shell" pos="0 0 0" euler="0 0 0">
            <joint name="shell_floor" type="free"/>
            <geom name="sphere_shell" type="sphere" pos="0 0 0" size=".2 .19" rgba=".0 .0 .0 .2" mass="0.2" friction="1 5 1"/>

            <body name="wheel">
                <joint name="left" type="ball" pos="0 0 0" frictionloss="0.0068"/>
                <geom name="wheel_" type="cylinder" fromto="-0.008 0 -0.15  0.008 0 -0.15" size="0.049 0.005" mass="0.7"/>
        	    <site name="decor_wheel" class="decor" fromto="-0.008 0 -0.15  0.008 0 -0.15" size=".025 .005 .012"/>
            </body>
      </body>

  </worldbody>

  <actuator>
	<motor name="wheel" gear="2" joint="left" ctrlrange="-1 1"/>
  </actuator>
</mujoco>
```

yuvaltassa, 2024-01-08:
That's expected. The wheel does not contact the sphere from the inside, it is merely the changing of the position of the com that makes it the whole thing roll. If you like, you can add another body with a joint at the wheel itself, and couple the joint you called "left" (why?) with this joint using and equality constraint, so that it looks like it is rolling on the inside. But that would be purely visual.

nor-code, 2024-01-08:
> That's expected. The wheel does not contact the sphere from the inside, it is merely the changing of the position of the com that makes it the whole thing roll. If you like, you can add another body [...]

Don't pay attention to this title ("left"). Why doesn't the wheel touch the shell from the inside? I connected them ` <joint name="left" type="ball" pos="0 0 0" frictionloss="0.0068"/>`. Tell me please how to do this then, so that there is not a visual effect. I've already spent a lot of time, but I couldn't do it the way I needed.

yuvaltassa, 2024-01-08:
`frictionloss` is joint friction, has nothing to do with contact.

Unless you use SDFs (which will be non trivial), you cannot collide with something that is concave. But why would you want to? Why do you care if the motion is generated by friction or not? The point is that the motion of the little wheel is coupled to the motion of the arm, which turns the sphere. This can be easily achieved, does it matter how?

nor-code, 2024-01-08:
> `frictionloss` is joint friction, has nothing to do with contact.
>
> Unless you use SDFs (which will be non trivial), you cannot collide with something that is concave. But why would you want to? Why [...]

Yes, actually it is important that the wheel rotates the spherical shell through their point of contact. And as i understand this possible in case, when i implement my SDF class (aka MySphericalShell)  in c++ ? Is there no other way?

yuvaltassa, 2024-01-09:
> Yes, actually it is important that the wheel rotates the spherical shell through their point of contact.

Why do you think this is the case? I have to tentatively disagree here. If the motion is the same then it doesn't matter if the force is generated by friction or a constraint, unless there is internal slipping of the contact point which you want to model, and I suspect this is not the case.

yuvaltassa, 2024-01-09:
Here, I made you an example model (also fixing various issues with your XML). I hope you can take it from here. Closing for now.

<details>
  <summary>example model</summary>

```XML
<mujoco>
  <visual>
    <headlight diffuse="0.6 0.6 0.6"  ambient="0.3 0.3 0.3" specular="0 0 0"/>
    <rgba haze="0.15 0.25 0.35 1"/>
  </visual>

  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="3072"/>
    <texture type="2d" name="groundplane" builtin="checker" mark="edge" rgb1="0.2 0.3 0.4" rgb2="0.1 0.2 0.3" markrgb="0.8 0.8 0.8" width="300" height="300"/>
    <material name="groundplane" texture="groundplane" texuniform="true" texrepeat="5 5" reflectance="0.2"/>
  </asset>

  <visual>
    <map force="0.1" zfar="30"/>
    <rgba haze="0.15 0.25 0.35 1"/>
    <quality shadowsize="2048"/>
    <global offwidth="800" offheight="800"/>
  </visual>

  <default>
    <joint damping=".01"/>

    <default class="decor">
      <site type="box" rgba=".5 1 .5 1"/>
    </default>
    <default class="internal">
      <geom contype="0" conaffinity="0"/>
    </default>
  </default>

  <worldbody>
    <light pos="0 0 1.5" dir="0 0 -1" directional="true"/>
    <geom name="floor" size="0 0 0.25" type="plane" material="groundplane" condim="3"/>
    <camera name="fixed" pos="0 0 1" zaxis="0 -1 0"/>

    <light pos="0 0 6"/>
    <light directional="false" diffuse=".2 .2 .2" specular="0 0 0" pos="0 0 5" dir="0 0 -1" castshadow="false"/>

    <body name="sphere" pos="0 0 .18">
      <freejoint/>
      <geom type="sphere" size=".181"  rgba=".0 .0 .0 .2"/>

      <body name="skeleton" pos="0 0 -.14" childclass="internal">
        <joint name="turn" axis="0 0 1" pos="0 0 .14"/>
        <joint name="roll" axis="0 1 0" pos="0 0 .14"/>
        <geom name="platform" type="cylinder" size=".15 .005" rgba=".0 .0 .3 .5" pos="0 0 .085"/>
        <site class="decor" pos=".10 0 .09" size=".006 .025 .001"/>
        <geom name="wheel_axis" type="cylinder" size=".005 .03" zaxis="0 1 0"  contype="0" conaffinity="0"/>
        <geom name="fork_one" pos="0 .03 .03" type="cylinder" size=".005 .035"/>
        <geom name="fork_two" pos="0 -.03 .03" type="cylinder" size=".005 .035"/>
        <geom name="wheel_axis_top" pos="0 0 .06" type="cylinder" size=".005 .03" zaxis="0 1 0"/>
        <geom name="rotor" pos="0 0 .075" type="cylinder" size=".005 .01"/>

        <body name="wheel" zaxis="0 1 0">
          <joint name="wheel" axis="0 0 -1"/>
          <geom type="cylinder" size=".04 .01" rgba=".5 .5 1 1" density="5000"/>
          <site class="decor" size=".006 .025 .012"/>
          <site class="decor" size=".025 .006 .012"/>
        </body>
      </body>
    </body>
  </worldbody>

  <equality>
    <joint joint1="wheel" joint2="roll" polycoef="0 4.525 0 0 0"/>
  </equality>

  <actuator>
    <motor name="turn" joint="turn" ctrlrange="-1 1" gear="0.1"/>
    <motor name="roll" joint="roll" ctrlrange="-1 1" gear="0.5"/>
  </actuator>
</mujoco>
```
</details>

closed 2024-01-09 by yuvaltassa

nor-code, 2024-01-12:
> Here, I made you an example model (also fixing various issues with your XML). I hope you can take it from here. Closing for now.
> 
> example model

Thanks!

LinShuang22, 2024-02-10:
> Why do you think this is the case? I have to tentatively disagree here. If the motion is the same then it doesn't matter if the force is generated by friction or a constraint, unless there is [...]

@yuvaltassa Excuse me, I'm currently facing an issue. I'm working on a similar spherical robot. However, my spherical robot is embedded in a hollow spherical shell, achieving steering through differential drive with different wheel speeds. Can this be implemented without using internal contacts, or is it necessary to model it using SDF?
It looks like this.
[image]

yuvaltassa, 2024-02-10:
I would avoid internal contacts if possible. I don't think this should require them. Imagine that you have two internal joints, one for roll and one for yaw. The sum of your wheel rotations should correspond to roll, while the difference should correspond to yaw, right? You just need to figure out the right coefficients to do the conversion.

LinShuang22, 2024-02-10:
> I would avoid internal contacts if possible. I don't think this should require them. Imagine that you have two internal joints, one for roll and one for yaw. The sum of your wheel rotations should [...]

@yuvaltassa Thank you! I understand what you mean. I will try this. Where should I add configurations for these parameters? <equality>?

LinShuang22, 2024-02-10:
> I would avoid internal contacts if possible. I don't think this should require them. Imagine that you have two internal joints, one for roll and one for yaw. The sum of your wheel rotations should [...]

@yuvaltassa Thank you! I understand what you mean. I will try this. Where should I add configurations for these parameters? \<equality\>? or anything else?

LinShuang22, 2024-02-10:
> I would avoid internal contacts if possible. I don't think this should require them. Imagine that you have two internal joints, one for roll and one for yaw. The sum of your wheel rotations should [...]

@yuvaltassa Thank you! I understand what you mean. I will try this. Where should I add configurations for these parameters? 
How do I relate differential drive to yaw in mujoco model? <equality>? or anything else?

LinShuang22, 2024-02-10:
> I would avoid internal contacts if possible. I don't think this should require them. Imagine that you have two internal joints, one for roll and one for yaw. The sum of your wheel rotations should [...]

@yuvaltassa Thank you! I understand what you mean. I will try this. Where should I add configurations for these parameters?
How do I relate differential drive to yaw in mujoco model?  \<equality\>? or anything else?

yuvaltassa, 2024-02-11:
You want to make 4 tendons:

1. Just the yaw joint.
2. Just the roll joint.
3. Sum of wheels.
4. Difference of wheels.

Then two equalities that couple 1-4 and 2-3. When the coefficients are correct and you actuate 1 or 2 you should see the wheels roll inside as if they are contacting with no slip.

LinShuang22, 2024-02-11:
> You want to make 4 tendons:
>
>     1. Just the yaw joint.
>
>     2. Just the roll joint.
>
>     3. Sum of wheels.
>
>     4. Difference of wheels.
>
>
> Then two equalities that couple 1-4 and 2-3. When the [...]

@yuvaltassa Thank you very much!!!