google-deepmind/mujoco #1312: Possible

TenguBoss, 2024-01-04:
Dear Colleagues,

I am using MuJoCo 3.1.1 in WSL,2 with Ubuntu 22.04.3 LTS. I have just cloned the MuJoCo Menagarie. I have included the model for Trossen VX300s in one of my own models. As required I have used the relative path like so

<include file="../../mujoco_menagerie/trossen_vx300s/vx300s.xml"/>

When I run

$~/mujoco-3.1.1/bin/simulate ~/hil_lab/slider_bots/beam_slider.xml

I get this error

Error: resource not found via provider or OS filesystem: '/home/xxx/hil_lab/slider_bots/assets/vx300s_1_base.stl'

I have fixed the problem by specifying the meshdir and texturedir with relative paths as well, in the top XML file, beam_slider.xml

<compiler meshdir="../../mujoco_menagerie/trossen_vx300s/assets"/>
<compiler texturedir="../../mujoco_menagerie/trossen_vx300s/assets"/>

Is the error above an expected behavior when one uses <include .../> due to my beginner's mistake of not properly specifying resource paths?

Thank you.

BU

rs545837, 2024-01-04:
```Error: resource not found via provider or OS filesystem:```
This error is because the mujoco viewer is not able to locate the mesh files. You can solve this in two ways:
- Add meshdir option in the compiler tag of xml file, and mention the path to the directory containing the mesh files

Or

- Use relative path directly of the files in the `<mesh>` under the `<asset>`
- Like this
[image]

TenguBoss, 2024-01-04:
@rs545837, thank you for your suggestions. I wasn't aware of the second option. 

However, I wonder if the MoJoCo documentation for `<include>` should mention that `meshdir` and `texturedir` are the responsibility of the user for the included model files. I have mistakenly assumed that simulate is aware of the folder structure specified in the included xml files.
BU

yuvaltassa, 2024-01-05:
Ya, <include> is generally a pretty shoddy mechanism, we are working on a more powerful replacement. In the meantime, the above can serve as temporary documentation. Thanks @rs545837

closed 2024-01-05 by yuvaltassa

TenguBoss, 2024-01-05:
Thank you @yuvaltassa Not sure if it matters, but I prefer to specify full paths to access resources.