google-deepmind/mujoco #1313: Problem implementing MJX environment from xml (NotImplementedError)

rs545837, 2024-01-04:
This problem is also discussed in the issue #1142. I have been facing same error i.e. ```NotImplementedError: {2} not implemented.``` but due to presence of an equality constraint in my xml file. I don't see anything about equality constraint in the unsupported table of [Feature Parity](https://mujoco.readthedocs.io/en/latest/mjx.html#feature-parity) section of MuJoCo. 

The part in xml which raises that error:
```python3
  <!-- The equality joint below raises:  NotImplementedError(f'{unsupported_typs} not implemented.'-->
  <equality>
          <joint solref="0.0025 1" polycoef="0 -1 0 0 0" joint1="BodyRight_Joint" joint2="BodyTopleft_Joint" name="equality_constraint4"/>
          <joint solref="0.0025 1" polycoef="0 1 0 0 0" joint1="BodyBot_Joint" joint2="BodyTopleft_Joint" name="equality_constraint5"/>
  </equality>
```
It might be possible that info about this is missing in the table or maybe i am missing something.

btaba, 2024-01-04:
Hi @rs545837 what's the full error you're getting with traceback? Can you also paste which version of MuJoCo/MJX you're using?

rs545837, 2024-01-04:
Hey @btaba, it's
mujoco==3.0.0
mujoco_mjx==3.0.0 

```
Traceback (most recent call last):
  File "/Users/rohansharma03/Desktop/SCALER_Mujoco/SCALER_XML/RL_Training/file2.py", line 455, in <module>
    env = envs.get_environment(env_name)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/brax/envs/__init__.py", line 63, in get_environment
    return _envs[env_name](**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/rohansharma03/Desktop/SCALER_Mujoco/SCALER_XML/RL_Training/file2.py", line 120, in __init__
    super().__init__(mj_model=mj_model, **kwargs)
  File "/Users/rohansharma03/Desktop/SCALER_Mujoco/SCALER_XML/RL_Training/file1.py", line 70, in __init__
    self.sys = mjx.device_put(mj_model)
               ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/mujoco/mjx/_src/device.py", line 186, in device_put
    _validate(value)  # type: ignore
    ^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/mujoco/mjx/_src/device.py", line 120, in _validate
    raise NotImplementedError(f'{unsupported_typs} not implemented.')
NotImplementedError: {2} not implemented.
```

btaba, 2024-01-04:
Hey @rs545837 , can you try the latest versions 3.1.1. I suspect the issue may be fixed with the latest versions

rs545837, 2024-01-04:
Yup, thanks @btaba, it seems it got resolved in the latest version. My bad that I forgot about the version update.

closed 2024-01-04 by rs545837