google-deepmind/mujoco #1314: Does mj_contactForce apply to contact->geom[0] or contact->geom[1]?

cstegel, 2024-01-04:
Hi,

I'm an engineer and I'm trying to use MuJoCo for analyzing contact forces between objects.

I'm looking for some help with [mj_contactForce](https://mujoco.readthedocs.io/en/stable/APIreference/APIfunctions.html#mj-contactforce). The documentation says it retrieves the force for a given contact but it doesn't say if the direction of this force is for [contact.geom[0]](https://mujoco.readthedocs.io/en/stable/APIreference/APItypes.html?highlight=mjdata#mjcontact) or [contact.geom[1]](https://mujoco.readthedocs.io/en/stable/APIreference/APItypes.html?highlight=mjdata#mjcontact). From experimenting, it looks like this force is always the force that the contact applies to `contact.geom[1]`. To get the force on `contact.geom[0]`, the vector must be inverted.

This matches what I see in the `simulate` sandbox's visualization code ([engine_vis_visualize.c lines 236-291](https://github.com/google-deepmind/mujoco/blob/721e2d5589d3fdafd440009374a31521214088b7/src/engine/engine_vis_visualize.c#L236-L291) for mjVIS_CONTACTFORCE).

Paraphrasing a bit, the code looks like this:

```c
mju_transpose(mat, con->frame, 3, 3);
mj_contactForce(m, d, i, confrc);
mju_copy(frc, confrc, mjMIN(3, con->dim));
mju_mulMatVec(vec, mat, frc, 3, 3);

// make sure arrow points towards body with higher id
if (m->geom_bodyid[con->geom[0]] > m->geom_bodyid[con->geom[1]]) {
  mju_scl3(vec, vec, -1);
}
```

This indicates that [mj_contactForce](https://mujoco.readthedocs.io/en/stable/APIreference/APIfunctions.html#mj-contactforce) gives the contact force to be applied to `geom[1]` but for some reason the visualization forces it to always point towards the body with the higher ID.

Clarification around contact force direction would be nice to have in the docs for [mjContact](https://mujoco.readthedocs.io/en/stable/APIreference/APItypes.html?highlight=mjdata#mjcontact) and [mj_contactForce](https://mujoco.readthedocs.io/en/stable/APIreference/APIfunctions.html#mj-contactforce).

Here is a model that you can turn on contact force visualization and contact split under "rendering" to see the force arrows:

```XML
<mujoco>
  <option
    timestep="0.01"
    integrator="implicitfast"
    gravity="0 0 -9.810"/>

  <!-- Free Camera Positioning -->
  <statistic
    center="0 0 0.5"
  />

  <visual>
    <map force="0.5" />
    <global azimuth="0" elevation="-20"/>
  </visual>

  <worldbody>
    <light pos="0 1 5" dir="0 -1 -2"/>

    <body name="plane" pos="0 0 -0.5">
      <geom type="plane" size="5 5 1"/>
    </body>

    <body name="box" pos="0 0 0.5" >
      <geom type="box" mass="1" size="1 1 1"/>
      <freejoint />
    </body>
  </worldbody>
</mujoco>
```

Here is a screenshot when "box" has the higher ID:

[image]

Here is a screenshot when "box" is moved above "plane" in the XML so it has a lower ID:

[image]

yuvaltassa, 2024-01-05:
Obviously the actual force is always a pair acting on both geoms (Newton's 3rd law), but in the case of one of the geoms being static only the component acting on the moving geom is rendered.

In terms of which of the 2 action/reaction forces is represented in `mjtContact` (and reported by mj_contactForce) I believe the rule is slightly different then what you think it is. [This line](https://github.com/google-deepmind/mujoco/blob/721e2d5589d3fdafd440009374a31521214088b7/src/engine/engine_collision_driver.c#L1461) shows you that the order of the con.geom[0, 1] has to do with the type order in [mjtGeom](https://mujoco.readthedocs.io/en/latest/APIreference/APItypes.html#mjtgeom), and then looking e.g. [this file](https://github.com/google-deepmind/mujoco/blob/main/src/engine/engine_collision_primitive.c), it looks to me like the frame is oriented such that the normal (`con.frame[0-2]`) always points from geom[0] to geom[1].

The actual force that is computed is in that frame and the normal component is never negative.

So I _believe_ the correct rule is 

**The normal direction (frame[0-2]) always points from geom[0] to geom[1], which are sorted according to their mjtGeom enum values. If the two geoms have the same mjtGeom value, then they are sorted according to their respective body indices**.

Do you think you could please confirm or find a counter-example to the above proposed rule? After we're happy that that is is correct, I will update the docs accordingly 🙂

cstegel, 2024-01-05:
Thanks for the quick response!

**I agree that the contact normal direction (frame[0-2]) always points from geom[0] to geom[1]:**

 - `con->frame` is populated by [ccdMPRPenetration](https://github.com/danfis/libccd/blob/7931e764a19ef6b21b443376c699bbc9c6d4fba8/src/ccd/ccd.h#L134-L146) from libccd in [engine_collision_convex.c](https://github.com/google-deepmind/mujoco/blob/feb92bf535683004e85dbf10fa3427c30dbb29d3/src/engine/engine_collision_convex.c#L283). [ccdMPRPenetration](https://github.com/danfis/libccd/blob/7931e764a19ef6b21b443376c699bbc9c6d4fba8/src/ccd/ccd.h#L134-L146) says the returned direction is from obj1 to obj2.
 
 - For plane-box contact, the plane normal is always the contact normal and the mjtGeom type results in box being `geom[1]` like you said

I couldn't find the contact force calculation code to know if the direction is always computed for `geom[0]` or `geom[1]` but experimentally **I've seen that it is always the correct force for geom[1]**:

- For plane-box, plane is `geom[0]` and box is `geom[1]`. Force is for `geom[1]` (points towards box).

- For box-box, `geom[0]` is the lower ID body. Force is for `geom[1]` (points towards the higher ID box).

- For mesh-mesh, `geom[0]` is the lower ID body. Force is for `geom[1]`. When I swap the body ordering but keep the scene the same, the force is for the new `geom[1]`.

I think the docs should describe:

- `frame[0-2]` as pointing from `geom[0]` to `geom[1]`
- Contact force is for `geom[1]`. To get the force for `geom[0]`, it must be scaled by (-1, -1, -1).

I don't think a MuJoCo user needs to know the rules for deciding geom ordering but that's up to you ¯\\\_(ツ)_/¯.

yuvaltassa, 2024-01-08:
I just saw this is [already documented](https://mujoco.readthedocs.io/en/latest/APIreference/APItypes.html#mjcontact):

```
mjtNum  frame[9];                // normal is in [0-2], points from geom[0] to geom[1]
```

Is `points from geom[0] to geom[1]` not clear enough? Closing for now.

closed 2024-01-08 by yuvaltassa

cstegel, 2024-01-08:
Ah, I forgot that `mjContact` already indicates it points from geom[0] to geom[1].

I still think there is missing documentation about which geom the contact force applies to. This is my main concern.

yuvaltassa, 2024-01-09:
But it always applies to both geoms... You want us to explain Newton's 3rd law in the documentation of mjContact?

cstegel, 2024-01-09:
It's reasonable for users to know that the contact force on one geom is the opposite of the contact force on the other geom.

What's not reasonable is knowing which geom needs to have `mj_contactForce` multiplied by (-1, -1, -1).