google-deepmind/mujoco #1316: Simulated air bearings partially "sink" in the floor after I include a ball joint

TenguBoss, 2024-01-05:
**How to ask for help**

Aloha Colleagues,

I'm an  engineer and I use MuJoCo to design and analyze a robot that slides on air bearings with (very) little friction on a (very) flat floor. I am using MiJoCo 3.1.1 on WSL2 Ubuntu Jammy.

The simplest way to explain an air bearing is an "inverted" air hokey table where the air is pumped through the puck towards the table. For detailed explanations I recommend the website of [New Way Air Bearings](https://www.newwayairbearings.com/).

I am new to MuJoCo and the only way I know how to simulate an air bearing is through frictionless contact with the floor. 
```
<body name="AB post -X+Y" pos=" 0.11 0.00 -0.02">
  <geom class="air bearing post"/>
  <!-- ball at bottom of post -->
  <body name="AB ball -X+Y" pos=" 0.00 0.00 -0.02">
    <geom class="air bearing ball"/>
    <!-- spherical joint between ball and pad -->
    <joint type="ball" damping="1e-3"/>
    <body name="AB pad -X+Y" pos="0 0 -0.01">
      <geom class="air bearing pad" condim="1"/>
    </body>
  </body> 
</body> 
```
where the classes are just visualization primitives, cylinders for the post and pad and sphere for the ball.  The `condim="1"` property of the pad geometry is matched by a  `condim="1"` for the floor geometry.

**Notes** 

1. Because I am interested in the static and dynamic and balancing of the robot I cannot use `slide` and `hinge` joints at the pads. They do not allow for the robot to tip over. 
2. I have noticed an unexpected behavior. If I include a spherical joint between the post and pad, the pads partially sink in the floor and robot has a (slight) tilt dependent on the location of the center of mass. The depth of the sink is proportional to the contact force, as illustrated in the screenshots. If I comment out the joint, the pads float on the floor, as expected. 

As far as I can tell, the dynamics of robot with partially sunk pads are realistic except the tilt due to the pads thinking. I have tried to understand what is happening with the simulation and I promise I have spent a bit of time on the MuJoCo documentation to no avail. 

Can you please help me understand what happens with the simulation? I think I'm missing something fundamental in setting up the spherical joints and the frictionless contact between the pads and the floor.

Thank you.

BU

[image]
[image]

TenguBoss, 2024-01-06:
T0 + 3 hr update (includes a lunch break)

I believe I fixed the problem of the air bearing pads sinking in the floor by adding an armature to the spherical joint like so

`<joint name="AB ball -X+Y" type="ball" damping="1e-5" armature="1e-4"/>`

However, I am leaving the topic open to get feedback on the solution.

Regards and a good weekend.

BU

yuvaltassa, 2024-01-08:
a. Try using the [elliptic cone](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option-cone) option, these contacts are often better behaved.
b. I just added [the formulas](https://mujoco.readthedocs.io/en/latest/modeling.html#reference) for penetration at rest, you might want to tweak your contact parameters.