google-deepmind/mujoco #1317: Problems Encountered in Rope Drive：(

chenxzu, 2024-01-06:
Hi,

I'm a student and I'm trying to use MuJoCo for cable drive.

Here is a model which explains my question:

Sorry to bother you again.

I want to be able to transfer force through the ropes to the joints by turning the dial, but I've tried many times and can't make it 

happen.

I have read all the issues here, and I would like to ask what I should do if I want to build a pulley block. It seems that it cannot be 

achieved through tendon. I know that cable should probably be used here, but I want to know if I can make it go specified when 

building the cable. path or through a specified point.

My mentor asked me to build a rope driven robotic arm, I've tried many things but still can't build it :(.

```XML
<mujoco>
    <compiler autolimits="true" />
  
    <default>
        <site type="sphere" rgba="1 0 0 1" size="0.01" />
        <tendon rgba="0 1 0 1" />
    </default>

    <worldbody>
        <light diffuse=".5 .5 .5" pos="0 0 3" dir="0 0 -1" />
        <geom name="floor" pos="0 0 -0.5" size="0 0 1" type="plane" />
        <site name="x1" pos=".1 0.2 0" />
       
        <body>
            <joint name="pin0" type="hinge" pos="-0.8 0 0" axis="0 0 1" />

            <geom name="lun0" type="cylinder" pos="-0.8 0 0" size="0.2 0.05" />
            <site name="s1" pos="-.8 .2 0" />
            <site name="s2" pos="-.8 -.2 0" />
            
            
        </body>
        <site name="s4" pos=".8 .2 0" />
        <site name="s11" pos="-0.8 .3 0" />
        <body pos="0 0 0">
            <geom name="body1" type="box" size="0.6 0.1 0.05" rgba="0 .9 0 1" mass="1" />
            <geom name="lun" type="cylinder" pos="0.6 0 0" size="0.2 0.05" />
            <site name="s3" pos="1.2 0.1 0" />
            
            <body pos="0.6 0 0">

                <joint name="pin" type="hinge" pos="0 0 0" axis="0 0 1" limited="true" range="0 120"
                    damping="0.1" />
                <geom name="lun2" type="cylinder" pos="0.4 0 0" size="0.2 0.05" />
                <geom name="body2" type="box" pos="1 0 0" size="0.6 0.1 0.05" rgba="0 0 .9 1"
                    mass="1" /> 
            </body>
        </body>

    </worldbody>
    <tendon>
        <spatial name="rope" width="0.009"  range="0 4" limited="true">
            <site site="s3" />
            <geom geom="lun2" />
            <site site="s4" />
            <geom geom="lun" />
            <site site="x1" />
            <geom geom="lun0" sidesite="s11"/>
            <site site="s1" />
        </spatial>
    </tendon>

    <actuator>
        <position name="actuator" joint="pin0" />
    </actuator>    

</mujoco>
```

Here is a screenshot / video, illustrating my question:
[image]

chenxzu, 2024-01-07:
Sorry to bother you again.

I think I solved part of the problem myself, but now I have a new problem that It keeps shaking even though I have added 

sidesite element. 

Is there a solution to this? Or is there something wrong with me? I really hope you can provide some suggestions.

```xml
<mujoco>
    <compiler autolimits="true" />
  
    <default>
        <site type="sphere" rgba="1 0 0 1" size="0.03" />
        <tendon rgba="0 1 0 1" />
    </default>

    <worldbody>
        <light diffuse=".5 .5 .5" pos="0 0 3" dir="0 0 -1" />
        <geom name="floor" pos="0 0 -0.5" size="0 0 1" type="plane" />
        <site name="x1" pos="-.8 .4 0"/>

        <body>
            <joint name="pin0" type="hinge" pos="-0.8 0 0" axis="0 0 1" limited="true" range="0 60" />
            <geom name="lun0" type="cylinder" pos="-0.8 0 0" size="0.2 0.05" />
            <site name="s3" pos="-.8 .2 0"/>

            <site name="s4" pos="-.8 -.2 0"/>
            
        </body>

        <body pos="0 0 0">
            <geom name="body1" type="box" size="0.6 0.1 0.05" rgba="0 .9 0 1" mass="1" />
            <geom name="lun" type="cylinder" pos="0.6 0 0" size="0.2 0.05" />
            <site name="s1" pos="0.6 .2 0"/>
            <site name="s5" pos="0.6 -.2 0"/>

            
            <body pos="0.6 0 0">

                <joint name="pin" type="hinge" pos="0 0 0" axis="0 0 1" limited="true" range="1 40" stiffness="0" />
                <geom name="lun2" type="cylinder" pos="0.4 0 0" size="0.2 0.05" />
                <geom name="body2" type="box" pos="1 0 0" size="0.6 0.1 0.05" rgba="0 0 .9 1"
                    mass="1" />
                <site name="s2" pos=".4 .2 0"/> 
                <site name="s6" pos=".4 -.2 0"/> 
                
            </body>
        </body>

    </worldbody>

    <tendon>
        <spatial  width="0.002" range="1.79 1.81" limited="true"  solreflimit=".01 2" damping="0.1">
            <site site="s3"/>
            <geom geom="lun0" sidesite="x1"/>
            <site site="s1"/>            
            <site site="s2"/>
        </spatial>

        <spatial width="0.002" range="1.79 1.81" limited="true"  solreflimit=".01 2" damping="0.1">
            <site site="s6"/>
            <geom geom="lun"/>
            <site site="s5"/>
            <geom geom="lun" />            
            <site site="s4"/> 
        </spatial>

    </tendon>

    <actuator>
        <position name="position" joint="pin0" kp="100" ctrllimited="true" ctrlrange="0 1 "/>
        <velocity name="velocity" joint="pin0" kv="100" ctrllimited="true" ctrlrange="0 1 "/>
    </actuator>    

</mujoco>
```

https://github.com/google-deepmind/mujoco/assets/148617991/c2283ea0-d2a4-48ca-a23f-f5757b13eb49

yuvaltassa, 2024-01-08:
The site needs to be properly **outside** the cylinder. You put it _exactly_ on the surface which means that numerically it is sometimes outside and sometimes inside. 
```xml
<site name="s3" pos="-.801 .2 0"/>
```

closed 2024-01-08 by yuvaltassa

chenxzu, 2024-01-09:
Oh~thank you very much.

chenxzu, 2024-01-09:
I would like to ask, if  use the cable element, can the rope be wrapped around a specified object, just like the way the rope is wound around a movable pulley ?