google-deepmind/mujoco #1322: Get internal mechanical stress in elasticity plugin

giorgionicola, 2024-01-08:
Hi,
I'm a researcher and I'm trying to use MuJoCo for simulating deformable objects.

I would like to get the internal stress status of a cable that I am simulating. Is it possible?
I can see that in [cable.h](https://github.com/google-deepmind/mujoco/blob/5308b485dcb84b08dfd1d19916e344445ce814ad/plugin/elasticity/cable.h#L48) the internal mechanical stress is stored. Instead, the stress tensor for other elements, such as the solid, does not seem to be saved.

I am working with Mujoco 3.1.1 and the Python bindings.

quagla, 2024-01-08:
Hi, I'm afraid this is not possible at the moment. 

The reason is that with the Saint Venant–Kirchhoff model we use, the Hessian of the elastic energy with respect to edge elongations is constant, so we precompute it using the invariants of the Lagrange-Green strain tensor E (see https://github.com/google-deepmind/mujoco/blob/main/plugin/elasticity/elasticity.h#L149). For the stress, you would need to take derivatives of the energy wrt E.  

However, since the energy is quadratatic in the edge elongations (where elongation = edge^T*E*edge) we have that S = dEnergy/dE = dEnergy/dElongation * dElongation/dE = metric * elongation * dElongation/dE so maybe you could use only  metric * elongation as a proxy for knowing the stress along the edges?I know this is not a precise answer but I'm trying to come up with a workaround..

giorgionicola, 2024-01-08:
Ok thank you very much. Is this also valid for the cable element?

quagla, 2024-01-09:
No this is valid for tensile stresses (Solid and Membrane) but it's not valid for bending/twisting stresses.

closed 2024-01-09 by quagla