google-deepmind/mujoco #1324: Movable pulley

chenxzu, 2024-01-09:
Hi, sorry to bother you.

I'm a student and I'm trying to use MuJoCo for Rope driven robotic arm.

I used the tendon element to implement a rope-driven demo before, but this didn't quite meet my needs. (This was mentioned in issue

 #1317 )  So I want to ask if it can be implemented using the cable element. I read the documentation and all related issues and 

examples. I only found how to connect two  objects through cable, but I did not find how to achieve a pulley-like effect. So I would like 

to ask how to use the cable element to implement a moving pulley model. I am not a native English speaker, which makes it difficult for 

me to read the documentation. Please forgive me if my expression is wrong or disrespectful.

Here is a picture, illustrating my question:
[image]

quagla, 2024-01-09:
Hi @chenxzu, it is in principle possible but you may need to do it in two steps:

- Load one of the cable examples in the simulation GUI, then press save XML.
- Open the saved XML and compare it with the one you loaded, you will see that the `composite` was expanded into a series of bodies (the individual segments of the cable), now you can edit this file and move or resize each body in any way you like in order to obtain something that looks like your picture.

Let me know if that makes sense and remember that the cable is inextensible.

quagla, 2024-01-09:
Also remember to set `flat=true` in the cable otherwise you will see that the cable with try to keep its initial bent configuration.

closed 2024-01-09 by quagla

rishabkr29, 2024-01-15:
can we fix the pulley and pull the cable with one end and other end might be connected with some other object ? if yes can you please share one example. I really need one.