google-deepmind/mujoco #1328: Insufficient memory error while loading a mesh file in the `flexcomp`

sangteak601, 2024-01-11:
Hi,

I'm trying to use MuJoCo for simulating grasping soft objects. I have a 3d CAD file for the object which I want to grasp and I want the object to be a soft body. Before integrating the actual model, I started testing `flexcomp` with simple models such as a cube and a sphere. I was able to load the cube model without a problem, but when I tried to load a sphere model, it threw a following error and crashed. 

```
ERROR: mj_stackAlloc: insufficient memory: max = 52428800, available = 3830528, requested = 4000000 (ne = 0, nf = 0, nefc = 0, ncon = 0)
```

The sphere model has 4902 vertices and 9800 faces according to MeshLab.
I tried both `solid` and `membrane` for plugins and also adjusted memory size and solver options.

These are my questions.
1. What could cause this problem and what would be a possible solution?
2. Is there any example of using mesh file with `flexcomp` ?

Here is a model which I used to create `flexcomp`:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
  <option timestep=".001" solver="CG"/>
  <size memory="50M"/>
  <extension>
    <plugin plugin="mujoco.elasticity.solid"/>
    <plugin plugin="mujoco.elasticity.membrane"/>
  </extension>
  <worldbody>
    <geom name="ground" type="plane" size="0 0 1" pos="0 0 0" quat="1 0 0 0" rgba=".5 .5 .5 1" condim="1"/>
    <flexcomp name="softbody" type="mesh" file="meshes/sphere.stl" pos="0 0 1.5"
             radius=".005" rgba="0 .7 .7 1" dim="2" mass="0.1" scale="0.001 0.001 0.001">
      <edge damping=".5"/>
      <plugin plugin="mujoco.elasticity.membrane">
         <config key="poisson" value="0.3"/>
         <config key="young" value="1e6"/>
         <config key="thickness" value=".01"/>
      </plugin>
   </flexcomp>
  </worldbody>
</mujoco>
```

</details>

Here is a zip file which also includes stl file
[flexcomp_test.zip](https://github.com/google-deepmind/mujoco/files/13901245/flexcomp_test.zip)

sangteak601, 2024-01-11:
My setup is as follows:

1. Ubuntu 22.04
2. Mujoco 3.1.1
3. Used `simulate` executable to load a model

quagla, 2024-01-12:
How much did you increase the memory to?

sangteak601, 2024-01-13:
Hi @quagla, Thanks for the reply.

I increased it to 50M, as you can see from the minimal XML <size memory="50M"/>. However, I assume my PC has only 3.8M available, as indicated by the error message.

I wonder if the number of vertices in the mesh causes the problem. The actual model I want to import is much more complicated than a simple sphere.

sangteak601, 2024-01-16:
I was able to load the model by adding a following line to `xml` file.
`<contact selfcollide="none"/>`

But I don't really understand why I need to disable `selfcollide`.

Now the simulation is not stable and throw warnings as follows.
```
WARNING: Nan, Inf or huge value in QACC at DOF 0. The simulation is unstable. Time = 0.0050
```

quagla, 2024-01-22:
Hi, try reduce the time step.

You can definitely increase the memory well above 50M.

closed 2024-01-22 by quagla

sangteak601, 2024-03-13:
Hi @quagla, Thanks for your help.

I'm still struggling to load mesh file as a `flexcomp`. It seems like complicated model(with more than 1000 vertices) is not possible to be loaded as `flexcomp`.

Have you tried loading mesh as a `flexcomp`? Do you have any idea how complicated(in terms of the number of vertices or faces) model can be comfortably loaded ?

arpit15, 2024-04-30:
I am getting a similar error in Mac. I also set the memory to 50M. My model doesn't load even with `<contact selfcollide="none"/>`. I am trying to create a `flexcomp` with `.msh` mesh
```
Element 'contact', line 13
ERROR: mj_stackAlloc: insufficient memory: max = 51475744, available = 19895992, requested = 31394888 (ne = 0, nf = 0, nefc = 2640, ncon = 660)
```

Pablo-M-V-99, 2024-07-28:
Hi @arpit15 and @sangteak601,
did you manage to solve your problem?