google-deepmind/mujoco #1329: I can't enable wireframe mode

traplotz, 2024-01-11:
Hi,
I'm an engineer and I'm trying to use MuJoCo python bindings for my new project.
I want to make a video rendered in wireframe mode because it gives a much clearer look for my model.

Here is a code:
```    
with mujoco.Renderer(model) as renderer:
        scene = mujoco.MjvScene()
        scene.flags[mujoco.mjtRndFlag.mjRND_WIREFRAME] = 1  # Enable wireframe mode
        duration = 10  # (seconds)
        framerate = 60  # (Hz)
        frames = []
        mujoco.mj_resetData(model, data)
        while data.time < duration:
            mujoco.mj_step(model, data)
            if len(frames) < data.time * framerate:
                renderer.update_scene(data, camera="Camera_0")
                pixels = renderer.render()
                frames.append(pixels)
renderer.close()
```

Those two lines with `scene` totally worked with `mjr_render`, but I don't know how to make them work with `Renderer`.
I would greatly appreciate it if you could let me know the way if there is any.
Thank you.

kevinzakka, 2024-01-12:
I believe this is a known issue, see #1190.

traplotz, 2024-01-12:
> I believe this is a known issue, see #1190.

So there isn't a solution yet? It's fine then, wasn't a big deal. 
Thank you.

saran-t, 2024-02-01:
This is not #1190 since it's got nothing to do with the viewer.

closed 2024-02-01 by saran-t