google-deepmind/mujoco #1333: MjrContext: mujoco.FatalError: gladLoadGL error

kankanzheli, 2024-01-12:
**How to ask for help**

Hi,

I'm a student  and I'm trying to use MuJoCo for research.

I'm looking for some help with getting depth data from mujoco's camera.

Here is a model which explains my question:
I am working with:
Ubuntu22
mujoco3.0.0
python3.9
glfw2.5.9
<details>

```XML
<?xml version="1.0"?>
<mujoco>
  <option gravity="0 0 0"/>
  <asset>
    <mesh name="zedmini" file="zedmini.stl"/>
  </asset>

  <worldbody>
    <body name="zedmini" pos="0.09475 0 0.02754">
      <inertial diaginertia="1e-4 1e-4 1e-4" mass="1e-2" pos="0 0.01 0"/>
      <geom name="zedmini" type="mesh" mesh="zedmini" pos="0 0 0" contype="0" conaffinity="0"/>
      <camera name="rgbd_camera" pos="0 0 0" euler="90 -90 0"/>
    </body>
  </worldbody>
</mujoco>
```

</details>

The python codes used for getting depth and RGB data from mujoco's camera.
`import mujoco
import numpy as np
import matplotlib.pyplot as plt

model = mujoco.MjModel.from_xml_path('robot.xml')
data = mujoco.MjData(model)

ctx = mujoco.MjrContext(model, 150)  

scene = mujoco.MjvScene(model, maxgeom=5000)

mujoco.mjv_updateScene(model, data, mujoco.MjvOption(), None, mujoco.MjvPerturb(), mujoco.mjtCatBit.mjCAT_ALL.value, scene)

camera_id = mujoco.mj_name2id(model, mujoco.mjtObj.mjOBJ_CAMERA, "rgbd_camera")

width, height = 800, 800

rgb_buffer = np.zeros((height, width, 3), dtype=np.uint8)
mujoco.mjr_render(0, scene, ctx)
mujoco.mjr_readPixels(rgb_buffer, None, ctx)

depth_buffer = np.zeros((height, width), dtype=np.float32)
mujoco.mjr_readPixels(None, depth_buffer, ctx)

depth_image_normalized = (depth_buffer - depth_buffer.min()) / (depth_buffer.max() - depth_buffer.min())

plt.figure(figsize=(10, 5))
plt.subplot(1, 2, 1)
plt.imshow(rgb_buffer)
plt.title("RGB Image")
plt.axis('off')

plt.subplot(1, 2, 2)
plt.imshow(depth_image_normalized, cmap='gray')
plt.title("Depth Image")
plt.axis('off')

plt.show()
`

Finally, the terminal return:
Traceback (most recent call last):
  File "camera.py", line 10, in <module>
    ctx = mujoco.MjrContext(model, 150) 
mujoco.FatalError: gladLoadGL error

I don't know why this error occured.

rs545837, 2024-01-14:
Hey @kankanzheli what version of mujoco are you using?

kankanzheli, 2024-01-16:
I have showed the version of mujoco in my question. The mujoco is downloaded in anconda enviroment by 'pip install'. The return of 'print(mujoco.__version__)' is 3.0.0. The version of python is 3.9.

kankanzheli, 2024-01-22:
Do you have any answers?

closed 2024-01-24 by kankanzheli