google-deepmind/mujoco #1336: How to route a Cable in MuJoCo ?

rishabkr29, 2024-01-15:
**How to ask for help**

First, read our quick guide to
[asking good questions](https://github.com/google-deepmind/mujoco#asking-questions).
Below is a template for you to use:

Hi,

I'm a (student / professor / engineer) and I'm trying to use MuJoCo for _____.

I'm looking for some help with ____.

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
  <worldbody>
    <light pos=".4 -.4 .3" dir="-2 2 -1.5" diffuse=".6 .6 .6"/>
    <light pos="-.2 -.4 .3" dir="1 2 -1.5" diffuse=".6 .6 .6"/>
    <geom type="plane" size=".5 .5 .01"/>
    <body name="ball" pos="0 0 0.05">
      <freejoint/>
      <geom size=".03"/>
    </body>
    <body name="hammer" pos="-.05 0 0.18">
      <joint axis="0 1 0"/>
      <geom type="capsule" size=".01" fromto="0 0 0 -.15 0 0"/>
      <geom type="box" size=".02 .02 .03" pos="-.15 0 0"/>
    </body>
  </worldbody>
</mujoco>
```

</details>

Here is a screenshot / video, illustrating my question:
[image]

rishabkr29, 2024-01-15:
basically, I am trying to model cable driven actuator system and with the help of python binding I tried to actuate the manipulator. In this model the slider was leaving the contact and the cable was becoming slack which I don't want. so after checking all the modal parameter. I decided to reconfigure the actuation method. 
```<mujoco>

  <option timestep="0.01" gravity="0 0 -9.81" integrator="Euler" jacobian="dense">
    <flag energy="enable" constraint = "enable" contact="enable" sensornoise="enable"/>   <!--#flag enable and disable different part of simulation pipeline.-->
  </option>

  <!--warmstart is enabled-->

  <compiler autolimits="true"/>   <!--settings are global and apply to entire model-->

  <size  njmax="15000" nconmax="500"  nkey="1"/>     <!--Key frame -->

  <size memory="5M"/> 

  <extension>
      <plugin plugin="mujoco.elasticity.cable"/>
  </extension>

  <default>
    <geom solref="1e-2 5" solimp=".5 .99 1e-3" friction=".1"/>
  </default>

  <visual>
    <rgba haze=".3 .3 .3 1"/>
  </visual>

 <visual>
    <rgba rangefinder="1 1 0.1 0.1"/>
    <map znear=".005" zfar="20"/>
  </visual>

  <asset>
    <hfield name="terrain" ncol="201" nrow="201" size="30 30 5 .1"/>
  </asset>

  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.6 0.6 0.6" rgb2="0 0 0" width="512" height="512"/>

    <texture name="texplane" type="2d" builtin="checker" rgb1=".25 .25 .25" rgb2=".3 .3 .3" width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>

    <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
  </asset>

  <worldbody>

        <light diffuse=".5 .5 .5" pos="0 0 10" dir="0 0 -1" />

        <geom type="plane" size="0.4 0.4 0.05" rgba="1 1 1 1" />

        <geom type="plane" pos="0 0 -1" size="10 10 10" rgba="0.5 0.5 0.5 1" />

        <site name="test" pos="-0.05 -0.05 0.06"  size="0.008" rgba="0.8 .9 0 1" />

        <body name="Extrusion">
            <geom type="box" pos="0 0.4 0"   euler="0 0 0" mass="0.2" size="0.4 0.05 0.05" rgba="0.5 0.5 0.5 1"/>
            <geom type="box" pos="0 -0.4 0"  euler="0 0 0" mass="0.2" size="0.4 0.05 0.05" rgba="0.5 0.5 0.5 1"/>
            <geom type="box" pos="0.4 0 0"   euler="0 0 0" mass="0.2" size="0.05 0.4 0.05" rgba="0.5 0.5 0.5 1"/>
            <geom type="box" pos="-0.4 0 0"  euler="0 0 0" mass="0.2" size="0.05 0.4 0.05" rgba="0.5 0.5 0.5 1"/>
        </body>
<!--
        <body name="Table Leg">
            <geom type="box" pos="0.4 0.4 0" euler="0 0 0" mass="0.2" size="0.05 0.01 0.5" rgba="0.1 0.1 0.1 1"/>
        </body>
-->
        <body name="slider_M1" pos="-0.4 0.15 0.06" euler="0 0 135">
            <inertial pos="0 0 0" quat="0 0 0 1" mass="0.1" diaginertia="0.1 0.1 0.1" />
            <geom type = "box" size=".01 0.01 0.01" rgba="1 1 1 1"/>
            <site name="M1" pos="0 0 0"  size="0.008" rgba="0.8 .9 0 1" />
            <joint type="slide" axis="1 0 0" damping="0.1" armature="0.00001" name="slider_joint1"/>
[... 93 lines cut ...]
    </contact>
-->

    <actuator>   
        <position name="x_axis1"     joint="slider_joint1"    ctrlrange="-0.2 0.2"  kp="100"/>
        <position name="x_axis2"     joint="slider_joint2"    ctrlrange="-0.2 0.2"  kp="100"/>  
        <position name="x_axis3"     joint="slider_joint3"    ctrlrange="-0.2 0.2"  kp="100"/>
        <position name="x_axis4"     joint="slider_joint4"    ctrlrange="-0.2 0.2"  kp="100"/> 
        <velocity name="x_dotaxis1"  joint="slider_joint1"    ctrlrange="-0.2 0.2"  kv="10"/>
        <velocity name="x_dotaxis2"  joint="slider_joint2"    ctrlrange="-0.2 0.2"  kv="10"/>
        <velocity name="x_dotaxis3"  joint="slider_joint3"    ctrlrange="-0.2 0.2"  kv="10"/>
        <velocity name="x_dotaxis4"  joint="slider_joint4"    ctrlrange="-0.2 0.2"  kv="10"/>
    </actuator>   

    <sensor>
        <force name="force_in_cable1"   site="l1"/>
        <force name="force_in_cable2"   site="l2"/>
        <force name="force_in_cable3"   site="l3"/>
        <force name="force_in_cable4"   site="l4"/>
        <torque name=" tau_at_1st "     site="1st Joint"/>
    </sensor>

</mujoco>

`

rishabkr29, 2024-01-15:
> basically, I am trying to model cable driven actuator system and with the help of python binding I tried to actuate the manipulator. In this model the slider was leaving the contact and the cable was becoming slack which I don't want. so after checking all the modal parameter. I decided to reconfigure the actuation method.
> 
> ```
> 
>   <option timestep="0.01" gravity="0 0 -9.81" integrator="Euler" jacobian="dense">
>     <flag energy="enable" constraint = "enable" contact="enable" sensornoise="enable"/>   
>   </option>
> 
>   
> 
>   <compiler autolimits="true"/>   
> 
>   <size  njmax="15000" nconmax="500"  nkey="1"/>     
> 
>   <size memory="5M"/> 
> 
>   <extension>
>       <plugin plugin="mujoco.elasticity.cable"/>
>   </extension>
> 
>   <default>
>     <geom solref="1e-2 5" solimp=".5 .99 1e-3" friction=".1"/>
>   </default>
> 
>   <visual>
>     <rgba haze=".3 .3 .3 1"/>
>   </visual>
> 
>  <visual>
>     <rgba rangefinder="1 1 0.1 0.1"/>
>     <map znear=".005" zfar="20"/>
>   </visual>
> 
>   <asset>
>     <hfield name="terrain" ncol="201" nrow="201" size="30 30 5 .1"/>
>   </asset>
> 
>   <asset>
>     <texture type="skybox" builtin="gradient" rgb1="0.6 0.6 0.6" rgb2="0 0 0" width="512" height="512"/>
> 
>     <texture name="texplane" type="2d" builtin="checker" rgb1=".25 .25 .25" rgb2=".3 .3 .3" width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>
> 
>     <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
>   </asset>
> 
>   <worldbody>
> 
>         <light diffuse=".5 .5 .5" pos="0 0 10" dir="0 0 -1" />
> 
>         <geom type="plane" size="0.4 0.4 0.05" rgba="1 1 1 1" />
> 
>         <geom type="plane" pos="0 0 -1" size="10 10 10" rgba="0.5 0.5 0.5 1" />
> 
>         <site name="test" pos="-0.05 -0.05 0.06"  size="0.008" rgba="0.8 .9 0 1" />
> 
>         <body name="Extrusion">
>             <geom type="box" pos="0 0.4 0"   euler="0 0 0" mass="0.2" size="0.4 0.05 0.05" rgba="0.5 0.5 0.5 1"/>
>             <geom type="box" pos="0 -0.4 0"  euler="0 0 0" mass="0.2" size="0.4 0.05 0.05" rgba="0.5 0.5 0.5 1"/>
>             <geom type="box" pos="0.4 0 0"   euler="0 0 0" mass="0.2" size="0.05 0.4 0.05" rgba="0.5 0.5 0.5 1"/>
>             <geom type="box" pos="-0.4 0 0"  euler="0 0 0" mass="0.2" size="0.05 0.4 0.05" rgba="0.5 0.5 0.5 1"/>
>         </body>
> 
>         <body name="slider_M1" pos="-0.4 0.15 0.06" euler="0 0 135">
>             <inertial pos="0 0 0" quat="0 0 0 1" mass="0.1" diaginertia="0.1 0.1 0.1" />
>             <geom type = "box" size=".01 0.01 0.01" rgba="1 1 1 1"/>
>             <site name="M1" pos="0 0 0"  size="0.008" rgba="0.8 .9 0 1" />
>             <joint type="slide" axis="1 0 0" damping="0.1" armature="0.00001" name="slider_joint1"/>
>        </body>
> 
>        <body name="slider_M3" pos="-0.4 -0.15 0.06" euler="0 0 -135">
[... 84 lines cut ...]
>     <actuator>   
>         <position name="x_axis1"     joint="slider_joint1"    ctrlrange="-0.2 0.2"  kp="100"/>
>         <position name="x_axis2"     joint="slider_joint2"    ctrlrange="-0.2 0.2"  kp="100"/>  
>         <position name="x_axis3"     joint="slider_joint3"    ctrlrange="-0.2 0.2"  kp="100"/>
>         <position name="x_axis4"     joint="slider_joint4"    ctrlrange="-0.2 0.2"  kp="100"/> 
>         <velocity name="x_dotaxis1"  joint="slider_joint1"    ctrlrange="-0.2 0.2"  kv="10"/>
>         <velocity name="x_dotaxis2"  joint="slider_joint2"    ctrlrange="-0.2 0.2"  kv="10"/>
>         <velocity name="x_dotaxis3"  joint="slider_joint3"    ctrlrange="-0.2 0.2"  kv="10"/>
>         <velocity name="x_dotaxis4"  joint="slider_joint4"    ctrlrange="-0.2 0.2"  kv="10"/>
>     </actuator>   
> 
>     <sensor>
>         <force name="force_in_cable1"   site="l1"/>
>         <force name="force_in_cable2"   site="l2"/>
>         <force name="force_in_cable3"   site="l3"/>
>         <force name="force_in_cable4"   site="l4"/>
>         <torque name=" tau_at_1st "     site="1st Joint"/>
>     </sensor>
> 
> </mujoco>
> 
> `
> ```

https://github.com/google-deepmind/mujoco/assets/110694658/5e4ad4a4-19c5-45ac-95e7-7170b7bbb278

quagla, 2024-01-16:
Thanks for the file and the video! 

If you don't want the cable to become slack, then you don't want to use the cable but the tendon, so I have to ask again why is the tendon not working for you? Is it because you don't want the tendon to stretch?

rishabkr29, 2024-01-16:
> Thanks for the file and the video!
> 
> If you don't want the cable to become slack, then you don't want to use the cable but the tendon, so I have to ask again why is the tendon not working for you? Is it because you don't want the tendon to stretch?

Thanks for your quick response, From biomechanics perspective tendons are meant to be actuated by the muscles. I am not sure weather I can model the tendons as cable and actuate it with the slider or winch. If you share your thoughts it will be of great help. 

Looking forward of your quick response, I hope what I wanted to achieve out of it.

quagla, 2024-01-22:
> Yes It should be inextensible and should behave like a rigid body.

I don't understand how the two requirements are possible. If it's inextensible it must become slack.

closed 2024-01-29 by quagla