google-deepmind/mujoco #1337: Need help for using the inbuilt PD controller of MuJoCo

rishabkr29, 2024-01-15:
Hello MuJoCo Community, 

I'm a student  and I'm trying to use MuJoCo for thesis.

I'm looking for some help related to inbuilt PD controller of MuJoCo.

Here is a model which explains my question: based on the model I am calculating the slider displacement and directly passing it to the data.ctrl (which means I am trying to use feedforward as my high level ) and rest I am dependent on the low level pd of the mujoco (I read somewhere mujoco have inbuilt PD). 

My question is :
1.  My output to the system is to reach the end effector to the refence value. Instead of giving the correct slider length and using the correct variable ids. why it is not able to achieve. 

<details>
  <summary>minimal XML</summary>

```

import mujoco as mj
from mujoco.glfw import glfw
import numpy as np
import matplotlib.pyplot as plt
import os
import math 
import sympy as sp
from numpy import array
from numpy.linalg import norm

xml_path = '2RCDM_SliderModel.xml' #xml file (assumes this is in the same folder as this file)
simend = 5 #simulation time
print_camera_config = 0 #set to 1 to print camera config
                        #this is useful for initializing view of the model)

# For callback functions

button_left = False
button_middle = False
button_right = False
lastx = 0
lasty = 0

#########################################################################
global l_init,ldot_init 

l0_init = 0.1414      # calculation from the model 
l1_init = 0.3905   
l2_init = 0.1414
l3_init = 0.3905
l_init = np.array([l0_init,l1_init,l2_init,l3_init])

ldot0_init = 0
ldot1_init = 0
ldot2_init = 0
ldot3_init = 0

ldot_init = np.array([ldot0_init,ldot1_init,ldot2_init,ldot3_init])
# initial cordinate of circle say x0 and y0 

global x0,y0,x0_dot,y0_dot,r,w
x0 = -0.1
y0 =  0
x0_dot = 0
y0_dot = 0
r = 0.005 # in m
w = 1 # rad/sec

#time 
t0 = 0
t1 = 5 

# link & Motor Parameter  
global L1,L2
[... 425 lines cut ...]
    viewport = mj.MjrRect(0, 0, viewport_width, viewport_height)

    #print camera configuration (help to initialize the view)
    if (print_camera_config==1):
        print('cam.azimuth =',cam.azimuth,';','cam.elevation =',cam.elevation,';','cam.distance = ',cam.distance)
        print('cam.lookat =np.array([',cam.lookat[0],',',cam.lookat[1],',',cam.lookat[2],'])')

    # Update scene and render
    mj.mjv_updateScene(model, data, opt, None, cam,
                       mj.mjtCatBit.mjCAT_ALL.value, scene)
    mj.mjr_render(viewport, scene, context)

    # swap OpenGL buffers (blocking call due to v-sync)
    glfw.swap_buffers(window)

    # process pending GUI events, call GLFW callbacks
    glfw.poll_events()

glfw.terminate()

```

</details>

Here is a screenshot / video, illustrating my question:

rishabkr29, 2024-01-15:
here is my xml model : `<mujoco>

  <option timestep="0.01" gravity="0 0 -9.81" integrator="Euler" jacobian="dense">
    <flag energy="enable" constraint = "enable" contact="enable" sensornoise="enable"/>   
  </option>

  

  <compiler autolimits="true"/>   

  <size  njmax="15000" nconmax="1000"  nkey="1"/>     
  
  <extension>
      <plugin plugin="mujoco.elasticity.cable"/>
  </extension>

  <default>
    <geom friction=".01"/>
  </default>

  <asset>
    <hfield name="terrain" ncol="201" nrow="201" size="30 30 5 .1"/>
  </asset>

  <worldbody>

        <light diffuse=".5 .5 .5" pos="0 0 10" dir="0 0 -1" />

        <geom type="plane" size="0.4 0.4 0.05" rgba="1 1 1 1" />

        <geom type="plane" pos="0 0 -1" size="10 10 10" rgba="0.5 0.5 0.5 1" />

        <site name="test" pos="-0.05 -0.05 0.06"  size="0.008" rgba="0.8 .9 0 1" />

        <body name="Extrusion">
            <geom type="box" pos="0 0.4 0"   euler="0 0 0" mass="0.2" size="0.4 0.05 0.05" rgba="0.5 0.5 0.5 1"/>
            <geom type="box" pos="0 -0.4 0"  euler="0 0 0" mass="0.2" size="0.4 0.05 0.05" rgba="0.5 0.5 0.5 1"/>
            <geom type="box" pos="0.4 0 0"   euler="0 0 0" mass="0.2" size="0.05 0.4 0.05" rgba="0.5 0.5 0.5 1"/>
            <geom type="box" pos="-0.4 0 0"  euler="0 0 0" mass="0.2" size="0.05 0.4 0.05" rgba="0.5 0.5 0.5 1"/>
        </body>

        <body name="slider_M1" pos="-0.4 0.15 0.06" euler="0 0 135">
            <inertial pos="0 0 0" quat="0 0 0 1" mass="0.1" diaginertia="0.1 0.1 0.1" />
            <geom type = "box" size=".01 0.01 0.01" rgba="1 1 1 1"/>
            <site name="M1" pos="0 0 0"  size="0.008" rgba="0.8 .9 0 1" />
            <joint type="slide" axis="1 0 0" damping="0.1" armature="0.00001" name="slider_joint1"/>
       </body>

       <body name="slider_M3" pos="-0.4 -0.15 0.06" euler="0 0 -135">
            <inertial pos="0 0 0" quat="0 0 0 1" mass="0.1" diaginertia="0.1 0.1 0.1" />
            <geom type = "box" size=".01 0.01 0.01" rgba="1 1 1 1" />
            <site name="M3" pos="0 0 0"  size="0.008" rgba="0.8 .9 0 1" />
            <joint type="slide" axis="1 0 0" damping="0.1" armature="0.00001"  name="slider_joint3"/>
        </body>

       <body name="slider_M2" pos="-0.4 0.3 0.06" euler="0 0 135">
            <inertial pos="0 0 0" quat="0 0 0 1" mass="0.1" diaginertia="0.1 0.1 0.1" />
            <geom type = "box" size=".01 0.01 0.01" rgba="1 1 1 1" />
            <site name="M2" pos="0 0 0"  size="0.008" rgba="0.8 .9 0 1" />
            <joint type="slide" axis="1 0 0" damping="0.1"  armature="0.00001" name="slider_joint2"/>
       </body>

       <body name="slider_M4" pos="-0.4 -0.3 0.06" euler="0 0 -135">
            <inertial pos="0 0 0" quat="0 0 0 1" mass="0.1" diaginertia="0.1 0.1 0.1" />
            <geom type = "box" size=".01 0.01 0.01" rgba="1 1 1 1" />
            <site name="M4" pos="0 0 0"  size="0.008" rgba="0.8 .9 0 1" />
            <joint type="slide" axis="1 0 0" damping="0.1" armature="0.00001" name="slider_joint4"/>
        </body>
     
        <body name = "Link1" pos="-0.3 0 0.06">
[... 49 lines cut ...]
            </body> 
        </body>
  </worldbody>

    <equality>
      <connect name="M1_boundary"  body1="DB_last"  body2="slider_M1" active = "true" anchor="0 0 0" /> #acnhor point is specified according to local coordinate of first body 
      <connect name="M3_boundary"  body1="CB_last"  body2="slider_M3" active = "true" anchor="0 0 0" />
      <connect name="M2_boundary"  body1="EB_last"  body2="slider_M2" active = "true" anchor="0 0 0" />
      <connect name="M4_boundary"  body1="FB_last"  body2="slider_M4" active = "true" anchor="0 0 0" />
    </equality>

    <actuator>   
        <position name="x_axis1"     joint="slider_joint1"    ctrlrange="-0.2 0.2"  kp="50"/>
        <position name="x_axis2"     joint="slider_joint2"    ctrlrange="-0.2 0.2"  kp="50"/>  
        <position name="x_axis3"     joint="slider_joint3"    ctrlrange="-0.2 0.2"  kp="50"/>
        <position name="x_axis4"     joint="slider_joint4"    ctrlrange="-0.2 0.2"  kp="50"/> 
        <velocity name="x_dotaxis1"  joint="slider_joint1"    ctrlrange="-0.2 0.2"  kv="5"/>
        <velocity name="x_dotaxis2"  joint="slider_joint2"    ctrlrange="-0.2 0.2"  kv="5"/>
        <velocity name="x_dotaxis3"  joint="slider_joint3"    ctrlrange="-0.2 0.2"  kv="5"/>
        <velocity name="x_dotaxis4"  joint="slider_joint4"    ctrlrange="-0.2 0.2"  kv="5"/>
    </actuator>   

    </sensor>

</mujoco>`

yuvaltassa, 2024-02-09:
The code block is empty?
Closing.

closed 2024-02-09 by yuvaltassa