google-deepmind/mujoco #1340: AttributeError: 'MjModel' object has no attribute 'actuator_actadr'

irvingZ11, 2024-01-16:
describe the bug:
AttributeError: 'MjModel' object has no attribute 'actuator_actadr'
and i did not find the attribute in my mujoco

joek5555, 2024-01-19:
Are you using the dm_control package? If so, please check the version to see if it is compatible with mujoco 2.2.0. [This link](https://github.com/google-deepmind/dm_control/releases) shows the different releases of dm_control and what version of mujoco they require. You might also need to update mujoco.

irvingZ11, 2024-02-19:
> Are you using the dm_control package? If so, please check the version to see if it is compatible with mujoco 2.2.0. [This link](https://github.com/google-deepmind/dm_control/releases) shows the [...]

thanks! I update mujoco and it works

closed 2024-02-19 by irvingZ11