google-deepmind/mujoco #1344: [MJX] Gradients w.r.t. frictionloss are always zero

bheijden, 2024-01-17:
Hi,

I'm a PhD student working on system identification for a real pendulum setup in our lab. While I've successfully determined parameters like inertia, length, mass, and gear ratio, the model overestimates damping. Incorporating dry friction in the joint could resolve this.

In transitioning from Brax to MJX, I noticed the mjx._src.types.Model class includes a dof_frictionloss attribute. This attribute seems ideal for modeling joint friction. However, I'm encountering an issue where gradients with respect to dof_frictionloss are consistently zero, in contrast to gradients for other parameters. This leads me to question if dof_frictionloss is effectively integrated into the dynamics computations. The absence of errors when loading the model via mjx.device_put suggests its implementation, yet its non-reactivity in the dynamics calculation is puzzling.

I'm using mujoco and mujoco-mjx version 3.1.1. Any insights into the functionality of dof_frictionloss or suggestions for alternative approaches to model dry friction are welcome.

```XML
<mujoco model="disk_pendulum">
    <compiler inertiafromgeom="auto" angle="radian" coordinate="local" eulerseq="xyz" autolimits="true"/>
    <option gravity="0 0 -9.81" timestep="0.01" iterations="10"/>
    <asset>
        <texture builtin="flat" height="1278" mark="cross" markrgb="1 1 1" name="texgeom" random="0.01" rgb1="0.8 0.6 0.4" rgb2="0.8 0.6 0.4" type="cube" width="127"/>
        <material name="geom" texture="texgeom" texuniform="true"/>
    </asset>

    <default>
        <geom contype="0" friction="1 0.1 0.1" material="geom"/>
    </default>

    <worldbody>
        <light cutoff="100" diffuse="1 1 1" dir="-0 0 -1.3" directional="true" exponent="1" pos="0 0 1.3" specular=".1 .1 .1"/>
        <geom name="table" type="plane" pos="0 0.0 -0.1" size="1 1 0.1" contype="8" conaffinity="11" condim="3"/>
        <body name="disk" pos="0.0 0.0 0.0" euler="1.5708 0.0 0.0">
            <joint name="hinge_joint" type="hinge" axis="0 0 1" range="-180 180" armature="0.00022993" damping="0.0001" limited="false" frictionloss="123"/>
            <geom name="disk_geom" type="cylinder" size="0.06 0.001" contype="0" conaffinity="0" condim="3" mass="0.0"/>
            <geom name="mass_geom" type="cylinder" size="0.02 0.005" contype="0" conaffinity="0"  condim="3" rgba="0.04 0.04 0.04 1"
                  pos="0.0 0.04 0." mass="0.05085817"/>
        </body>
    </worldbody>

    <actuator>
        <motor joint="hinge_joint" ctrllimited="true" ctrlrange="-3.0 3.0"  gear="0.01"/>
    </actuator>
</mujoco>
```

btaba, 2024-01-17:
Hi @bheijden , frictionloss is not yet implemented in MJX, we'll update the docs to reflect that [here](https://mujoco.readthedocs.io/en/stable/mjx.html#feature-parity)

btaba, 2024-01-29:
Closing for now

closed 2024-01-29 by btaba

kassasin, 2024-12-23:
> Hi @bheijden , frictionloss is not yet implemented in MJX, we'll update the docs to reflect that [here](https://mujoco.readthedocs.io/en/stable/mjx.html#feature-parity)

Hello , i am doing some work about robot whole body system identification .I want to using mjx to compute robot dynamics and joint friction is a big part of it .(high motor gear ratio) . Does frictionloss is still unavailable in mjx?

kassasin, 2024-12-23:
> Closing for now
Hello,I am doing some work about humanoid robot whole body system identification . I want using mjx to compute dynamics and joint friction is a big part (hight gear reduction ratio) .Does frictionloss still unavailable in mjx?