google-deepmind/mujoco #1356: [MJX] visualize parallel environments in one viewer

panwliu, 2024-01-21:
First of all, thanks for the great work! MJX could make a great impact in the robotics community.

I'm doing massive parallel simulation using MJX. During the development phase, I'd like to visualize the parallel environments. For example, I spawned 4096 humanoids, and I'd like to visualize them all in a viewer as the image (a screenshot of Isaac Gym). I saw that [mjx/mujoco/mjx/viewer.py](https://github.com/google-deepmind/mujoco/blob/main/mjx/mujoco/mjx/viewer.py) is used to visualize one environment. How to visualize the parallel environments in one viewer?

Thanks
[image]

rs545837, 2024-01-22:
Yes I am also very curious to know if there's a way to do it @yuvaltassa.

btaba, 2024-01-26:
Hi folks,

For the MJX part, you can simulate N environments and write them back to MjData via [`get_data`](https://github.com/google-deepmind/mujoco/blob/3b0a680fa7acf69a3cc3fc72f91402bfeb50490d/mjx/mujoco/mjx/_src/io.py#L232) which accepts a batched `mjx.Data`.

Now for visualizing N environments, the simplest way I can think of is to create an `MjModel`/`MjData` with N characters (procedurally generate the XML or use something like `dm_env` composer), and then feed the qpos from the above step back into your `MjData`.  Others should comment on this replication bit though. I know that there has been some active discussion on better ways to `repeat` bodies.

panwliu, 2024-01-31:
Hi @btaba , thanks for the response.

Since mujoco doesn't allow repeated bodies in the XML, it actually makes procedurally generating N characters not trivial. Also needs careful handling in feeding the batched `mjx.Data`  to the right bodies.

Another possible way would be a "batched" version of `mjModel`. Would you think this as an option?

yuvaltassa, 2024-02-05:
I think what you want is visual only, no? Like, your humanoids don't need to touch each other at all, right?

This is something that @vikashplus (and, as it turns out, #1395) have asked for.

What you actually want, IIUC, is something like [mjv_updateScene](https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#mjv-updatescene), except you want "`mjv_addToScene`", i.e. some way to add together the geoms from multiple `mjData` instances into one visual scene. Correct?

vikashplus, 2024-02-05:
+1 `mjv_addToScene` will be amazing. I have been trying to hack this in multiple ways.

panwliu, 2024-02-06:
> some way to add together the geoms from multiple mjData instances into one visual scene

@yuvaltassa, exactly!

yuvaltassa, 2024-02-06:
Okay, this is MUCH easier than your initial proposal. I'll try to do something like this in the next few days, will update here.

AlexS28, 2024-02-25:
@yuvaltassa was wondering if there was any update on this feature. I think it would be really great to have, and not only good to get a nice overview for how the robots are behaving, but also nice for making videos for research papers etc.

alexhansson, 2024-03-25:
@yuvaltassa Any update on this? Would definitely be huge if this is somehow possible!

yuvaltassa, 2024-09-10:
So this turned out to be rather easy. All you need is the API function [mjv_addGeoms](https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#mjv-addgeoms) which, as the name suggests, adds geoms to an existing scene. I added a section to the [tutorial colab](https://github.com/google-deepmind/mujoco?tab=readme-ov-file#getting-started) showing how to do this.

closed 2024-09-10 by yuvaltassa