google-deepmind/mujoco #1357: [force sensing] I have a question

2-woo-10, 2024-01-22:
**How to ask for help**

First, read our quick guide to
[asking good questions](https://github.com/google-deepmind/mujoco#asking-questions).
Below is a template for you to use:

Hello

I am a student who is conducting an experiment to measure force using a Mujoko simulator.

I wrote this article to get advice on the following.

Currently, the force is measured when the contact between A1, A2, B1, and B2 and the object occurs as shown in the figure below. Through this, I'm done measuring the force, but I want to measure the force applied from the box by declaring the above green box as "geom" on the body rather than declaring it as a freepoint. Is it possible? I'm attaching the xml file.

It's hard to explain in words, so I attached a picture together.

[image]

[image]

[image]

<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
    <option gravity="0 0 -9.81"/>
    <asset>
        <texture name="grid" builtin="checker" height="500" rgb1=".1 .2 .3" rgb2=".2 .3 .4" type="2d" width="500"/>
        <material name="grid" texture="grid" texrepeat="1 1" texuniform="true" reflectance=".2"/>
        <texture name="skybox" type="skybox" builtin="gradient" rgb1=".1 .2 .3" rgb2="0 0 0" width="800" height="800" mark="random" markrgb="1 1 1"/>
        <material name="silver" specular="1" shininess="0.5" reflectance="1" emission="0.25" rgba="0.82 0.82 0.82 1"/>
        <material name="grey" specular="1" shininess="0.5" reflectance="1" emission="0.1" rgba="0.5 0.5 0.5 1"/>
        <material name="blue" specular="1" shininess="0.5" reflectance="1" emission="0" rgba="0.678 0.847 0.902 1"/>
        <material name="black" specular="1" shininess="0.5" reflectance="1" emission="0" rgba="0.2 0.2 0.2 1"/>
    </asset>
      <visual>
    <rgba haze=".3 .3 .3 1"/>
  </visual>
    <worldbody>
      <light directional="true" diffuse=".4 .4 .4" specular="0.1 0.1 0.1" pos="0 0 5.0" dir="0 0 -1" castshadow="false"/>
      <light directional="true" diffuse=".6 .6 .6" specular="0.2 0.2 0.2" pos="0 0 4" dir="0 0 -1" castshadow="false"/>
      <geom name="ground" type="plane" size="0 0 1" pos="0 0 0" quat="1 0 0 0" material="grid" condim="6"/>

      <body name="uskin_body" pos="0 0 0">
        <geom name="uskin_frame" condim="6" mass="10" type="box" pos="0 0 0.005" size="0.02 0.02 0.005" rgba="0 0 0 0.4"/>
        
        <body name="uskin_11" pos="0 0 0">
          <geom name="uskin_11" condim="6" mass="10" type="box" pos="0.01 0.01 0.011" size="0.007 0.007 0.001" rgba="1 1 1 0.6"/>
        </body>
        
        <body name="uskin_12" pos="0 0 0">
          <geom name="uskin_12" condim="6" mass="10" type="box" pos="-0.01 0.01 0.011" size="0.007 0.007 0.001" rgba="1 1 1 0.6"/>
        </body>

        <body name="uskin_21" pos="0 0 0">
          <geom name="uskin_21" condim="6" mass="10" type="box" pos="0.01 -0.01 0.011" size="0.007 0.007 0.001" rgba="1 1 1 0.6"/>
        </body>
        
        <body name="uskin_22" pos="0 0 0">
          <geom name="uskin_22" condim="6" mass="0.001" type="box" pos="-0.01 -0.01 0.011" size="0.007 0.007 0.001" rgba="1 1 1 0.6"/>
          <site name="force_sensor" type="box" pos="-0.01 -0.01 0.011" size="0.007 0.007 0.001" rgba="0.2 0 0 0.5"/>
        </body>

        <body name="box" pos="0 0.0 0.011" euler="0 0 0">
          <geom name="object" mass="1" condim="6" type="box" pos="0 0 0.01" size="0.02 0.02 0.01" rgba="0 0.2 0 0.3"/>
        </body>

      </body>

      <body name="act" pos="0 -0.1 0.022" >
        <joint name="act" type="slide" damping="10" axis="0 1 0"/>
        <geom name="act" mass="2.5" condim="6" type="box" pos="0 0 0" size="0.005 0.01 0.005" rgba="0 0 0.7 0.3"/>
      </body>

    </worldbody>

    

    <actuator>
      <motor name="act" joint="act" ctrllimited="true" ctrlrange="0 10" />
    </actuator>

</mujoco>
```

</details>

Here is a screenshot / video, illustrating my question:

rs545837, 2024-01-22:
From what I can understand you want to detect collision among the geometries. Detecting collision simply means that there will be a contact force. If that's what you are looking for, try to read about contype and conaffinity or the other way to do the same thing can be by using 
```xml 
<connect>
```
to declare the collision detection among the geometries individually.

yuvaltassa, 2024-01-22:
Also, look at [touch sensors](https://mujoco.readthedocs.io/en/latest/XMLreference.html#sensor-touch) and the [contactForce](https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#mj-contactforce) API function.

closed 2024-01-30 by quagla