google-deepmind/mujoco #1358: Workflow for creating custom environments

dantes-ai, 2024-01-22:
Hi,

I'm an engineer and I'm trying to use MuJoCo for scene-interaction tasks in various environments.

I would like to create rich environments in some 3D editor, and export them as an XMLs for MuJoCo. What's the recommended approach for achieving that? Currently, the only way to create environments is by changing the XML directly, or programmatically.

yuvaltassa, 2024-01-22:
> Currently, the only way to create environments is by changing the XML directly, or programmatically.

That is correct. We are actively working on procedural access to the model compiler (this has been requested multiple times, e.g. in #364) which is slated for H1 2024.

Closing as duplicate.

yuvaltassa, 2024-01-22:
cc @quagla for visibility