google-deepmind/mujoco #1361: AttributeError: 'NoneType' object has no attribute 'get_current_context'

hub2nature, 2024-01-23:
Hi,

I'm a student and I'm trying to use MuJoCo for hopper-v4.

I'm looking for some help with the error I got ( mentioned below). I am using mujoco 2.3.3, gymnasium 0.28.1, PyOpenGL=3.1.7, glfw=2.5.0

Exception ignored in: <function OffScreenViewer.__del__ at 0x7ca17d9facb0>
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/gymnasium/envs/mujoco/mujoco_rendering.py", line 191, in __del__
  File "/usr/local/lib/python3.10/dist-packages/gymnasium/envs/mujoco/mujoco_rendering.py", line 188, in free
  File "/usr/local/lib/python3.10/dist-packages/mujoco/glfw/__init__.py", line 35, in free
AttributeError: 'NoneType' object has no attribute 'get_current_context'
Exception ignored in: <function GLContext.__del__ at 0x7ca17d9e92d0>
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/mujoco/glfw/__init__.py", line 41, in __del__
  File "/usr/local/lib/python3.10/dist-packages/mujoco/glfw/__init__.py", line 35, in free
AttributeError: 'NoneType' object has no attribute 'get_current_context'

kevinzakka, 2024-01-24:
This should be fixed in https://github.com/google-deepmind/mujoco/commit/1c7809e54261009698b5baab569faf700c8f8dad and will be available in the next MuJoCo release.

kankanzheli, 2024-01-27:
Are you referring to the next MuJoCo release as the future version of MuJoCo or is the version I am currently using too old.

nutsintheshell, 2024-02-07:
> Are you referring to the next MuJoCo release as the future version of MuJoCo or is the version I am currently using too old.

I have the similar error like yours. I add `renderer.__del__()`at the end of the code and solve the problem. It's probably the problem of the current version.

thakur-sachin, 2024-08-27:
I had this error: `AttributeError: 'NoneType' object has no attribute 'eglQueryString'`

This error looks similar to something I faced. I resolved it by following these steps:

I resolved it by downloading mujoco zip, extracting mujoco210 to $home/.mujoco/

I then added the libglew.so to the LD_PRELOAD using 
export LD_PRELOAD=$LD_PRELOAD:/home/sachin/.mujoco/mujoco210/bin/libglew.so

closed 2024-08-30 by kevinzakka