google-deepmind/mujoco #1365: Why dont we expose MjStepArgs in the PreUpdate and PostUpdate EventHandler?

vidurvij-apptronik, 2024-01-24:
**Is your feature request related to a problem? Please describe.**
Would love to have access to model and data through PreUpdate and PostUpdate handles. I know I can get it using a reference to MjScene but it just makes the code cleaner.

**Describe the solution you'd like**
https://github.com/google-deepmind/mujoco/blob/c8e839a6662199cf94153906e1b11790dbe3e04d/unity/Runtime/Components/MjScene.cs#L82
https://github.com/google-deepmind/mujoco/blob/c8e839a6662199cf94153906e1b11790dbe3e04d/unity/Runtime/Components/MjScene.cs#L84C3-L84C39
Convert this to a template with MjStepsArgs

Balint-H, 2024-01-25:
I added the EventHandlers originally, and since using them now I agree with all of them using MjStepArgs, and I'm happy to submit a PR for that. Furthermore I'm considering marking the EventHandlers as `event`s explicitly. This means only MjScene is allowed to perform any operation on them (e.g. invoke them, clear them etc), everywhere else only subscribe/unsubscribe is available.

Would it break any functionality for you if I swap them for `event`s?

vidurvij-apptronik, 2024-01-26:
It wont break things for me as of now. I am using the event handlers with += and -= accesors which are the standard way of doing things anyway! Thanks for the jiffy reply.

Balint-H, 2024-02-05:
@erez-tom This issue can be closed now

closed 2024-03-27 by erez-tom