google-deepmind/mujoco #1368: Configuration for flexcomp upon model init

sachinkundu, 2024-01-26:
I'm using MuJoCo to do cloth simulation using [Flexcomp](https://mujoco.readthedocs.io/en/stable/XMLreference.html#body-flexcomp).

The definition is as follows 

```XML
<mujoco model="cloth">
    <option timestep="0.01" integrator="implicitfast" viscosity="0.3" solver="CG" tolerance="1e-6"/>

    <extension>
        <plugin plugin="mujoco.elasticity.shell"/>
    </extension>

    <worldbody>
        <geom name="floor" type="plane" size="0 0 .1"/>
        <light diffuse=".6 .6 .6" specular="0.2 0.2 0.2" pos="0 0 4" dir="0 0 -1"/>
        <body>
            <body name="object" pos="0 0 0.9">
                <flexcomp type="grid" count="20 10 1" spacing=".025 .025 .025" mass="1.0"
                          name="cloth" radius="0.003">
                    <contact condim="5" solref="0.001" friction="2"/>
                    <edge equality="true" damping="1.0"/>
                    <plugin plugin="mujoco.elasticity.shell">
                        <config key="poisson" value="0.4032740512363587"/>
                        <config key="thickness" value="0.0005"/>
                        <config key="young" value="85242.02369946481"/>
                    </plugin>
                </flexcomp>
                <site rgba="0 0 0 0" size="0.005" pos="0 0 -0.045" name="bottom_site"/>
                <site rgba="0 0 0 0" size="0.005" pos="0 0 0.03" name="top_site"/>
                <site rgba="0 0 0 0" size="0.005" pos="0.03 0.03 0" name="horizontal_radius_site"/>
            </body>
        </body>
    </worldbody>
</mujoco>
```

which works fine. I would like to understand how to set a configuration for the generated bodies(which have a pos and 3 slider joints in the compiler generated xml)

The genearated xml has

```XML
<body name="cloth_main" pos="0 0 0.9">
    <body name="cloth_cloth_0" pos="-0.2375 -0.1125 0">
          <inertial pos="0 0 0" mass="0.005" diaginertia="8.33333e-08 8.33333e-08 8.33333e-08"/>
          <joint pos="0 0 0" axis="1 0 0" type="slide"/>
          <joint pos="0 0 0" axis="0 1 0" type="slide"/>
          <joint pos="0 0 0" axis="0 0 1" type="slide"/>
          <plugin instance="flexcomp_cloth_cloth"/>
        </body>
199 other bodies...

   <deformable>
      <flex name="cloth_cloth" radius="0.003" body="cloth_cloth_0 cloth_cloth_1 ... " element="0 10 ...">
        <contact condim="5" friction="2 0.005 0.0001" solref="0.001 1"/>
        <edge damping="1"/>
      </flex>
    </deformable>
```

I would like to initialize the cloth in random crumpled configuration. How can I do that? Should I set the joint of each body?

I hope the question makes sense.

(the xml is being generated in robosuite but I think that should be of no significance)

yuvaltassa, 2024-01-26:
Probably easiest to save a keyframe?

1. Add `<size nkey="n"/>` to your model with some `n` > 0.
2. Load the model in `simulate`, repeat `n` times:
  a. Select keyframe `i` using the keyframe slider.
  b. Crumple the cloth.
  c. Click "save key".
3. Click "save XML", copy the keyframes from the saved XML into your XML (maybe deleting unused fields, e.g. you probably don't need "qvel").

closed 2024-01-26 by yuvaltassa

sachinkundu, 2024-01-29:
@yuvaltassa Thanks for the method. I could not make it work though.

I added `<size nkey="10"/>` to the model and loaded in simulate. First of all I cannot seem to modify pose of any point on the cloth. Double click on the cloth and use ALT+Mouse but it does not seem to change anything( I loaded other sample models and they seem to work). Not sure why it does not work with deformable objects.

I saved two keyframes in any case, and saved the xml and compared it to the original output where there was no  `<size nkey="10"/>`. When I diff the two outputs all I see is as follows.

```bash
diff mjmodel_nokf.xml mjmodel.xml 
3a4
>   <size nkey="10"/>
```

Not sure what I am doing wrong here.