google-deepmind/mujoco #1370: TypeError: 'type' object is not subscriptable @ mjx/__init__.py line 122

Servo97, 2024-01-26:
Hi, 
I'm trying to make mjx work on an m3 max mac. I have a conda env with python 3.8 (since jax-metal requires jax==0.4.11). 
I installed mujoco, mujoco-mjx, and brax using pip. But I get the following error when I try to execute the import block from the [tutorial](https://colab.research.google.com/github/google-deepmind/mujoco/blob/main/mjx/tutorial.ipynb#scrollTo=T5f4w3Kq2X14). 
My OS is Sonoma 14.1.1

TypeError                                 Traceback (most recent call last)
Cell In[1], [line 14](vscode-notebook-cell:?execution_count=1&line=14)
     [11](vscode-notebook-cell:?execution_count=1&line=11) import numpy as np
     [12](vscode-notebook-cell:?execution_count=1&line=12) from typing import Any, Dict, Sequence, Tuple, Union
---> [14](vscode-notebook-cell:?execution_count=1&line=14) from brax import base
     [15](vscode-notebook-cell:?execution_count=1&line=15) from brax import envs
     [16](vscode-notebook-cell:?execution_count=1&line=16) from brax import math

File [~/Applications/miniconda/envs/py38/lib/python3.8/site-packages/brax/__init__.py:19](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py38/lib/python3.8/site-packages/brax/__init__.py:19)
     [15](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py38/lib/python3.8/site-packages/brax/__init__.py:15) """Import top-level classes and functions here for encapsulation/clarity."""
     [17](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py38/lib/python3.8/site-packages/brax/__init__.py:17) __version__ = '0.9.4'
---> [19](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py38/lib/python3.8/site-packages/brax/__init__.py:19) from brax.base import Motion
     [20](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py38/lib/python3.8/site-packages/brax/__init__.py:20) from brax.base import State
     [21](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py38/lib/python3.8/site-packages/brax/__init__.py:21) from brax.base import System

File [~/Applications/miniconda/envs/py38/lib/python3.8/site-packages/brax/base.py:29](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py38/lib/python3.8/site-packages/brax/base.py:29)
     [27](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py38/lib/python3.8/site-packages/brax/base.py:27) from jax.tree_util import tree_map
     [28](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py38/lib/python3.8/site-packages/brax/base.py:28) import mujoco
---> [29](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py38/lib/python3.8/site-packages/brax/base.py:29) from mujoco import mjx
     [31](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py38/lib/python3.8/site-packages/brax/base.py:31) # f: free, 1: 1-dof, 2: 2-dof, 3: 3-dof
     [32](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py38/lib/python3.8/site-packages/brax/base.py:32) Q_WIDTHS = {'f': 7, '1': 1, '2': 2, '3': 3}

File [~/Applications/miniconda/envs/py38/lib/python3.8/site-packages/mujoco/mjx/__init__.py:18](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py38/lib/python3.8/site-packages/mujoco/mjx/__init__.py:18)
     [15](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py38/lib/python3.8/site-packages/mujoco/mjx/__init__.py:15) """Public API for MJX."""
...
    [120](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py38/lib/python3.8/site-packages/mujoco/mjx/_src/dataclasses.py:120) def tree_replace(
    [121](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py38/lib/python3.8/site-packages/mujoco/mjx/_src/dataclasses.py:121)     self, params: Dict[str, Optional[jax.typing.ArrayLike]]
    [122](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py38/lib/python3.8/site-packages/mujoco/mjx/_src/dataclasses.py:122) ) -> 'PyTreeNode':

TypeError: 'type' object is not subscriptable

Is that a known bug? Any ideas on how to fix it?

btaba, 2024-01-26:
Hi @Servo97 , I believe this is a python 3.8 issue, can you update the python version to at least 3.9 and try again?

See [here](https://github.com/google/brax/commit/a99298692ed21b2f959a629d68494ab54352ec6c) for context

Servo97, 2024-01-27:
Thanks @btaba! The import function is now working, but as soon as I run
`jit_step = jax.jit(mjx.step)

frames = []
mujoco.mj_resetData(mj_model, mj_data)
mjx_data = mjx.put_data(mj_model, mj_data)
while mjx_data.time < duration:
  mjx_data = jit_step(mjx_model, mjx_data)
  if len(frames) < mjx_data.time * framerate:
    mj_data = mjx.get_data(mj_model, mjx_data)
    renderer.update_scene(mj_data, scene_option=scene_option)
    pixels = renderer.render()
    frames.append(pixels)

media.show_video(frames, fps=framerate)` 

I get an XLA runtime error:

`XlaRuntimeError                           Traceback (most recent call last)
Cell In[9], [line 7](vscode-notebook-cell:?execution_count=9&line=7)
      [5](vscode-notebook-cell:?execution_count=9&line=5) mjx_data = mjx.put_data(mj_model, mj_data)
      [6](vscode-notebook-cell:?execution_count=9&line=6) while mjx_data.time < duration:
----> [7](vscode-notebook-cell:?execution_count=9&line=7)   mjx_data = jit_step(mjx_model, mjx_data)
      [8](vscode-notebook-cell:?execution_count=9&line=8)   if len(frames) < mjx_data.time * framerate:
      [9](vscode-notebook-cell:?execution_count=9&line=9)     mj_data = mjx.get_data(mj_model, mjx_data)

    [... skipping hidden 14 frame]

File [~/Applications/miniconda/envs/py39/lib/python3.9/site-packages/jax/_src/compiler.py:255](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py39/lib/python3.9/site-packages/jax/_src/compiler.py:255), in backend_compile(backend, module, options, host_callbacks)
    [250](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py39/lib/python3.9/site-packages/jax/_src/compiler.py:250)   return backend.compile(built_c, compile_options=options,
    [251](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py39/lib/python3.9/site-packages/jax/_src/compiler.py:251)                          host_callbacks=host_callbacks)
    [252](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py39/lib/python3.9/site-packages/jax/_src/compiler.py:252) # Some backends don't have `host_callbacks` option yet
    [253](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py39/lib/python3.9/site-packages/jax/_src/compiler.py:253) # TODO(sharadmv): remove this fallback when all backends allow `compile`
    [254](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py39/lib/python3.9/site-packages/jax/_src/compiler.py:254) # to take in `host_callbacks`
--> [255](https://file+.vscode-resource.vscode-cdn.net/Users/nagababa/Sarvesh/Simulation/~/Applications/miniconda/envs/py39/lib/python3.9/site-packages/jax/_src/compiler.py:255) return backend.compile(built_c, compile_options=options)

XlaRuntimeError: UNKNOWN: [/var/folders/80/fpjtzy0572dbqzxnjspxvn2r0000gn/T/ipykernel_26874/889828542.py:7:0](https://file+.vscode-resource.vscode-cdn.net/var/folders/80/fpjtzy0572dbqzxnjspxvn2r0000gn/T/ipykernel_26874/889828542.py:7:0): error: failed to legalize operation 'mhlo.pad'
[/var/folders/80/fpjtzy0572dbqzxnjspxvn2r0000gn/T/ipykernel_26874/889828542.py:7:0](https://file+.vscode-resource.vscode-cdn.net/var/folders/80/fpjtzy0572dbqzxnjspxvn2r0000gn/T/ipykernel_26874/889828542.py:7:0): note: called from
[/var/folders/80/fpjtzy0572dbqzxnjspxvn2r0000gn/T/ipykernel_26874/889828542.py:7:0](https://file+.vscode-resource.vscode-cdn.net/var/folders/80/fpjtzy0572dbqzxnjspxvn2r0000gn/T/ipykernel_26874/889828542.py:7:0): note: see current operation: %1438 = "mhlo.pad"(%1437, %103) {edge_padding_high = dense<1> : tensor<1xi64>, edge_padding_low = dense<0> : tensor<1xi64>, interior_padding = dense<1> : tensor<1xi64>} : (tensor<1xsi32>, tensor<si32>) -> tensor<2xsi32>`

Not sure if I should open a new issue. But any help on how to make mjx work on m3 mac would be greatly appreciated!

erikfrey, 2024-02-02:
@Servo97 we've played a bit with `jax-metal` and have seen similar issues.  We're quite excited to support MJX on metal and have chatted with the `jax-metal` folks a few times.

I think issues created at https://github.com/google/jax will be routed to the `jax-metal` team, if you'd like to open an issue there please mention this issue and we'll keep an eye on it.

closed 2024-02-02 by erikfrey