google-deepmind/mujoco #1378: How to embed the viewer into a QT window?

Zzx-J, 2024-01-30:
Hi all,

I'm new to MuJoCo and I'm trying to use MuJoCo for realtime rendering.

In my project I have made a client GUI using pyQT, for the ease of use I would like to combine the passive viewer and the QT GUI together as a single window.

Currently I am able to create the QT GUI and the viewer as two seperate windows, the code structure looks like this:

```
class MuJoCoViewerThread(threading.Thread):
    def __init__(self):
        super().__init__()

    def run(self):
    
        model = mujoco.MjModel.from_xml_path(filename=xml, assets=None)
        data = mujoco.MjData(model)
        with mujoco.viewer.launch_passive(model, data) as viewer:
               #do some tasks
```
        
```
class UDPServerWindow(QMainWindow):
    def __init__(self):
        super().__init__()
        self.initUI()

    def initUI(self):
        self.viewer_thread = MuJoCoViewerThread()
        self.viewer_thread.start()
        #do some tasks
````

Waht should I do to combine them to a single window? Ideally I would like the window look like this:
[image]

Zzx-J, 2024-04-08:
> Hey @Zzx-J, I am looking for a solution as well for exactly what you asked. Were you able to embed mujoco into a single Qt app?

Sadly, no. I am still actively looking for solutions. My program currently uses 2 windows.

bakehamster, 2024-04-08:
Hey @Zzx-J,
Check these out, they might help you out.
[mujoco_qt](https://gist.github.com/JeanElsner/755d0feb49864ecadab4ef00fd49a22b)
[qmujocosim](https://github.com/hesic73/QMuJoCoSim?tab=readme-ov-file)
[mujoco_viewer](https://github.com/rohanpsingh/mujoco-python-viewer)

Zzx-J, 2024-04-10:
> Hey @Zzx-J, Check these out, they might help you out. [mujoco_qt](https://gist.github.com/JeanElsner/755d0feb49864ecadab4ef00fd49a22b) [qmujocosim](https://github.com/hesic73/QMuJoCoSim?tab=readme-ov-file) [mujoco_viewer](https://github.com/rohanpsingh/mujoco-python-viewer)

Awesome! The first one seems really helpful! Thank you very much!!!

closed 2024-04-10 by Zzx-J